fwupd/firmware-lenovo #304: Lenovo Thinkpad T490s fails to upgrade IME 192.85.1869

bowlofeggs, 2022-12-21:
This issue was originally posted by @jknaus in https://github.com/fwupd/firmware-lenovo/issues/162#issuecomment-1274452039 and I will just copy their text here:

> Same issue here with a T490s:
> 
> BIOS Revision: 1.75
> Firmware Revision: 1.19 (N2JET97W)
> 
> Going into UEFI graphical interface at boot, the BIOS shows
> Intel Management Engine ("ME Firmware Version") is already 12.0.85.1869
> 
> However, fwupdmgr get-updates still erroneously offers
> 
> [...]
> Intel Management Engine
> [...]
> Current version: 192.81.1753
> [...]
> New version: 192.85.1869
> [...]
>  • Update Management Engine Firmware version to 12.0.85.1869.
> 
> This is also offered by Gnome-Software (running Fedora36 workstation).
> My fwupdmgr version is 1.8.5
> 
> Is this issue the same as https://github.com/fwupd/firmware-lenovo/issues/224?

kmauleon, 2022-12-21:
hi @bowlofeggs thank you for reporting... will ping owner

kmauleon, 2023-01-17:
According to owner, they will implement fix in BIOS 1.79.... please wait until release...

kmauleon, 2023-03-07:
BIOS 1.79 has been put to stable... please check... thank you

kmauleon, 2023-03-10:
please check with BIOS 1.79 should be fixed... closing

closed 2023-03-10 by kmauleon