fwupd/firmware-lenovo #305: ThinkPad X1 Carbon 10th / X1 Yoga 7th:         New version:      0.1.34 breaks wifi

nbecker, 2022-12-21:
After applying latest upgrades wifi did not work.  On boot, hung on 
Job NetworkManager-wait-online.service/start running
but then would start after maybe 30s.
But then only 2.4G wifi worked, not 5G wifi.

I started downgrading the firmware.   I believe the problem was caused by
└─System Firmware:
  │   Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
  │   Previous version:   0.1.34

I downgraded to 0.1.32.  That caused a hard lock, which is another issue.  But on eventually rebooting, wifi is fine.
Attached:
output of 
fwupdmgr history
fwupdmgr get-updates

[history.txt](https://github.com/fwupd/firmware-lenovo/files/10278958/history.txt)
[upgrades.txt](https://github.com/fwupd/firmware-lenovo/files/10278964/upgrades.txt)

mrhpearson, 2022-12-21:
Hi - I just checked this out on my X1Y7 and I'm connected to my 5G Wifi and it looks to be working fine with that FW version. Tested under Ubuntu 22.04; Fedora 37 and Debian testing

Are you able to check with a different access point? I can flag this to the FW team - but having reproduction steps makes it so much easier to investigate.

Anything in the kernel logs that might point at a problem?

nbecker, 2022-12-21:
I tested on Fedora 37, x1 carbon gen 10.

I'm sorry but I'm not willing to try this again.  After installing the offending update which caused the wifi issue, I used fwupdmgr downgrade to revert it.  This cause the machine to totally lock up and be unbootable.  It hung at the end of installing the downgrade, at 100%, machine will reboot/shutdown.  At this point even holding the power button did nothing.  I only recovered by letting the battery die and then powering up.  I'd like to help, but I'm not trying this again.

I don't think it's an access point issue because when the problem occurred the boot hung for about 30s on  
Job NetworkManager-wait-online.service/start running
I believe it's not connecting to access point at boot, but at login (I could be wrong though).

I tried searching journalctl but so far haven't spotted anything.  I'll try to keep searching.  One problem is there have been many reboots since and I haven't been able to spot a boot where the problem occurred.  BTW I tried both 
kernel-6.0.12-300.fc37.x86_64
kernel-6.0.13-300.fc37.x86_64

MalikMlitat, 2023-01-01:
same within ubuntu 22.04.1, updated from 0.1.9 -> 0.1.34 

lucky that TP reset to the older version (0.1.9)