fwupd/firmware-lenovo #306: Can't update ME firmware on Thinkpad L13 gen1 to 225.67.2046

p4block, 2023-01-02:
I'm on latest UEFI firmware, even downgraded to see if that fixed anything and tried some workarounds people discussed on similar threads such as enabling csm to no avail.

Current ME fw version is 225.53.1649, verified by looking at the UEFI firmware menu.

When rebooting to update fwupd efi binary briefly prints to screen something about capsule/device error and reboots. 

```
 sudo fwupdmgr update
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 225.53.1649 to 225.67.2046?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad L13 Consumer ME Firmware en_US update MEFW 14.1.67.2046      ║
║                                                                              ║
║ Hot Fix Release U/Y/S/H  Disable auto-sleep when the device is connected to  ║
║ a USB port  Improve performance of the analog sticks Some new functionality  ║
║ has also been added:  Add support for Joystick Mode to OS X                  ║
║                                                                              ║
║ 20R30004SP must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
```

Additionally, I think that changelog for an ME update is quite weird and may be a mixup (macOS? Joysticks?? :confused:  )

hughsie, 2023-01-03:
I've fixed the update release notes; the default text was used by the uploader somehow :)

> something about capsule/device error

Can you video it and get the exact message please?

p4block, 2023-01-03:
[image]

I forced a refresh and changelog still the same though, I guess it takes time to propagate :thinking:

kmauleon, 2023-01-16:
hi... will ping the owner for this one... thank you for reporting

kmauleon, 2023-02-03:
hi... new firmware was reuploaded can you please try?  thank you very much.
Corporate : https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR15RM.firmware
Consumer : https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR15RN.firmware

p4block, 2023-02-03:
Updated now without issues :smile:

closed 2023-02-03 by p4block