fwupd/firmware-lenovo #307: Lenovo L13 Yoga (20R6): Could not apply capsules: Device Error (Intel Management Engine (225.65.1969 → 225.67.2046))

Swiftb0y, 2023-01-12:


kmauleon, 2023-01-13:
hi... are you updating MEFW together with other firmware? if yes... can you please just try updating firmware one at a time

Swiftb0y, 2023-01-13:
Hi there :)
Initially yes, I tried updating the "system firmware" and MEFW at the same time. Then I found an issue where someone fixed their problem by doing the upgrade separately. By doing that, I was able to update the other firmware successfully. Now I'm just updating the single ME firmware, which still fails. 
Also I just found #306 after filing this one which seems exactly like the issue I have (also same device), I'm sorry for creating a duplicate issue.

kmauleon, 2023-01-16:
hi... will ping the owner for this one... thank you

Swiftb0y, 2023-01-29:
Hey there, I'm sorry for pinging this again. Has there been any response from the owner? I'd gladly work with them to provide more information if needed. Is there anything else I could help with in order to move this forward?

kmauleon, 2023-01-29:
hi @Swiftb0y sorry for the delay but due to Chinese New Year holidays... I have not received any feedback yet from owner...will follow up again... thank you very much

Swiftb0y, 2023-01-29:
No stress, thank you for the quick response.

kmauleon, 2023-02-03:
hi... new firmware was reuploaded can you please try?  thank you very much.
Corporate : https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR15RM.firmware
Consumer : https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR15RN.firmware

Swiftb0y, 2023-02-03:
Thank you, I can also confirm that the plain update via `fwupdmgr update` is working now.

closed 2023-02-03 by Swiftb0y