fwupd/firmware-lenovo #308: Firmware update for Samsung PM9B1 NVMe

stuarthayhurst, 2023-01-19:
As documented by [this thread](https://lore.kernel.org/all/20221116171727.4083-1-git@augustwikerfors.se/T/), the PM9B1 has an issue where it doesn't wake up correctly from suspend. This was fixed by a firmware update, which has been released for Dell laptops, but is nowhere to be seen for Lenovo.

hughsie, 2023-01-20:
If it helps, this is the one for Dell: https://fwupd.org/lvfs/devices/com.dell.PM9B1.firmware

mrhpearson, 2023-01-21:
The Yoga 7 isn't in the Linux program so I don't have any levers to pull on there. Don't suppose you happen to know if it's on any Thinkpads?

I'll check with the SSD team and see if I can find out what is going on; from the thread it looked like we would be doing an update but if that part isn't on any Linux certified platforms I can't, sadly, promise it will make it onto LVFS.

Mark

stuarthayhurst, 2023-01-21:
I can't see it on any Thinkpads, unless it's on the L13/14 Gen 3, E14 Gen 4 or a couple others but those might just be other Samsung 512GB PCIe 4 2242 NVMes, as they aren't marked as PM9B1 like they are on the parts search for the Yoga 7 Gen 7 (14ARB7).

I have a couple questions about the firmware update though, but I can't really find anyone to answer them, so if you're able to get an answer to any, that'd be really appreciated:
 - If it won't be available through the LVFS, which tool will it be on (Windows Update, Lenovo Vantage, something else)?
   - I've seen a few be available through some Lenovo NVMe Firmware Update tool, but I can't find any download for the latest version of this
 - Do you know roughly how long these sorts of firmware updates take to be released?
 - Do you know how I'd be able to tell when an update is ready?

Thanks

mrhpearson, 2023-01-25:
I'm still trying to track down who would know about this - but it's Chinese New Year so the people I think I need to talk to are on vacation. I think I got confirmation that it's not on the Thinkpads; need to check on the ThinkCenters.

For how this is delivered from Windows and the timelines - afraid I don't know the answer (I never run Windows :) ) but I would assume it was offered through Vantage once it is ready? Once people are back I'll see if I can find out.

Mark

stuarthayhurst, 2023-01-25:
Thank you, I'll keep my eye out for an update then

am-silex, 2023-02-05:
I've got IdeaPad 5 Pro 16ARH7 - Type 82SN with SAMSUNG PM9B1 512G 2242 PCIe G4V x4. It does has a major problem with resume after suspend cycle. I've tried suggested GRUB options like acpiphp.disable to no avail. Not only gets nvme lost after resume, but even during normal workload. 
It's interesting that Lenovo points that updating this particular ssd is restricted [here](https://fwupd.org/lvfs/search?csrf_token=IjQzYzY1YTFhY2NmYzE1MTc4Yzk5MDk5NDEyZTRlZGFjZGY1MWZkMjMi.Y95rBA.h8tyMDtBBGiBPm4gsvhfrI2ieIA&value=pm991)

stuarthayhurst, 2023-02-05:
If you're comfortable compiling a kernel, you can use the patch from [this kernel thread](https://lore.kernel.org/all/20221116171727.4083-1-git@augustwikerfors.se/T/) to work around this issue for now. As Samsung patched it in firmware, kernel policy says it won't be patched in the kernel, so we have to wait for the firmware update for a proper fix :/

FSchumacher, 2023-03-11:
The  IdeaPad Flex 5 16ABR8 has the same problem. It can be fixed by the kernel patch, but I would like to run a standard distro kernel instead of compiling my own kernels.
I haven't found a firmware update for the drive on Lenovos site or Samsungs.

mrhpearson, 2023-03-13:
I'm stuck on this one - the FW team reached out to Samsung to see if there were fixes that we should be picking up and Samsung reported back that there are no Linux issues reported against this part :(

I'll try reaching out to the Samsung person on the upstream mailing list...because this is going nowhere fast right now.

stuarthayhurst, 2023-03-13:
Thank you for taking the time to chase this up :)

takuvata, 2023-03-24:
> If you're comfortable compiling a kernel, you can use the patch from [this kernel thread](https://lore.kernel.org/all/20221116171727.4083-1-git@augustwikerfors.se/T/) to work around this issue for now.

ThinkBook 13s gen4 ARB. I have tried building latest Fedora 38 beta kernel with that patch, but it did not solve the issue. However I can confirm that while using live media - it suspends and resumes perfectly fine. Also couple of times it managed to sort of resume while installed and behaved pretty much how one would expect when OS block device vanishes. So - pretty sure that ssd is the culprit.

stuarthayhurst, 2023-03-24:
Since the PM9B1 seems to be on some ThinkBook 14 systems, can it be uploaded to the LVFS?

mrhpearson, 2023-03-27:
Hi - which Thinkbook 14? If it's on the G4 Intel I have a chance of making something happen.

Mark

SeekingGoodTech, 2023-04-19:
This issue is affecting my own Lenovo Yoga 7, but also other examples have shown up on the Lenovo Linux forum, including the IdeaPad 3 Slim:
https://forums.lenovo.com/t5/Ubuntu/Ideapad-Slim-3-Ubuntu-Suspend-Causes-System-to-Crash-Requiring-a-Hard-Reboot/m-p/5217318?page=1#5954366

0x9fff00, 2023-04-20:
@mrhpearson Any updates about the ThinkBook?

mrhpearson, 2023-04-21:
No update I'm afraid. I've forwarded the thread we had on the Lenovo forum about this issue to the SSD team to point out that there is an issue impacting SSDs but not heard back yet (but it's been a crazy week). I'll send them a nudge.

stuarthayhurst, 2023-05-12:
Recompiling my kernel to patch this every kernel update (which I can't really miss out, as they keep fixing other issues) since November is starting to get tedious. Is there any sort of contact form I can fill out that would point out users are still having issues?

I appreciate you taking the time to help out here, especially since I haven't seen a single mention of this from anyone else at Lenovo.

mrhpearson, 2023-05-12:
I nudged the FW team on this again this week and they've asked Samsung to look at it. I don't have any updates back yet though I'm afraid.

stuarthayhurst, 2023-08-04:
Well this is promising https://fwupd.org/lvfs/devices/com.lenovo.PM9B1.firmware.v220116

Sadly, I have the `MZAL4512HBLU-00BL2`, not the `MZAL4512HBLU`, so my SSD is ignored by both the Windows tool and `fwupd`

takuvata, 2023-08-08:
> Sadly, I have the `MZAL4512HBLU-00BL2`, not the `MZAL4512HBLU`, so my SSD is ignored by both the Windows tool and `fwupd`

Same here

mrhpearson, 2023-08-09:
Looks like I managed to miss updating this thread for the original update....apologies. As you saw - FW updates were released.

I've sent a question to the SSD team asking what the difference is for the 00BL2 device and why it's not covered. Hopefully it's just a snafu and they missed some IDs....

mrhpearson, 2023-08-10:
OK, I hope I have this correct - but the 00BL2 device is specific to Lenovo, and FW hasn't been released for that as it is still under going some testing. An issue was found at one of the stages that is being reviewed by the team and Samsung.
Once that issue has been resolved then it will be released. Not sure if that particular version will be on LVFS as that device isn't in (to my knowledge) any of our Linux certified platforms....but I'll keep an eye out for any updates.
Mark

Jalopy-Tech, 2023-08-19:
A patch has been released for Linux kernels 6.5-rc6, 6.4.11 and 6.1.46. I have model MZAL41T0HBLB-00BL2. I have tried kernels 6.5-rc6 and 6.4.11: suspend and wake up work without a problem.

Notes:
I used the application Mainline Kernels to install the kernels.

I use Open Razer. I had to install gcc-13 to avoid a clash with 6.5-rc6.

liepumartins, 2023-08-21:
@Jalopy-Tech which patch are You talking about? The [one](https://lore.kernel.org/all/20221116171727.4083-1-git@augustwikerfors.se/T/) previously linked seems to have hit a dead end.
From the thread:
> We don't want to add quirks for things fixed by firmware updates,
see the recently posted features and quirks policy.

Is there another patch? 
Using 6.2 kernel (ubuntu) here.

Jalopy-Tech, 2023-08-21:
https://github.com/liepumartins I'm talking about patches that were incorporated into Linux kernels 6.5-rc6,, 6.4.11 and 6.1.46. These kernels were released last week.

See discussion:
https://github.com/tomsom/yoga-linux/issues/9

0x9fff00, 2023-08-21:
@liepumartins It’s the same patch but the recent discussion is in another thread: https://lore.kernel.org/all/36319a0f-34a6-9353-bc52-4d4d0fac27a5@amd.com/T/#u

adggit, 2023-08-23:
My Yoga 6 13ABR8 has an MZAL4256HBJD-00BL2 and I can confirm that kernel 6.4.11 has fixed the resume issue.  Waiting for Lenovo to release proper drive firmware (for a permanent/hardware fix). Thanks to Mark Pearson for following up on this issue.

coldtobi, 2023-08-24:
FWIIW, my X13 Gen 2 (WK20), purchased in March, did come with a  MZVL41T0HBLB-00BL7, and (Debian unstable) fwupdtool offers me an update from BL2QHXC7 -> CL2QHXC7
(I'm experiencing the issue that sometimes the device does not come back from S2R, so I hope this will improve things…)

liepumartins, 2023-08-29:
Using 6.4.11 kernel on Lenovo Legion with MZAL4512HBLU-00BL2 helps.
Before that I was pulling my hair, trying myriad of acpi/nvme related kernel parameters. :sweat_smile:

Weird that one kernel discussion states the opposite (will not patch) of the other (patch made). 

Still waiting on Samsung/Lenovo to release actual firmware update for -00BL2

am-silex, 2023-08-29:
It's interesting that this discussion has been going on from the start of this year. I can't even imagine how someone is able to use this series of laptops with Linux productively. However, I'm using IdeaPad 5 Pro 16ARH7 - Type 82SN with Windows and it's unstable too - constantly rebooting while in sleep mode, or can't be woken up after night/couple of hours of sleep.

liepumartins, 2023-09-22:
Any update on -00BL2? @mrhpearson maybe knows?

Currently using patched kernel as a workaround, however using custom kernel is a hassle when it comes to maintained  packages like virtualbox.

mrhpearson, 2023-11-03:
SSD team confirmed that there were issues found with the planned update release that are still under investigation by Samsung. So no FW update ready yet I'm afraid.

bitx01, 2023-12-05:
Hi, any update on -00BL2?
Just bought an Ideapad Flex 5 with MZAL4512HBLU-00BL2 and having the same issue with Ubuntu.
I'm thinking in buying a new SSD

eebssk1, 2024-03-02:
Getting the error when setting up lvm on the second handed -00BL2. Device vanished after setup.

TanZhiwen2001, 2024-03-15:
> SSD team confirmed that there were issues found with the planned update release that are still under investigation by Samsung. So no FW update ready yet I'm afraid.

There's an update for PM9B1 actually.
https://support.lenovo.com/ky/en/downloads/ds119265
You can just download the FWNVA66, and it should include the FWNVA63's update.

FWNVA63 2023/09/21
[Important updates]
- Samsung PM9B1 Firmware update mandatory replacement.

[Problem fixes]
Samsung PM9B1
1.       Fixed an issue the performance might drop.
2.       Fixed an issue that BSOD or system hang up may occur during operation.

stuarthayhurst, 2024-08-24:
With `DL2QHXC7` also available on the LVFS now, is there any progress on getting the update for `MZAL4512HBLU-00BL2`?