fwupd/firmware-lenovo #309: Brand new T14s Gen 3 (AMD) self bricked an hour after unboxing (Windows, using Lenovo app)

zorrobyte, 2023-01-20:
This seems to be a Linux repo, this is a Windows machine ordered new from Lenovo.com

Sorry, I don't have firmware versions, etc. Just unboxed, ran the Lenovo tool for updates, machine rebooted and was doing a firmware update, rebooted, then bricked. No fan.

Tried:
* reset button. 3 flashes, reboot. Also tried 4 times, then hold for 30 seconds.
* Pulling battery and cmos battery, holding reset/power, reboot
* Everything above in various conditions like plugged in/unplugged

Seems I'll be going Framework and BIOS/firmware issues are pretty common with Lenovo stuff, eh?

hughsie, 2023-01-20:
> ran the Lenovo tool for updates

What Lenovo tool please?

@superm1 @mrhpearson any ideas?

zorrobyte, 2023-01-20:
> > ran the Lenovo tool for updates
> 
> What Lenovo tool please?
> 
> @superm1 @mrhpearson any ideas?

Lenovo Advantage. On the phone today to see if I can get an RMA, ha!

Again, likely wrong repo. Just wanted to at least try as it's the first machine I ever had commit suicide right out of the box and it looks like there's some issues with the T14s

zorrobyte, 2023-01-20:
Update, they are doing an express replacement.

Question is, I'm terrified of doing another BIOS update or auto updating the new machine when it gets here with Lenovo Advantage. Do I just never update (bad security practice), or is there a recommended way to update the BIOS without it bricking?

hughsie, 2023-01-20:
> is there a recommended way to update the BIOS without it bricking?

Well, it should be fwupd... @mrhpearson any ideas on what went wrong here? Do we get any post-replacement debug dumps or something?

mrhpearson, 2023-01-21:
Afraid nothing clever to add as they were using Vantage which I have no insight too - following up with Lenovo support is the right option. 
If you know what you were updating from and going to that might be useful - I could check the internal update notes and check. Last year we had a lot of issues with updates on the AMD platforms if going from very old to very new - but that was all pre-production....it shouldn't impact you if you got a production unit (though stranger things have happened)

zorrobyte, 2023-01-21:
> Afraid nothing clever to add as they were using Vantage which I have no insight too - following up with Lenovo support is the right option. If you know what you were updating from and going to that [...]

Will get version info and re-try updating once I receive the new unit! Closing for now, thank you!

closed 2023-01-21 by zorrobyte