fwupd/firmware-lenovo #310: Issue updating Lenovo Thinkpad P14s AMD Gen1 UEFI 0.1.41 -> 0.1.42

BentHaase, 2023-01-22:
When trying to update my Lenovo Thinkpad P14s AMD Gen 1 (`20Y1`) from UEFI `0.1.41` -> `0.1.42` I get the following error and the update fails:

[image]

Text form:

```
fwupd-efi version 1.3
WARNING: QueryCapsuleCapabilities failed, assuming EfiResetWarn: Unsupported
WARNING: Could not apply capsule update: Not Found
WARNING: Could not apply capsules: Not Found
Reset System
```

Could this issue be caused by my disk encryption (LUKS) which is not unlocked in the pre initramfs / UEFI state?

Version info:

```bash
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.9
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.3
runtime   org.freedesktop.fwupd         1.8.9
runtime   org.kernel                    6.0.18-300.fc37.x86_64

$ lsb_release -a
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 37 (Thirty Seven)
Release:	37
Codename:	ThirtySeven

$ uname -a
Linux fedora 6.0.18-300.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Sat Jan 7 17:10:00 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

```
---

When trying to run the update as [per Lenovos README](https://download.lenovo.com/pccbbs/mobiles/r1bul73w.txt) (manually) (checksum of downloaded .zip archive has been checked) I get this error:

```bash
$ sudo fwupdmgr install r1bul73w.cab
Decompressing…           [   -                                   ]
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer
```

mrhpearson, 2023-01-25:
I don't think LUKS should matter here. The main issue I know of around FW updates is that the TPM PCR0 measurement can change on a BIOS update - which can sometimes throw things, but I don't believe that would apply in this case.

I tried this update on my T14 AMD G1 (which is essentially the same as the P14s) and it went through successfully. In my case I just triggered it with 'fwupdmgr update'.

There's a very similar issue on the X1 Carbon G10 which the FW team are working on so it's a bit worrying that this is showing up in multiple places now.

Are there any other settings you've changed in the BIOS? Can you try with secure boot disabled - just to rule that out (I have it disabled on mine currently)

Thanks
mark

mrhpearson, 2023-01-25:
Tagging @kmauleon in case there is any input from debugging on the X1C10

kmauleon, 2023-01-26:
hi @mrhpearson we are still waiting for user's reply https://github.com/fwupd/firmware-lenovo/issues/301#issuecomment-1376949536
hi @BentHaase yes please try with Secure Boot disabled and update firmware one at a time, BIOS first... thank you very much

BentHaase, 2023-01-27:
> hi @mrhpearson we are still waiting for user's reply [#301 (comment)](https://github.com/fwupd/firmware-lenovo/issues/301#issuecomment-1376949536) hi @BentHaase yes please try with Secure Boot disabled and update firmware one at a time, BIOS first... thank you very much

I will have to try and downgrade my BIOS first (if possible). In the meantime I did the update manually via USB stick in hopes this resolves some nasty Suspend / Sleep issues (spoiler: *it doesn't*) which the AMD Gen 1 and Gen 2 devices have been suffering for ages now.

If I can downgrade my BIOS I will try upgrading via `fwupdmgr` and Secure Boot disabled (if enabled).

kmauleon, 2023-02-01:
hi please see recommendation from dev owner....  thank you very much...

Could you help ask customer to disable “Secure Boot” in BIOS Setup then do BIOS online upgrade with 'fwupdmgr update'? If still fail, please help collect log by ‘fwupdmgr get-devices --show-all-devices’ and send to us.
We checked BIOS 1.42 downloading statistics from LVFS([https://fwupd.org/lvfs/firmware/14723/analytics/reports](https://apc01.safelinks.protection.outlook.com/?url=https%3A%2F%2Ffwupd.org%2Flvfs%2Ffirmware%2F14723%2Fanalytics%2Freports&data=05%7C01%7Ckmauleon%40lenovo.com%7C30cedfc029044a2ac9d708db03fce224%7C5c7d0b28bdf8410caa934df372b16203%7C0%7C0%7C638108156932366746%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=TisA07xoXvWoNbNzYocT0Kqz%2BC9ODFJDkG8CJvT6fRo%3D&reserved=0)), there is no other failure report. We suppose there is something special that need to be distinguished in the failed system.

ehaag, 2023-02-03:
Not sure how helpful this is (versus creating a separate issue), but I've had the exact error messages displayed by BentHaase when attempting to update my T15g Gen 1 from 0.1.30 to 0.1.32 for the past few months. Secure boot already turned off, drive not LUKS encrypted.

enyone, 2023-02-12:
Same problem here
```
$ sha256sum r1bul73w.zip
7b757df11c1990dcf1b75b9485d64ef14f6a1678f525265f3cd579bf78130a0e  r1bul73w.zip
$ unzip r1bul73w.zip 
Archive:  r1bul73w.zip
  inflating: r1bul73w.cab            
$ sudo fwupdmgr install r1bul73w.cab
Decompressing…           [   -                                   ]
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer
```

kmauleon, 2023-02-16:
hi all as per owner... can you please send results of fwupdmgr get-devices --show-all-devices

ehaag, 2023-02-18:
To be clear, I'm referring to these error messages:
````
fwupd-efi version 1.3
WARNING: QueryCapsuleCapabilities failed, assuming EfiResetWarn: Unsupported
WARNING: Could not apply capsule update: Not Found
WARNING: Could not apply capsules: Not Found
Reset System
````

@kmauleon , see output of `[~/]$ fwupdmgr get-devices --show-all-devices`:
[get_devices.txt](https://github.com/fwupd/firmware-lenovo/files/10772726/get_devices.txt)

kmauleon, 2023-02-20:
hi @ehaag thank you for the logs... forwarded to ODM. let's wait for feedback

kmauleon, 2023-03-02:
hi @ehaag replied by ODM owner as below....
        We can’t duplicate the issue, and checked the LVFS web there just one failed report, but the failed is caused by ESP too small.
        We supposed there is something environmental differences in failed system need to clear.
        BTW, the log get_devices.txt from the previous mail seem is no error.

ehaag, 2023-03-02:
If you cannot reproduce then I shall manually update with the vendor's bootable USB method.

kmauleon, 2023-03-03:
hi @ehaag have you tried updating via bootable usb?

ehaag, 2023-03-03:
@kmauleon Yes, the bootable USB method succeeded on the first attempt. Now showing:
``` 
    $fwupdmgr get-devices
    . . .
    ─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.32
│ │   Minimum Version:    0.1.15
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              e1678770-bdc9-47f9-8d90-340e28b6f196
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
    . . .
```

enyone, 2023-03-03:
But this bug supposed to be about `0.1.41` -> `0.1.42` update ?

kmauleon, 2023-03-06:
hi @enyone  please try as below....

set OnlyTrusted=false in /etc/fwupd/daemon.conf
sudo fwupdmgr install r1bul73w.cab
after successful update set OnlyTrusted=true in /etc/fwupd/daemon.conf

kmauleon, 2023-03-07:
this might be related as well >> https://github.com/fwupd/fwupd/issues/1042
@hughsie is there a similar fix in fedora?

hughsie, 2023-03-07:
@kmauleon 1.2.5 went into Fedora on Feb 25 **2019** :)

enyone, 2023-03-07:
@kmauleon it would be glad to do it for you but I'm not going to flash a firmware to this only laptop unless it has went through a valid signature verification.

kmauleon, 2023-03-07:
@hughsie oh...haha so this might be different root cause... though similar error
@enyone was the cab file downloaded from Lenovo Support Site or LVFS?

enyone, 2023-03-07:
@kmauleon from assumed Lenovo support site
```
$ sha256sum r1bul73w.zip
7b757df11c1990dcf1b75b9485d64ef14f6a1678f525265f3cd579bf78130a0e  r1bul73w.zip
```

kmauleon, 2023-03-07:
have you tried downloading the file from LVFS site >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1BET.firmware

hughsie, 2023-03-07:
@kmauleon why does Lenovo host the unsigned versions? can I make it easier for you to download the right cab archive so you can sync it to the lenovo download site?

enyone, 2023-03-07:
I believe the one provided by Lenovo is messed up...
```
$ sha256sum r1bul73w.cab
8cf140eee6c0a87f40c1dc9ce5db510cb886de921c167e746015661454665730  r1bul73w.cab

$ binwalk r1bul73w.cab

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Microsoft Cabinet archive data, 8382684 bytes, 2 files
```

And then the one from LVFS:
```
$ sha256sum r1bul73w_2.cab
6b9395807b51b62fcb62492b2c09f8e66d4a12f3760ef887fdc15a566cb8fac3  r1bul73w_2.cab

$ binwalk r1bul73w_2.cab

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Microsoft Cabinet archive data, 17922251 bytes, 5 files
845           0x34D           Copyright string: "copyright holders of the LVFS be liable for"
324462        0x4F36E         SHA256 hash constants, little endian
329526        0x50736         Zlib compressed data, best compression
464726        0x71756         Zlib compressed data, best compression
566830        0x8A62E         Zlib compressed data, best compression
.
.
.
```

kmauleon, 2023-03-08:
> @kmauleon why does Lenovo host the unsigned versions? can I make it easier for you to download the right cab archive so you can sync it to the lenovo download site?

sorry @hughsie can't answers to this... not sure how ODM BIOS owner process their cab files
@ChiWei-Chen can we get answers from ODM BIOS owner for this case?

kmauleon, 2023-03-08:
@hughsie I compared the bin files of the lenovo cab file and that of lvfs... they are just the same... so I think LVFS is adding some files to cab file during upload to LVFS.... lenovo cab file has 2 files in it (bin and xml).  so I guess the warning "firmware signature missing or not trusted" appeared just because the cab file came from another site (which is Lenovo) and not from LVFS itself.  please correct if I misunderstood... thank you very much...

mrhpearson, 2023-03-10:
I don't know why we would host it on the Lenovo site as well...I have enough trouble getting the FW cab uploaded in a timely manner without trying to track two separate locations. Having too many options just increases the number of things that can go wrong in my opinion and the LVFS upload has all the extra checks implemented that catches mistakes. I would rather Linux users use LVFS and not manually track/download/install from the support site.

@kmauleon - let's discuss internally; but posting my thoughts publicly in case I'm missing something obvious that for some users that prevents them using LVFS (let me know)

hughsie, 2023-03-10:
Agreed with all that, thanks @mrhpearson  -- if there's anything I can provide to make it easier for Lenovo -- let me know.

kmauleon, 2023-03-12:
@enyone let's close this one as you were able to update to latest already... thank you very much
@mrhpearson noted on this... thank you very much

closed 2023-03-12 by kmauleon