fwupd/firmware-lenovo #311: Incompatible update displayed sporadically on ThinkPad P14s Gen 3 AMD

jluebbe, 2023-01-25:
From time to time, `fwupdmgr get-upgrades` displays an update which seems to be for a different product. It's not always displayed, but I've seen it several times by now:
```
LENOVO 21J5CTO1WW
│
├─System Firmware:
│ │   Device ID:          f95c9218acd12697af946874bfe4239587209232
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.36
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              e40809b1-7d9f-4dc4-7c0f-8ace63b43687
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │
│ └─ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD:
│       New version:      0.1.38
│       Remote ID:        lvfs
│       Release ID:       16804
│       Summary:          Lenovo ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD System Firmware
│       License:          Proprietary
│       Size:             34,6 MB
│       Created:          2021-07-13
│       Urgency:          High
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:
│       Problem fixes:
│
│       • Fixed security issues.
│       • Fixed issue that keyboard 50% backlight will get off under Linux after the system resumes from suspend.
│       • Fixed an issue where WMI change the BIOS display setting not work in setup.
│       Issues:           309662
│                         CVE-2022-34303
│                         CVE-2022-34302
│                         CVE-2022-34301
│
```

According to dmidecode, I'm currently using:
```
…
Handle 0x0014, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: R23ET60W (1.30 )
        Release Date: 09/14/2022
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
…
                UEFI is supported
        BIOS Revision: 1.30
        Firmware Revision: 1.23

Handle 0x0015, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 21J5CTO1WW
        Version: ThinkPad P14s Gen 3
        Serial Number: <redacted>
        UUID: c580b7cc-2fd9-11b2-a85c-8a48952c5600
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_21J5_BU_Think_FM_ThinkPad P14s Gen 3
        Family: ThinkPad P14s Gen 3
…
Handle 0x0040, DMI type 140, 15 bytes
ThinkPad Embedded Controller Program
        Version ID: R23HT57W
        Release Date: 10/20/20
…
```

This matches the previous version listed on https://support.lenovo.com/us/en/downloads/ds557681-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-t14-gen-3-type-21cf-21cg-t16-gen-1-type-21ch-21cj-p16s-gen-1type-21ck-21cl. According to that page, the correct update would be to 1.32.

Compared to the Lenovo page, the update offered by fwupd has a different device name and a different version. What's going on here?

mrhpearson, 2023-01-25:
The P14s G3 and T14 G3 share the same BIOS Firmware (along with the T16/P16s).

This is just a case of the FW team missing the name out of the title - I'll let them know as it looks like they've done it for all the updates for this platform (for my reference -  LO-2246)

Mark

jluebbe, 2023-01-25:
Is there a reason for the version difference (1.32 on lenovo.com compared to 0.1.38 reported by fwupd)?

mrhpearson, 2023-01-27:
That is weird....I don't know why and will have to ask the FW team who are away for Chinese New Year this week.
I can ask why when they return....it's peculiar.

mrhpearson, 2023-01-30:
Hi @jluebbe - they confirmed we are adding P14s to the platform list for the next version.

For the version numbers - apparently this was a mistake with the version numbers and I've not completely understood how they did it, but by mixing up decimal and hex they accidentally skipped 7 versions in LVFS...
I'm confirming how they will fix this. Sigh.

mrhpearson, 2023-03-09:
Should all be fixed now. Let me know if any questions; and thanks for flagging this.

jluebbe, 2023-03-09:
> Should all be fixed now. Let me know if any questions; and thanks for flagging this.

Hmm, should this be fixed for the currently available firmware? Or only for the next release?
```
% fwupdmgr refresh --force
Updating lvfs
Downloading…             [    \                                  ]
Successfully downloaded new metadata: 4 local devices supported
% fwupdmgr get-updates    
Devices with no available firmware updates: 
 • ELAN0678:00 04F3:3195
 • Integrated RGB Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • MZVL22T0HBLB-00BL7
LENOVO 21J5CTO1WW
│
├─Embedded Controller:
│ │   Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.23
│ │   Minimum Version:    0.1.23
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               6be552b4-eda3-46a7-ab1b-d8ba1ff451bd
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14 Gen 3 AMD/T16 Gen 1 AMD/P16s Gen 1 AMD:
│       New version:      0.1.25
│       Remote ID:        lvfs
│       Release ID:       17503
│       Summary:          Lenovo ThinkPad T14 Gen 3 AMD/T16 Gen 1 AMD/P16s Gen 1 AMD Embedded Controller Firmware
│       License:          Proprietary
│       Size:             1,2 MB
│       Created:          2022-12-26
│       Urgency:          High
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:      
│       This stable release fixes the following issues:
│       
│       • Fixed vantage will show a few seconds AC Power when plugging in a USB-C adapter
│       • Fixed system do not pop up slow charge when plugging in 30W adapter.[DFT0197186]
│       
│       Some new functionality has also been added:
│       
│       • Modify Charge limit table for the requirements of thermal team
│     
├─System Firmware:
│ │   Device ID:          f95c9218acd12697af946874bfe4239587209232
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.36
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              e40809b1-7d9f-4dc4-7c0f-8ace63b43687
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD:
│       New version:      0.1.38
│       Remote ID:        lvfs
│       Release ID:       16804
│       Summary:          Lenovo ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD System Firmware
│       License:          Proprietary
│       Size:             34,6 MB
│       Created:          2021-07-13
│       Urgency:          High
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:      
│       Problem fixes:
│       
│       • Fixed security issues.
│       • Fixed issue that keyboard 50% backlight will get off under Linux after the system resumes from suspend.
│       • Fixed an issue where WMI change the BIOS display setting not work in setup.
│       Issues:           309662
│                         CVE-2022-34303
│                         CVE-2022-34302
│                         CVE-2022-34301
…
```

So I now additionally see a FW update for the embedded controller, but both updates still use the 0.* versions and don't mention my model (P14s Gen 3 AMD).

mrhpearson, 2023-03-09:
I'm sorry - I managed to mixup the Intel and the AMD platforms. This issue is still open :(

jluebbe, 2023-04-14:
@mrhpearson Do you have any news on this issue?

mrhpearson, 2023-04-17:
Looks like they've been having some issues with the releases - there is activity on LVFS and the latest version was uploaded last week, but is under embargo still. 
I've asked for an update on when it will be released

jluebbe, 2023-08-03:
It seems there have been some releases in the meantime, but there is *still* a version mismatch:
```
  ├─ThinkPad T14/P14s Gen 3 T16/P16s Gen 1 AMD:
  │     New version:      0.1.41
  │     Remote ID:        lvfs
  │     Release ID:       26179
  │     Summary:          Lenovo ThinkPad T14/P14s Gen 3 T16/P16s Gen 1 AMD System Firmware
  │     License:          Proprietary
  │     Size:             34,6 MB
  │     Created:          2021-07-13
  │     Urgency:          High
  │     Vendor:           Lenovo
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     This version will fix the following issue:
  │     
  │     • Fixed an issue where garbage displays when perform 'Reset system to Factory Default' after change 'Setup UI' to Simple Text.
```
While this is referred to `0.1.41` here, https://download.lenovo.com/pccbbs/mobiles/r23uj66wd.txt has this change description for `<R23UJ65WD> UEFI: 1.35 / ECP: 1.25`, issued on 2023/04/06 instead of 2021-07-13. So something is still very wrong here.

The description seems better now:
``` diff
- Summary:          Lenovo ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD System Firmware
+ Summary:          Lenovo ThinkPad T14/P14s Gen 3 T16/P16s Gen 1 AMD System Firmware
```

mrhpearson, 2023-08-09:
I believe they're 'stuck' on the versioning because of the original error. They can't go back to an older version number because it would really mess things up so I believe they've just decided to roll with it (which is confusing). I was hoping they'd skip a whole bunch so that everything synced up...I'll ask if they can do that for the next round.

jluebbe, 2023-08-09:
Perhaps they could just drop the 0. prefix, which is also confusing on its own. :⁠-⁠)

jluebbe, 2023-11-28:
Is there any news on this issue? Currently, no firmware update is offered, although I've not updated the FW so far.
```
% fwupdmgr get-updates 
Devices with no available firmware updates: 
 • ELAN0678:00 04F3:3195
 • Integrated RGB Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • MZVL22T0HBLB-00BL7
 • System Firmware
No updates available
```

mrhpearson, 2023-11-29:
I'm having to refresh my memory as this has been a while..

72W (which is being labelled as 1.48 on LVFS) is up but under embargo. I need to check why it hasn't been released, but I know the AMD platforms have had some issues recently so it might be delayed
70W (1.46) is released to stable. If you're on an older version than that let me know...

Mark

jluebbe, 2023-11-29:
Ah, thanks. It seems that even after `fwupdmgr refresh --force` it needed some time to recognize the available updates.

I now see:
```
  ├─ThinkPad T14/P14s Gen 3 T16/P16s Gen 1 AMD:
  │     New version:      0.1.46
  │     Remote ID:        lvfs
  │     Release ID:       53536
  │     Summary:          Lenovo ThinkPad T14/P14s Gen 3 T16/P16s Gen 1 AMD System Firmware
  │     License:          Proprietary
  │     Size:             34,6 MB
  │     Created:          2021-07-13
  │     Urgency:          High
  │     Vendor:           Lenovo
  │     Release Flags:    • Is upgrade
  │     Description:      
  │     This release contains the following changes:
  │     
  │     • [Important] Remove MSC mode.
  │     • (New) Add support for Maori keyboard.
  │     • (New) Enable smart card preboot function.
  │     • (Fix) Fixed issue where Lenovo logo interface lost during restart after set Supervisor Password/Power-On Password/NVMe1Password.
  │     • (Fix) Fixed issue where keyboard backlight does not remain after returning from modern standby.
  │     • (Fix) Fixed issue where system hang up at Pre-Boot Authentication interface after perform a normal scan operation with unregistered fingerprint for 3 attempts.
  │     • (Fix) Fixed issue where system will shutdown when dock power button is pressed for 4s in the process of BIOS upgrade.
  │     • (Fix) Fixed issue where “invalid critical threshold” issue under linux.
  │     • (Fix) Fixed issue where system can't change power-saver mode under linux after wake up from DC+Lid open.
  │   
```

This mentions 0.1.46 (not 1.46 as you said or 1.40 as the current lenovo website offers), which is confusing. Also, does the `Created:          2021-07-13` indicate that this is something else?

How does one match the LVFS releases to the ones directly from Lenovo?