fwupd/firmware-lenovo #313: T570P51s 0.1.51 and 0.1.52 require 0.1.64 to install

jonbrenner, 2023-01-29:
Firmware obtained from lvsh: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN1VET.firmware

The following firmware incorrectly requires a higher version (0.1.64) to be installed:
* [Lenovo-ThinkPad-T570P51s-SystemFirmware-0.1.51.cab](https://fwupd.org/downloads/891ce6c766966ea09590c899cf930765526a1ad50eabd60cdf62d144730fcf26-Lenovo-ThinkPad-T570P51s-SystemFirmware-0.1.51.cab)
* https://fwupd.org/downloads/2e7b3c7bb16aae742bc10dc606b03aaa37139af39346c9e72e36b4e4988bcf27-Lenovo-ThinkPad-T570P51s-SystemFirmware-0.1.52.cab

Error message: `Not compatible with firmware version 0.1.50, requires >= 0.1.64`

hughsie, 2023-01-30:
@mrhpearson can you tag the right person please -- I think one of the requirements is screwy.

mrhpearson, 2023-01-31:
Of course. I've raised internal ticket  LO-2253 to get this resolved. 
I'd like to understand how this happened as well - it's a peculiar mistake to make (and not catch in testing more worryingly)

hughsie, 2023-01-31:
@mrhpearson there are no signed tests, so I don't think anyone actually tested this before uploading. If they did, they didn't send the signed report. Maybe this gives more weight to making them compulsory? https://lists.linuxfoundation.org/pipermail/lvfs-announce/2022-November/000075.html

mrhpearson, 2023-02-01:
Hi @hughsie - Let's sync up offline so I can flag it to the team here but no concerns from me; seems a no-brainer. 

Testing and review is part of our sign-off process so should have been done. I was going to pick your brains on if this was maybe something that wouldn't have been caught with an older fwupdmgr maybe? I was waiting to get details back from the FW team first though - China is just coming back online after their new year holiday so I should hear back from them soon.

julian-klode, 2023-02-03:
P52s 0.1.35 (not on lvfs afaict) is affected too.

ChiWei-Chen, 2023-03-13:
Hi,
The FW team released 1.53 and the requirement seems removed, could anyone give it a try and see if it works? Thanks!
https://fwupd.org/downloads/b79c9b8908f270e705b8975616e3183a8ee5e41fe14c60b2293e041124b3db28-Lenovo-ThinkPad-T570P51s-SystemFirmware-0.1.53.cab

SillieWous, 2023-03-13:
I can confirm that it works, writing this on a 0.1.53 machine. ^^

jonbrenner, 2023-05-30:
1.53 was downloaded and it installed correctly. Thanks for the resolution.

There is a separate issue where updating blows away my EFI boot entry, which is annoying. I'll create/update a separate issue for that.

closed 2023-05-30 by jonbrenner