fwupd/firmware-lenovo #314: Is it possible to update 40AJ docking station firmware from a linux box?

laur89, 2023-01-29:
Don't have access to a windows machine that's compatible with named docking station -- any way to upgrade from linux?
Note running `fwupdtool get-devices` docked vs undocked, then this block is present only when docked:
```
LENOVO 20Y1CTO1WW
│
├─USB2.0 Hub:
│ │   Device ID:          81c9a317b30acf7c0cd967f2ce6925d518ef19a8
│ │   Summary:            USB 3.x hub
│ │   Current version:    50.33
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              xxx
│ │                       xxx
│ │                       xxx
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Unsigned Payload
│ │ 
│ └─USB2.0 Hub:
│       Device ID:        51af0a4d70ae1190499dc16830f8f4e856f07aef
│       Summary:          USB 3.x hub
│       Current version:  50.34
│       Vendor:           VIA Labs, Inc. (USB:0x17EF)
│       Install Duration: 15 seconds
│       GUIDs:            xxx
│                         xxx
│                         xxx
│                         xxx
│                         xxx
│       Device Flags:     • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│                         • Unsigned Payload
│     
├─USB3.1 Hub:
│ │   Device ID:          38fe7919e3b6102195936a2b7486d9f5a54afe92
│ │   Summary:            USB 3.x hub
│ │   Current version:    50.33
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              xxx
│ │                       xxx
│ │                       xxx
│ │                       xxx
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Unsigned Payload
│ │ 
│ └─USB3.1 Hub:
│       Device ID:        67048ba3beaccc7f5ac6a70cbc62a663c9fb82f4
│       Summary:          USB 3.x hub
│       Current version:  50.34
│       Vendor:           VIA Labs, Inc. (USB:0x17EF)
│       Install Duration: 15 seconds
│       GUIDs:            xxx
│                         xxx
│                         xxx
│                         xxx
│                         xxx
│                         xxx
│       Device Flags:     • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│                         • Unsigned Payload
│     
```

mrhpearson, 2023-01-31:
Afraid not, at least officially - the docking team aren't putting the FW for that dock on LVFS.

laur89, 2023-01-31:
Thanks for the confirmation!

closed 2023-01-31 by laur89