fwupd/firmware-lenovo #315: Lenovo T14 System Firmware update fail due to low battery

StefanKaeser, 2023-02-01:
**Describe the bug**
I have a Lenovo T14 Gen 2 Machine Types 20W0 with a system manufacture date prior to 2021/07 and like described [here](https://support.lenovo.com/eg/en/solutions/ht512607) I also have an issue with not being able to charge the battery.

Unfortunately I am below 25% charge can can therefore not do a system firmware update using `fwupdmgr get-updates
`.
I have found a similar issue https://github.com/fwupd/firmware-lenovo/issues/301, but for another laptop and battery model, where an battery firmware was provided on LVFS. I will provide the information that were asked for in this issue.

**Outputs**
sudo fwupdtool get-devices –verbose

```
Loading…                 [-                                      ]15:51:14:0042 FuEngine             failed to get releases for UEFI Device Firmware: No releases found: no HWIDs matched 8a5186b0-5fcb-5247-8296-7b21b0bea516
15:51:14:0044 FuEngine             failed to get releases for Intel Management Engine: No releases found: no HWIDs matched 8a5186b0-5fcb-5247-8296-7b21b0bea516
Loading…                 [***************************************]
20W0000EGE
│
├─11th Gen Intel Core™ i5-1135G7 @ 2.40GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000a4
│     Vendor:             Intel
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│
├─EM120R GL:
│     Device ID:          bba54f76de66958e5d70964b6ddd2d1e6c7dae9e
│     Summary:            Mobile broadband device
│     Current version:    EM120RGLAPR02A07M4G_10.010.10.010
│     Vendor:             quectel (PCI:0x1EAC)
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              53f8dd6d-c5ee-5a79-9da1-f98d39e28699 ← PCI\VID_1EAC&PID_1001&REV_0000&CARRIER_ROW_COMMERCIAL
│                         4294f9ad-e7b2-53fb-bd16-be64092ebb1d ← PCI\VID_1EAC&PID_1001&REV_0000
│                         3b2b9e7f-2620-546f-becb-727b6e4b662f ← PCI\VID_1EAC&PID_1001
│                         8efebe65-85c5-58ff-937c-51f626d8fe2a ← PCI\VID_1EAC
│     Device Flags:       • System requires external power source
│                         • Supported on remote server
│                         • Updatable
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.8
│     Minimum Version:    0.1.8
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Error:       Cannot install update when the lid is closed, Cannot install update when system battery is not at least 25%
│     GUID:               2a5d161c-0ba2-4136-94b3-c027f0a7e5fc
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│
├─Integrated Camera:
│     Device ID:          3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627
│     Current version:    60.4
│     Vendor:             Azurewave (USB:0x13D3)
│     Serial Number:      0000
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUIDs:              1b2c4ed0-8e31-5d65-884b-a599e07e2260 ← USB\VID_13D3&PID_5406&REV_6004
│                         db842a92-1141-541b-965a-2733a5726e17 ← USB\VID_13D3&PID_5406
│     Device Flags:       • Updatable
│
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
[... 385 lines cut ...]
  battery
    present:             yes
    state:               pending-charge
    warning-level:       none
    energy:              10.86 Wh
    energy-full:         47.67 Wh
    energy-rate:         0 W
    charge-cycles:       N/A
    percentage:          22%
    icon-name:          'battery-low-charging-symbolic'

Daemon:
  daemon-version:  0.99.17
  on-battery:      no
  lid-is-closed:   yes
  lid-is-present:  yes
  critical-action: PowerOff
```

**Additional questions**
- Operating system and version: Ubuntu 22.04.1 LTS
- Have you tried rebooting? Yes
- Is secure boot enabled? Disabled.

Is it possible to also provide me a battery firmware on LVFS?

mrhpearson, 2023-02-06:
Hi,
The T14 G2 Intel battery FW is on LVFS: 

https://fwupd.org/downloads/46c7820e9575bb2d9926b3df9c86b75f51e1b5f6738a0f4759164a837ff40176-Lenovo-ThinkPad-T14-Gen-2i-P14s-Gen-2i-BatteryFirmware-258.0.0.10.cab 

But the GUID in there doesn't match any of your devices. I think you're also on a really old BIOS/EC FW so I wonder if that's part of the problem.

I'll ask the FW team what the recommendations are. Internal ticket is LO-2267

ChiWei-Chen, 2023-02-07:
Hi @StefanKaeser , I'm working with @mrhpearson.
We got feedback from FW team, they are asking if it is possible to try temporarily upgrading the FW with another power source(AC adapter)? Thanks.

StefanKaeser, 2023-02-07:
Hi @ChiWei-Chen,

thank you very much for your effort. I am right now using my laptop with an AC adapter, but can not update the FW due to restriction that the battery has to be at least 25% full, even through I am plugged in.

mrhpearson, 2023-02-07:
Hi @StefanKaeser

As a note - I was incorrect above; I must have looked at the wrong file and the GUID does match one of your device IDs.

I think you can do:
fwupdmgr update f7576b01-100e-4ec1-96cf-ccb69f4ce87e

This worked for the user in the thread you linked to
That should try and update just the battery FW. If it works then hopefully your battery will charge and you can update the other FW items normally. If it doesn't work please let us know and we'll follow up with the FW team.

Mark

StefanKaeser, 2023-02-08:
Hi @mrhpearson,

Thanks for your suggestion.

I tried running the command your posted, but unfortunatley without success, see the output:
```
>fwupdmgr update f7576b01-100e-4ec1-96cf-ccb69f4ce87e
No updatable devices
```
I also downloaded the file you linked to and tried installing it, but this also didn't work, see the output:
```
>fwupdmgr install 46c7820e9575bb2d9926b3df9c86b75f51e1b5f6738a0f4759164a837ff40176-Lenovo-ThinkPad-T14-Gen-2i-P14s-Gen-2i-BatteryFirmware-258.0.0.10.cab
Decompressing…           [***************************************]
No supported devices found
```

mrhpearson, 2023-02-08:
That's a shame - I was hoping that would work. We're asking the FW team for suggestions.

ChiWei-Chen, 2023-02-10:
@StefanKaeser I just got reply from FW team:

> could you try this ,because 258.0.0.10.cab was different FU package.
> 
> fwupdmgr install 862c7b9c9b7f9d70a6f1091860c0949495043778f60622566c600e26c8af1e84-Lenovo-ThinkPad-T14-Gen-2i_ThinkPad-P14s-Gen-2i-BatteryFirmware-260.0.0.9.cab
> 
You can get it from this link:
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN34BL_00060109.firmware

Thanks.

StefanKaeser, 2023-02-10:
@ChiWei-Chen Thank you very much for this new suggestion.
It first complained that I was not on AC adapter , but after rebooting with AC adapter plugged in i unfortunateley get the battery < 25% restriction again.

```
>fwupdmgr install 862c7b9c9b7f9d70a6f1091860c0949495043778f60622566c600e26c8af1e84-Lenovo-ThinkPad-T14-Gen-2i_ThinkPad-P14s-Gen-2i-BatteryFirmware-260.0.0.9.cab
Decompressing…           [***************************************]
Device UEFI Device Firmware [f95c9218acd12697af946874bfe4239587209232] does not currently allow updates: Cannot install update when system battery is not at least 25%
```

ChiWei-Chen, 2023-02-14:
@StefanKaeser Thanks for sharing the information, I see the battery team is discussing with the BIOS team about this, I'll keep you updated if we got a conclusion.

hughsie, 2023-02-15:
@ChiWei-Chen if this is a limit imposed on updating from fwupd, would you like me to remove the battery check for specific GUIDs? If you can give me a list I can add them to the quirks list.

hughsie, 2023-02-15:
e.g. something like https://github.com/fwupd/fwupd/pull/5536

StefanKaeser, 2023-02-15:
> @StefanKaeser , the BIOS team suggested to use **fwupdate** instead of **fwupdmgr** as below. Could you give it a try?
> 
> ```
>         # su
>         # fwupdate -a xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx firmware.bin
> 
> Here, xxxxx is GUID for target device.
> ```
This did the trick, my battery is charging again. If it is charged above 25% I will update the other firmwares.
**Thank you all so much!**