fwupd/firmware-lenovo #316: Upgrade Intel Management Engine fails on X13Yoga

bmn-m, 2023-02-03:
**Describe the bug:**
Upgrade Intel Management Engine from 225.60.1807 to 225.67.2046 on
Lenovo ThinkPad X13 Yoga Gen 1 20SX002XGE fails with: Cannot find ME client as seen in #217 

**Outputs**
See #217

**Additional questions**

Operating system and version: Ubuntu 22.04 Kernel 6.1.8
Have you tried rebooting? Yes
Is secure boot enabled? Disabled.

Thanks

Blaimi, 2023-04-28:
Same here with fedora 38, secure boot enabled. Also facing the message mentioned in #217.

```
user@ningal ~ $ sudo fwupdmgr refresh --force
Wird aktualisiert lvfs
Herunterladen …          [    \                                  ]
Neue Metadaten wurden erfolgreich heruntergeladen: 6 lokale Geräte unterstützt
user@ningal ~ $ sudo fwupdmgr get-devices
LENOVO 20SX002XGE
│
├─CometLake-U GT2 [UHD Graphics]:
│     Gerätekennung:      5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Aktuelle Version:   02
│     Anbieter:           Intel Corporation (PCI:0x8086)
│     GUIDs:              5589d313-88e5-54d3-8f4c-b2ee3882d4bc ← PCI\VEN_8086&DEV_9B41
│                         9779d679-0f31-5dd5-b3c2-f6ff275a9ca9 ← PCI\VEN_8086&DEV_9B41&REV_02
│                         59c26d2f-d3ad-5f25-bda8-0d42ccd93659 ← PCI\VEN_8086&DEV_9B41&SUBSYS_17AA22AD
│                         f048bb9a-6083-5672-a6f5-9b3b64d28e6f ← PCI\VEN_8086&DEV_9B41&SUBSYS_17AA22AD&REV_02
│                         25be981a-c8a2-5d3a-82cc-39c2a795effb ← PCI\VEN_8086&DEV_9B41&REV_00
│                         05383be5-bd62-5d13-9e7f-66f01f10ee24 ← PCI\VEN_8086&DEV_9B41&SUBSYS_17AA22AD&REV_00
│     Device Flags:       • Internes Gerät
│                         • Kryptografische Hash-Verifizierung ist verfügbar
│   
├─Core™ i7-10510U CPU @ 1.80GHz:
│     Gerätekennung:      4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Aktuelle Version:   0x000000f0
│     Anbieter:           Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         d5092916-1868-5182-8c70-12f83dab6372 ← CPUID\PRO_0&FAM_06&MOD_8E&STP_C
│     Device Flags:       • Internes Gerät
│   
├─Integrated Camera:
│     Gerätekennung:      0fef0a0c55f6442bffaebd774ae771341c89571b
│     Aktuelle Version:   59.18
│     Anbieter:           Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Seriennummer:       0001
│     GUIDs:              95b07a8e-2063-5025-80b5-1fcf4ca8e9e3 ← USB\VID_04F2&PID_B6CB
│                         fd6f6106-dc43-55ad-a882-54904f5c3fbc ← USB\VID_04F2&PID_B6CB&REV_5918
│     Device Flags:       • Aktualisierbar
│   
├─Intel Management Engine:
│     Gerätekennung:      349bb341230b1a86e5effe7dfe4337e1590227bd
│     Zusammenfassung:    UEFI ESRT device
│     Aktuelle Version:   225.53.1649
│     Minimale Version:   0.0.1
│     Anbieter:           Lenovo (DMI:LENOVO)
│     Aktualisierungsstatus:Erfolg
│     GUID:               e84807e1-f83c-4cb0-ac65-b598a55d844e
│     Device Flags:       • Internes Gerät
│                         • Aktualisierbar
│                         • System benötigt externe Stromquelle
│                         • Unterstützt auf dem entfernten Server
│                         • Benötigt einen Neustart nach der Installation
│                         • Das Gerät ist während der Dauer der Aktualisierung nutzbar
│   
├─MZVLB512HBJQ-000L7:
│     Gerätekennung:      3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Zusammenfassung:    NVM Express solid state drive
│     Aktuelle Version:   5M2QEXF7
│     Anbieter:           Samsung (NVME:0x144D)
│     Seriennummer:       S4ENNX3N639011
│     GUIDs:              47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         0b4d773a-7ac3-58c1-a541-e22ef1cdfe02 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801&REV_00
│                         f4ceb2a8-09c4-55c0-af2a-13ccd63925b1 ← SAMSUNG MZVLB512HBJQ-000L7
│     Device Flags:       • Internes Gerät
│                         • Aktualisierbar
[... 160 lines cut ...]
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Intel Management Engine von 225.53.1649 auf 225.67.2046 aktualisieren?       ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad X13 Yoga Gen 1 Consumer ME Firmware                          ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Disable auto-sleep when the device is connected to a USB port              ║
║ • Improve performance of the analog sticks                                   ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for Joystick Mode to OS X                                      ║
║                                                                              ║
║ 20SX002XGE muss während der gesamten Dauer der Aktualisierung an eine        ║
║ Stromquelle angeschlossen bleiben, um Schäden zu vermeiden.                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Operation durchführen? [Y|n]: n
Request canceled
```

kmauleon, 2023-05-02:
hi @bmn-m do you still see the error?
hi @Blaimi can you please try with secure boot disabled?

Blaimi, 2023-05-03:
sorry, I did the update already with Lenovo Vantage.

bmn-m, 2023-05-04:
> hi @bmn-m do you still see the error? 
Hi, unfortunately the problem persists.
Devices that were not updated correctly:
 • Intel Management Engine (225.60.1807 → 225.67.2046)
"UpdateError" : "failed to run update on reboot".

kmauleon, 2023-05-04:
hi @bmn-m can we have the results of below command please
fwupdmgr get-devices

bmn-m, 2023-05-07:
```
20SX002XGE
│
├─Integrated Camera:
│     Device ID:          0fef0a0c55f6442bffaebd774ae771341c89571b
│     Current version:    59.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Serial Number:      0001
│     GUID:               fd6f6106-dc43-55ad-a882-54904f5c3fbc ← USB\VID_04F2&PID_B6CB&REV_5918
│                         95b07a8e-2063-5025-80b5-1fcf4ca8e9e3 ← USB\VID_04F2&PID_B6CB
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    225.60.1807
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               e84807e1-f83c-4cb0-ac65-b598a55d844e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─MZVLB512HBJQ-000L7:
│     Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:            NVM Express solid state drive
│     Current version:    5M2QEXF7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S4ENNX3N605470
│     GUID:               0b4d773a-7ac3-58c1-a541-e22ef1cdfe02 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801&REV_00
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d ← NVME\VEN_144D&DEV_A808&REV_00
│                         47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         f4ceb2a8-09c4-55c0-af2a-13ccd63925b1 ← SAMSUNG MZVLB512HBJQ-000L7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─Prometheus:
│ │   Device ID:          d432baa2162a32c1554ef24bd8281953b9d07c11
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3478575
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   Serial Number:      245612081112399
│ │   GUID:               09e8bf16-3e69-50f5-bb66-c7a040248352 ← USB\VID_06CB&PID_00BD&REV_0000
│ │                       8088f861-6318-5b1e-9ce4-fbddbedb09ac ← USB\VID_06CB&PID_00BD
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:        8dcffb5d059857368caa2b7b22371ebd831b0c0c
│       Summary:          Fingerprint reader config
│       Current version:  0028
│       Minimum Version:  0028
│       Vendor:           Synaptics (USB:0x06CB)
│       GUID:             7c5a1e62-38fa-5859-9337-09dbac6377e4 ← USB\VID_06CB&PID_00BD-cfg
│                         21a62213-7012-552c-9f1a-a51d07c87242 ← USB\VID_06CB&PID_00BD&CFG1_3462&CFG2_0
│       Device Flags:     • Updatable
│                         • Supported on remote server
[... 79 lines cut ...]
      Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
      Summary:            UEFI ESRT device
      Current version:    22808
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               6842df28-6cc4-4ecf-a861-0ab176892a1e
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
________________________________________________

Devices that were not updated correctly:

 • Intel Management Engine (225.60.1807 → 225.67.2046)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]:
y
Target:                  https://fwupd.org/lvfs/firmware/report

```

kmauleon, 2023-05-25:
hi @bmn-m please see reply from ODM as below....  thank you very much.
------------------------
>> The MEFW of Consumer/Corporate have been upload the LVFS before and the both file already display “Stable” without problem.
Currently, there is a problem on Consumer MEFW, It’s ok on Corporate MFW.

The problem already solved without issues on Consumer MEFW, please double check it. 
-------------------------

Can you please try this again...
fwupdmgr refresh --force
fwupdmgr update

bmn-m, 2023-05-28:
No change, it does not work.

kmauleon, 2023-06-08:
can you please try this instead
download cab file here >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2URN.firmware
navigate to the location of the cab file
execute as below

fwupdmgr --allow-older --allow-reinstall install [cab filename].cab

bmn-m, 2023-09-14:
Sorry for my late answer, still no change, it does not work. Tested it even with the newest Update.

kmauleon, 2023-09-14:
@bmn-m can you please check fwupd version?

bmn-m, 2023-09-15:
Package: fwupd
Version: 1.7.9-1~22.04.3

kmauleon, 2023-09-18:
sorry @bmn-m no feedback yet from MEFW owner... in the meantime, you may want to try this >> https://github.com/fwupd/firmware-lenovo/issues/217#issuecomment-1105950678

kmauleon, 2023-10-17:
@bmn-m ODM dev said he just put it to stable... can you please check again... please execute
fwupdmgr reinstall

bmn-m, 2023-10-17:
@kmauleon the released Update is only for Corporate ME Firmware N2UR**M**16W not for Consumer ME Firmware N2UR**N**16W.
So the installation fails with `No supported devices found`. The latest release for Consumer Hardware is from 2023-07-24 
`fwupdmgr reinstall` fails with `Unable to locate release for Intel Management Engine version 225.60.1807` which is my current firmware.

kmauleon, 2023-10-17:
@bmn-m can you please try the following

fwupdmgr refresh --force
fwupdmgr update DEVICE-ID

bmn-m, 2023-10-17:
@kmauleon 

I did the following:

```
fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 4 local devices supported

fwupdmgr update
Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 225.60.1807 to 225.67.2046?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad X13 Yoga Gen 1 Consumer ME Firmware                          ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Disable auto-sleep when the device is connected to a USB port              ║
║ • Improve performance of the analog sticks                                   ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for Joystick Mode to OS X                                      ║
║                                                                              ║
║ 20SX002XGE must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
wird entpackt …          [***************************************]
wird entpackt …          [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Gerät wird neu gestartet …[***************************************]
Schreibvorgang …         [***************************************]
wird eingeplant …        [***************************************]
wird entpackt …          [***************************************]
Gerät wird neu gestartet …[***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.

```
Then a Reboot.

After POST the Terminal output was:
```
fwupd-efi version 1.4
reset system
```
and after a short black screen
```
Cannot find ME client
```

No difference whether I specified the device ID or not.

kmauleon, 2023-10-18:
@bmn-m please download this 
[Lenovo-ThinkPadX13YogaGen1-ConsumerMEFirmware-225.70.2228.cab.zip](https://github.com/fwupd/firmware-lenovo/files/12972322/Lenovo-ThinkPadX13YogaGen1-ConsumerMEFirmware-225.70.2228.cab.zip)

extract the package and navigate to the location of the cab file
execute as below

fwupdmgr --allow-older --allow-reinstall install [cab filename]

reboot the system
this should install the firmware... wait until completed

petenewcomb, 2024-01-20:
In case this helps anyone else still struggling with this issue: the update worked correctly in the normal way after I upgraded the BIOS using the bootable ISO provided by Lenovo: https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x13-yoga-gen-1/downloads/ds544145-bios-update-utility-bootable-cd-for-windows-10-64-bit-linux-thinkpad-x13-yoga-gen-1-type-20sx-20sy?category=BIOS%2FUEFI

bmn-m, 2024-02-05:
> In case this helps anyone else still struggling with this issue: the update worked correctly in the normal way after I upgraded the BIOS using the bootable ISO provided by Lenovo: [...]

I can Confirm this.

closed 2024-02-05 by bmn-m

kmauleon, 2024-02-06:
thank you very much @petenewcomb and @bmn-m !