fwupd/firmware-lenovo #317: System Firmware Update Fails on Lenovo Thinkpad X1 Yoga 3rd Gen

taaem, 2023-02-14:


hughsie, 2023-02-14:
Can someone poke the ODM please? This has no QA tests on the LVFS and looks like a simple case where the ESRT wasn't updated.

mrhpearson, 2023-02-15:
Flagged to FW team - internal ticket is LO-2278.

JeroenGoddijn, 2023-02-16:
I'm experiencing similar issue, but mine won't install 0.1.48 (even though it downloads/installs/reboots without errors), but next update available is always still 0.1.48 (even though 0.1.49 is already available too)

mrhpearson, 2023-02-16:
FW team are looking to fix this - there were a few issues with the version uploaded that need correcting. For now recommend holding off on the updates.

ChiWei-Chen, 2023-05-16:
@JeroenGoddijn @taaem there is a new version 1.49 uploaded after we reported to FW team, could you let me know if the issue is still there? Thanks!
[ThinkPad X1 Yoga 3rd System Update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN25ET.firmware)

taaem, 2023-05-16:
@ChiWei-Chen the new update worked!
```
fwupdmgr get-devices --show-all-devices
...
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.49
│ │   Minimum Version:    0.0.1
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              fc5552dc-7882-409b-9b98-699361c74fc4
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
...
```

closed 2023-05-16 by taaem