fwupd/firmware-lenovo #318: Can't upgrade a thinkpad t14 gen3 amd

marco44, 2023-02-23:
Hi, can't apply this update:

```
LENOVO 21CF002GMX
│
└─Embedded Controller:
  │   Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
  │   Summary:            UEFI ESRT device
  │   Current version:    0.1.11
  │   Minimum Version:    0.1.11
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               6be552b4-eda3-46a7-ab1b-d8ba1ff451bd
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14 Gen 3 AMD/T16 Gen 1 AMD/P16s Gen 1 AMD:
        New version:      0.1.25
        Remote ID:        lvfs
        Release ID:       17503
        Summary:          Lenovo ThinkPad T14 Gen 3 AMD/T16 Gen 1 AMD/P16s Gen 1 AMD Embedded Controller Firmware
        License:          Proprietary
        Size:             1.2 MB
        Created:          2022-12-26
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Fixed vantage will show a few seconds AC Power when plugging in a USB-C adapter
        • Fixed system do not pop up slow charge when plugging in 30W adapter.[DFT0197186]
        
        Some new functionality has also been added:
        
        • Modify Charge limit table for the requirements of thermal team
```
I get this error : `Update Error:       failed to run update on reboot` from `get-history`

I've already applied about 10 firmwares on this machine, so I suppose it's set-up correctly, but I don't know how to get more debug output.

The machine reboots, I see the firmware update screen, goes from 0 to 100%, reboots, and get-history tells me that it failed

marco44, 2023-02-23:
Oh I forgot to say this: I tried twice, it failed twice, the exact same way

mrhpearson, 2023-02-23:
I've flagged this to the FW team - internal ticket  LO-2290

Just to check my understanding - when it goes from 0 to 100% that's the FW update progress bar? Strange to have it fail after that.

Can you check the FW version from the BIOS setup screen (F1 during early boot). Just to rule out this being an ESRT table update issue....

Thanks
Mark

marco44, 2023-02-24:
Yes, it goes from 0 to 100% on the progress bar. I also see the boot on the firmware updater in the bios log.
The Controller's FW version displays R23HT45W(1.11) in the BIOS

ChiWei-Chen, 2023-02-24:
@marco44 I was working with @mrhpearson and got feedback from FW team.

> LVFS is only available on ECFW after version 1.16. Customers may need to upgrade EC to 1.16 or later under the shell before LVFS can be used normally.

You get the later ECFW from the link below: 
https://support.lenovo.com/jp/en/downloads/ds557681-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-t14-gen-3-type-21cf-21cg-t16-gen-1-type-21ch-21cj-p16s-gen-1type-21ck-21cl

Thanks!

marco44, 2023-02-24:
Ok, So i tried the linux procedure…

```
fwupdmgr install r23ul62w.cab
Decompressing…           [      \                                ]
Specified firmware is already installed '0.1.38
```

ChiWei-Chen, 2023-02-24:
@marco44 , thanks, I'll give you back once I got FW team's feedback for the question above

ChiWei-Chen, 2023-02-24:
hi @marco44 , I got a separate cab package which can update Embedded Controller Firmware only, please drop me a mail (cchen50 [at] lenovo [dot] com) if you want to try it. Thanks.

ChiWei-Chen, 2023-02-27:
@marco44 , as discussed in the mail, it seems the EC Frimwared could be installed successfully, please let us know if there is any further questions, or we can close this. Thanks!

marco44, 2023-02-27:
Hi,
My problem is solved. But I suppose other users will encounter it if everything stays in this status

ChiWei-Chen, 2023-02-27:
@marco44 thanks, we'll ask the FW team to document this troubleshooting instruction in Lenovo support site.

gchudnov, 2023-02-27:
Hello,

it seems that I have a similar issue (?), but on `P16s Gen 1`.

cannot update
```
└─Embedded Controller:
  │   Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
  │   Previous version:   0.1.11
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
```
to
```
0.1.25
```

and

`sudo fwupdate -L` prints:

```
failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
```

If there is a way to solve it, please let me know...

Thank you!

gchudnov, 2023-02-28:
Forgot to mention here..
the problem was solved, thank you!

dahaupt, 2023-04-01:
I experience the same issue when updating the EC from `0.1.11` to `0.1.25` (LENOVO 21CF004NGE).

I already tried to install the cab file from [fwupd.org](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR23HT.firmware) manually without success.

Looking helpfully for any help or further investigation! :pray:

ChiWei-Chen, 2023-04-03:
@dahaupt , I know the FW is working in progress to document the instruction tips. It may take a while.

Please drop me a mail as https://github.com/fwupd/firmware-lenovo/issues/318#issuecomment-1443356092 if you want to give it a try. Thanks!

aderumier, 2023-04-15:
@ChiWei-Chen 

I just send you a mail, I have same problem with my thinkpad p16s gen1

ChiWei-Chen, 2023-04-17:
@aderumier , thanks! I sent you the instruction, please give it a try.

techi602, 2023-07-25:
I just see 0.1.28 update available. Can anyone confirm if it works before I brick my laptop? :smile: 

```
21CJS04U00
│
├─Embedded Controller:
│ │   Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.11
│ │   Minimum Version:    0.1.11
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               6be552b4-eda3-46a7-ab1b-d8ba1ff451bd
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─ThinkPad T14 Gen 3 AMD/T16 Gen 1 AMD/P16s Gen 1 AMD:
│ │     New version:      0.1.28
│ │     Remote ID:        lvfs
│ │     Release ID:       42646
│ │     Summary:          Lenovo ThinkPad T14 Gen 3 AMD/T16 Gen 1 AMD/P16s Gen 1 AMD Embedded Controller Firmware
│ │     License:          Proprietary
│ │     Size:             1,2 MB
│ │     Created:          2023-06-02
│ │     Urgency:          High
│ │     Vendor:           Lenovo
│ │     Release Flags:    • Is upgrade
│ │     Description:      
│ │     Some new functionality has also been added:
│ │     
│ │     • Modify thermal table for the requirements of thermal team.
│ │     • Modify the initialization sequence of expander.
│ │     • Add password protection for MPS2845A.

```

ChiWei-Chen, 2023-07-26:
@techi602 , I saw some user update to 0.1.28 from 0.1.11, please give it a try.
If it doesn't work for you, please send me a mail as this https://github.com/fwupd/firmware-lenovo/issues/318#issuecomment-1443356092 . I can send you the workaround upgrading to 1.24 first.

ChiWei-Chen, 2023-08-25:
Hi,
The tip is published. 
Please follow the instructions to upgrade EC firmware first, then you can upgrade EC firmware via LVFS afterward.
https://support.lenovo.com/us/en/solutions/ht515021-ecfw-is-unable-to-be-upgraded-by-wufu-or-lvfs-thinkpad

ragibkl, 2023-12-23:
> Hi, The tip is published. Please follow the instructions to upgrade EC firmware first, then you can upgrade EC firmware via LVFS afterward. https://support.lenovo.com/us/en/solutions/ht515021-ecfw-is-unable-to-be-upgraded-by-wufu-or-lvfs-thinkpad

Hi, I'm on `Thinkpad T14s Gen 3 AMD`.
I had the same issue of not being able to update the embedded controller.

I can confirm that the instructions on the tip page works!
Thanks for the hard work!