fwupd/firmware-lenovo #319: Firmware Update Error on T14 Gen3 AMD (21CG)

je-nix, 2023-02-25:
Hi there

I encounter issues updating the firmware of my brand new T14 Gen3 AMD (Series 21CG) with Arch Linux. The BIOS verion is 1.17. I tried installing the new firmware according to the guide in the Arch wiki with fwupd (basically using 'fwupdmgr refresh', 'fwupdmgr get-updates' und 'fwupdmgr update'), using the online database as source first. This successfully updated some firmware (e.g. the one of my docking). I'd say it could install all firmware which does not need a reboot to apply.
When trying to update the system firmware, fwupdmgr creates the corresponding files on the EFI partition and rebooting boots into the EFI flash tool. Unfortunately, the update on boot fails with the following error:

> fwupd-efi version 1.4
> WARNING: QueryCapsuleCapabilities failed, assuming EfiResetWarn: Unsupported
> WARNING: Could not apply capsule update: Not Found
> WARNING: Could not apply capsule: Not Found
> Reset System

I tried installing the .cab file with the update manually ('fwupdmgr install r23ul62w.cab'), disabling the signature verification (as it is suggested in other issue threads). But I'm facing the same issue with the update error during the boot flashing.
Next, I tried updating the BIOS using the boot CD iso (extracted with geteltorito and written to a usb drive). The update manager from the usb drive boots, but then fails with a generic "unexpected error" during the upgrade ("An unexpected error has occured, and the process did not complete"). Sadly. there is no more information about what went wrong. Given that it takes a moment before the error occurs and I can see access to the usb drive (blinking LED) I assume the updater did an integrity check (probably checksum verification of the update) and failed then trying to apply it.

I tried all the different BIOS versions available for download (1.29, 1.30, 1.32) but am always facing the same issue. I also tried resetting the BIOS to defaults ("Load defaults" in the "Startup" section and the full reset via the "Security" section). The issue is still the same.
Now it seems I'm stuck as I don't see any further option to try. Since both the update via fwupd nor the boot CD seems to work, I suspect something within my BIOS is stuck / broken (as I don't suspect that 3 different BIOS updates publicly released are all broken). I can't say if it was the way from the beginning or if the attempted update via fwupd caused this situation.
Does someone sees an option how I can update my BIOS from this point?

je-nix, 2023-02-25:
Adding a bit more information. Right now, 'fwupdmgr get-history' shows:

```
├─Embedded Controller:
│ │   Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
│ │   Previous version:   0.1.11
│ │   Update State:       Success
│ │   GUID:               6be552b4-eda3-46a7-ab1b-d8ba1ff451bd
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14 Gen 3 AMD/T16 Gen 1 AMD/P16s Gen 1 AMD:
│       New version:      0.1.25
│       Remote ID:        lvfs
│       Release ID:       17503
│       Summary:          Lenovo ThinkPad T14 Gen 3 AMD/T16 Gen 1 AMD/P16s Gen 1 AMD Embedded Controller Firmware
│       License:          Proprietary
│       Size:             1.2 MB
│       Created:          2022-12-26
│       Urgency:          High
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:      
│       This stable release fixes the following issues:
│       
│       • Fixed vantage will show a few seconds AC Power when plugging in a USB-C adapter
│       • Fixed system do not pop up slow charge when plugging in 30W adapter.[DFT0197186]
│       
│       Some new functionality has also been added:
│       
│       • Modify Charge limit table for the requirements of thermal team
(.....)
└─System Firmware:
  │   Device ID:          d96de5c124b60ed6241ebcb6bb2c839cb5580786
  │   Previous version:   0.1.17
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2023-02-19 10:33
  │   GUID:               e40809b1-7d9f-4dc4-7c0f-8ace63b43687
  │   Device Flags:       • Internal device
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      0.1.36
        License:          Unknown
        Description:      
        The vendor did not supply any release notes.
```

However, 'fwupdmgr get-updates' states that there are no updates for the controller nor system firmware available. I can't say if this has been the case previously (meaning that version 1.32 is not in the database) or if it is confused with version. Also, version 0.1.36 seems strange to me as it is documented nowhere.

je-nix, 2023-02-25:
Another strange thing I just noticed:

'fwupdmgr get-releases d96de5c124b60ed6241ebcb6bb2c839cb5580786' shows:

```
LENOVO 21CGS0TV00
│
├─ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD:
│     New version:        0.1.38
│     Remote ID:          lvfs
│     Release ID:         16804
│     Summary:            Lenovo ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD System Firmware
│     License:            Proprietary
│     Size:               34.6 MB
│     Created:            2021-07-13
│     Urgency:            High
│     Vendor:             Lenovo
│     Release Flags:      • Is upgrade
│     Description:        
│     Problem fixes:
│     
│     • Fixed security issues.
│     • Fixed issue that keyboard 50% backlight will get off under Linux after the system resumes from suspend.
│     • Fixed an issue where WMI change the BIOS display setting not work in setup.
│     Issues:             309662
│                         CVE-2022-34303
│                         CVE-2022-34302
│                         CVE-2022-34301
│   
├─ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD:
│     New version:        0.1.36
│     Remote ID:          lvfs
│     Release ID:         15843
│     Summary:            Lenovo ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD System Firmware
│     License:            Proprietary
│     Size:               34.6 MB
│     Created:            2021-07-13
│     Urgency:            High
│     Vendor:             Lenovo
│     Release Flags:      • Is upgrade
│     Description:        
│     This stable release is only regular update, not fixes Linux issue
│   
└─ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD:
      New version:        0.1.23
      Remote ID:          lvfs
      Release ID:         14536
      Summary:            Lenovo ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD System Firmware
      License:            Proprietary
      Size:               34.6 MB
      Created:            2021-07-13
      Urgency:            High
      Tested by Lenovo:
        Tested:           2022-07-24
        Distribution:     ubuntu 20.04
        Old version:      0.1.22
        Version[fwupd]:   1.7.5
      Vendor:             Lenovo
      Release Flags:      • Is upgrade
      Description:        
      Lenovo ThinkPad T14 Gen 3 T16/P16s Gen 1 AMD System Firmware
      
      This stable release fixes the following issues:
      
      • Fix Lenovo AMD RMB platform DMIC audio Linux support
      • Fix Enable L3 running BiosSetup_ChangeSettings causes machine BSOD
      • Follow TCO Certified Generation 9 for Notebooks standard transition to dropped e-label and logo showing in boot display.
      • Update Phoenix patch to fix Linux system hang after resume with HDD password set with MS.
      • Fixed the issue of e-label not display
      
      Some new functionality has also been added:
      
      • Enable TPM WUFU support
    
```

There are newer versions of the firmware available than on the Lenovo support download page (versions 0.1.36 / 0.1.38) which where released far before the release of the current versions (" Created: 2021-07-13"). Maybe fwupd tried to installed one of those wrong versions (as they are a newer version) and broke something?

mrhpearson, 2023-02-27:
Hi,

1.38 is the latest BIOS version on LVFS for the T14 G3 AMD (and 1.25 for the EC). Checking on the internal FW servers and those are valid releases - I have no idea why they're not up on the Lenovo site though.

I've checked the internal release notes and I can't see any limitations so you should be able to jump directly from 1.17 to 1.25 (though sometimes the bigger jumps make me nervous as they don't get tested as much....)

The fact it can't find the capsule makes me wonder if you have an issue with your EFI partition. Not sure exactly how Arch does it but on Fedora I can see cap files under:
/boot/efi/EFI/fedora/fw
Maybe check that you have enough free space in your EFI partition; and that the file is actually installed?

je-nix, 2023-02-27:
Thanks for checking and getting back to me. The latest available firmware on the Lenovo page is 1.32, that's why I was wondering.

I can still see a .cab file in my boot partition, which is located at /boot/EFI/arch/fw/fwupd-e40809b1-7d9f-4dc4-7c0f-8ace63b43687.cap
I notice the different of the ESP, which is located at /boot/efi/EFI/ in some examples while my structure is /boot/EFI/. Not sure if this is relevant, I followed the setup guide from Arch (which basically is "mount the partition at /boot/ and run the grub installert").

What's also strange from my point of view is that the upgrade failed with my BIOS version still being 1.17, but 'fwupdmgr get-updates' claimes that there is no system firmware update available:
> Devices with no available firmware updates: 
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • UEFI Device Firmware
>  • ELAN0678:00 04F3:3195
>  • Integrated RGB Camera
>  • Prometheus IOTA Config
>  • SKHynix HFS001TEJ9X102N
>  • Unifying Receiver
> Devices with the latest available firmware version:
>  • Embedded Controller
>  • System Firmware
>  • Prometheus
>  • ThinkPad Universal USB-C Dock
>  • UEFI dbx
> No updates available

Furthermore (if this is relevant here), the update process using the Lenovo boot CD approach (via geteltorito and usb drive) also fails to update. That may be a different issue, but may also be an indicator that something is broken with my BIOS.

Smith4545, 2023-02-28:
I can confirm and reproduce this on Fedora 37 Workstation on identical hardware.
```
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.1.13-200.fc37.x86_64
```

je-nix, 2023-03-03:
Thanks for your feedback @Smith4545. How is your /boot/ partition looking? Do you have the update .cabs in /boot/EFI/fedora/ or in /boot/efi/EFI/fedora/ ?

Smith4545, 2023-03-03:
Only `/boot/efi/EFI/fedora/` exists on my installation.

Note that SecureBoot isn't enabled.

```
[root@laptop fedora]# tree
.
├── BOOTIA32.CSV
├── BOOTX64.CSV
├── fw
│   └── fwupd-6be552b4-eda3-46a7-ab1b-d8ba1ff451bd.cap
├── fwupdx64.efi
├── gcdia32.efi
├── gcdx64.efi
├── grub.cfg
├── grubia32.efi
├── grubx64.efi
├── mmia32.efi
├── mmx64.efi
├── shim.efi
├── shimia32.efi
└── shimx64.efi

2 directories, 14 files
```

je-nix, 2023-03-05:
Thanks for the confirmation. Seems like my EFI partition layout may be odd, but not wrong.

I did some more testing with updating via fwupdmgr. I can reproduce the "QueryCapsuleCapabilities failed, assuming EfiResetWarn: Unsupported" error in fwupd-efi with both available version (0.1.36 and 0.1.38). A third version, 0.1.23, is listed in the releases, but can't be installed (Error "System Firmware is not currently updatable: (null)").

I noticed that the update file placed in EFI/arch/fw/ has the same name and size for both release versions (/boot/EFI/arch/fw/fwupd-e40809b1-7d9f-4dc4-7c0f-8ace63b43687.cap in my case with a size of 34561776 bytes). Not sure if this is correct. The releases are listed to have the same size though.

However, I'm out of ideas at this point. With neither fwupd nor the Lenovo standalone Update using the boot CD working, I can't tell if the issue is with my device or the current state of firmware update stability on Lenovos site. Sadly, the Lenovo Premier support refuses to help me because I don't want to do a "quick windows installation so see if the updates works there".

@mrhpearson : Kind request if you have some inside view into this issue and can help. Is this a known issue which is being addressed?

mrhpearson, 2023-03-06:
This ticket has very similar https://github.com/fwupd/firmware-lenovo/issues/301

@kmauleon - do you think this is related? I am wondering in this case whether there might have been something that causes issues when jumping from an older version (1.17) and if we need to update manually to something closer to the final version first?

@Smith4545 - which version are you updating from?

kmauleon, 2023-03-07:
they have the same error though but not sure... only BIOS dev can confirm this.  but I believe it's safer to update gradually than jumping to latest from a very old BIOS version

hughsie, 2023-03-07:
> I believe it's safer to update gradually

If that's true then the requirements on firmware versions need to be set so that the user is forced to update and reboot multiple times. I would really check if this is required, as making the user reboot 15 times to install 15 firmware updates is a really bad experience.

kmauleon, 2023-03-07:
@je-nix @mrhpearson this might help >> https://github.com/fwupd/fwupd/issues/1042

je-nix, 2023-03-08:
Thank you all so much for the help. To my shame I must admit that the initial assumption from @mrhpearson (incorrect EFI partition) was indeed correct. On review I notified 2 issues, which I fixed:
- I created the partition table in 'dos' format
- I forgot the EFI partition and installed everything on my /boot/ partition

After correcting both issues (converting the partition table to GPT and shrinking my root partition, allowing for a seperate /efi partition) and reinstalling my bootloader, I was able to update the BIOS using fwupdmgr without any issues. The upgrade to 0.1.38 went flawless (which updated the BIOS to 1.32 and the Controller to 1.25).

So from my side, this issue is resolved. Another case of "read the fine manual'. Thanks again for your support!

kmauleon, 2023-03-08:
thank you very much @je-nix for the update... closing

closed 2023-03-08 by kmauleon