fwupd/firmware-lenovo #320: Failed to update firmware of ThinkPad L14L15 Gen3 AMD (v0.1.16 -> v0.1.22)

TWeber132, 2023-03-02:
Hello there,

I tried to install the firmware update several time, to no avail. I hope someone can help me.
I don't know much about the procedure of firmware updates, so please bear with me.

My journey began in the snap store (Ubuntu 22.04.2 LTS) where a firmware update poped in. I choose to update, restarted the laptop, saw a loading bar and booted into Ubuntu. Nothing out of the ordinary. BUT after entering the snap store again, there still was this firmware update.

I read on the internet about fwupd, tried standard procedures like "refresh" and "update", still to no avail.

fwupdmgr get-history gets me this:
```
│
├─Prometheus:
│ │   Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
│ │   Previous version:   10.01.3333223
│ │   Update State:       Success
│ │   Last modified:      2023-01-11 13:11
│ │   GUID:               d67c4925-e30f-53e6-abc6-e7af0e4c5884
│ │   Device Flags:       • Updatable
│ │                       • Is in bootloader mode
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─  New version:      10.01.3478575
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
└─System Firmware:
  │   Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
  │   Previous version:   0.1.16
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2023-03-02 10:36
  │   GUID:               7d789c51-d050-4bda-875a-80224c0cb1ed
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad L14L15 Gen3 AMD:
        New version:      0.1.22
        Remote ID:        lvfs
        Release ID:       17625
        Summary:          ThinkPad L14L15 Gen3 AMD System Firmware
        License:          Proprietary
        Size:             34,6 MB
        Created:          2022-12-23
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        This release includes following changes:
        
        • Update E-label version to OneKeyLabel_v6.1.0.25
        • Follow setup layout L1415_AMD_20221107.xlsx modify "UMA Frame buffer size" item
        • Update PI 100B
        • Fix DFT184169 KVM window does not display BIOS interface in static IP

```

efibootmgr -v gets me this:
```
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0002,0001,001A,001B,001C,001D,001E,001F,0020,0021
Boot0001* ubuntu	HD(1,GPT,a8642227-d297-44a5-85a6-ecae7720a8ce,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* Linux-Firmware-Updater	HD(1,GPT,a8642227-d297-44a5-85a6-ecae7720a8ce,0x800,0x100000)/File(\EFI\ubuntu\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Asset Information	FvFile(da465b87-a26f-4c12-b78a-0361428fa026)
Boot0015  Regulatory Information	FvFile(e4a83242-deee-f12e-15ff-0102036cc3ce)
Boot0016  ThinkShield secure wipe	FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
Boot0017  ThinkShield Passwordless Power-On Device Manager	FvFile(08448b41-7f83-49be-82a7-0e84790ab133)
Boot0018  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0019  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot001A* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot001B* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001C* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001D* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001E* PXE BOOT	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001F* LENOVO CLOUD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,ad38ccbbf7edf04d959cf42aa74d3650)/Uri(https://download.lenovo.com/pccbbs/cdeploy/efi/boot.efi)
Boot0020  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35400)
Boot0021  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot0022* Boot Next Boot Option	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,a7ca6d35b2c2684783721826a7404894)
```

fwudpmgr get-updates:
```
Devices with no available firmware updates: 
 • ELAN067F:00 04F3:3209
 • Integrated RGB Camera
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • KBG5AZNT256G LA KIOXIA
 • Prometheus
________________________________________________

Devices that were not updated correctly:

 • System Firmware (0.1.16 → 0.1.22)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]:
```

Something i found interesting is that on the retries of the firmware update the loading bar never fully completes. There were some loading hick ups until 60 % , but it stops after 72 % and blanks.
Linux-Firmware-Updater announces that there is nothing to update and proceeds booting into Ubuntu.
Strange isn't it?

Anyways, i really hope there is a way to fix this behavior. If more information is needed i will happily provide it.
Thanks in advance!

Dinastyh, 2023-03-16:
Hi,
I also have this issue on the same latop
fwupdmgr get-history:
```
│     
└─System Firmware:
  │   Device ID:          d96de5c124b60ed6241ebcb6bb2c839cb5580786
  │   Previous version:   0.1.16
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2023-03-16 14:50
  │   GUID:               7d789c51-d050-4bda-875a-80224c0cb1ed
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad L14L15 Gen3 AMD:
        New version:      0.1.22
        Remote ID:        lvfs
        Release ID:       17625
        Summary:          ThinkPad L14L15 Gen3 AMD System Firmware
        License:          Proprietary
        Size:             34.6 MB
        Created:          2022-12-23
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        This release includes following changes:
        
        • Update E-label version to OneKeyLabel_v6.1.0.25
        • Follow setup layout L1415_AMD_20221107.xlsx modify "UMA Frame buffer size" item
        • Update PI 100B
        • Fix DFT184169 KVM window does not display BIOS interface in static IP
      
```
efibootmgr -v:
```
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,001A,001B,001C,001D,001E,001F,0020,0021,0001
Boot0000* Linux Boot Manager	HD(1,GPT,f3340198-2deb-4d89-aa2e-9e6632270826,0x800,0xff800)/File(\EFI\systemd\systemd-bootx64.efi)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 f8 0f 00 00 00 00 00 98 01 34 f3 eb 2d 89 4d aa 2e 9e 66 32 27 08 26 02 02 / 04 04 46 00 5c 00 45 00 46 00 49 00 5c 00 73 00 79 00 73 00 74 00 65 00 6d 00 64 00 5c 00 73 00 79 00 73 00 74 00 65 00 6d 00 64 00 2d 00 62 00 6f 00 6f 00 74 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0001* Linux-Firmware-Updater	HD(1,GPT,f3340198-2deb-4d89-aa2e-9e6632270826,0x800,0xff800)/File(\EFI\arch\fwupdx64.efi)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 f8 0f 00 00 00 00 00 98 01 34 f3 eb 2d 89 4d aa 2e 9e 66 32 27 08 26 02 02 / 04 04 32 00 5c 00 45 00 46 00 49 00 5c 00 61 00 72 00 63 00 68 00 5c 00 66 00 77 00 75 00 70 00 64 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
      dp: 04 06 14 00 66 8b 1c 72 6c 42 86 4e 8e 99 34 57 c4 6a b0 b9 / 7f ff 04 00
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
      dp: 04 06 14 00 2d 76 6a 12 58 57 ca 4f 85 31 20 1a 7f 57 f8 50 / 7f ff 04 00
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
      dp: 04 06 14 00 a6 d9 d8 a7 b0 6a eb 4a ad 9d 16 3e 59 a7 a3 80 / 7f ff 04 00
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
      dp: 04 06 14 00 5b 61 7e 3f 45 0d 80 4f 88 dc 26 b2 34 95 85 60 / 7f ff 04 00
Boot0014  Asset Information	FvFile(da465b87-a26f-4c12-b78a-0361428fa026)
      dp: 04 06 14 00 87 5b 46 da 6f a2 12 4c b7 8a 03 61 42 8f a0 26 / 7f ff 04 00
Boot0015  Regulatory Information	FvFile(e4a83242-deee-f12e-15ff-0102036cc3ce)
      dp: 04 06 14 00 42 32 a8 e4 ee de 2e f1 15 ff 01 02 03 6c c3 ce / 7f ff 04 00
Boot0016  ThinkShield secure wipe	FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
      dp: 04 06 14 00 d5 a0 93 35 52 bd a0 43 80 8e cb ff 5e ce 24 77 / 7f ff 04 00
Boot0017  ThinkShield Passwordless Power-On Device Manager	FvFile(08448b41-7f83-49be-82a7-0e84790ab133)
      dp: 04 06 14 00 41 8b 44 08 83 7f be 49 82 a7 0e 84 79 0a b1 33 / 7f ff 04 00
Boot0018  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
      dp: 04 06 14 00 f4 e6 6e f4 85 47 a3 43 92 3d 7f 78 6c 3c 84 79 / 7f ff 04 00
Boot0019  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
      dp: 04 06 14 00 60 3f 5d 66 3e ad ad 4c 8e 26 db 46 ee e9 f1 b5 / 7f ff 04 00
Boot001A* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
[... 64 lines cut ...]
  │     This release includes following changes:
  │     
  │     • Update E-label version to OneKeyLabel_v6.1.0.25
  │     • Follow setup layout L1415_AMD_20221107.xlsx modify "UMA Frame buffer size" item
  │     • Update PI 100B
  │     • Fix DFT184169 KVM window does not display BIOS interface in static IP
  │   
  └─ThinkPad L14L15 Gen3 AMD:
        New version:      0.1.17
        Remote ID:        lvfs
        Release ID:       22122
        Summary:          ThinkPad L14L15 Gen3 AMD System Firmware
        License:          Proprietary
        Size:             34.6 MB
        Created:          2023-01-10
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        ThinkPad L14L15 Gen3 AMD System Firmware
        
        This release includes following changes:
        
        • Apply Core Update : Security ID 003-013
```

artificial-intelligence, 2023-04-18:
I also got this problem

artificial-intelligence, 2023-04-18:
if you do need additional information to debug this, please ask.

TWeber132, 2023-04-21:
Hi @artificial-intelligence ,
First of all thank you for the further reporting of the bug. I didn't quite get where the appropriate place for this issue is, but either way.
I read through your issue and a bit through the bugtrackes you linked. The solutions that were mentioned didn't work for me. Neither disabling boot order lock nor disabling secure boot did the trick.

If you can provide me with some troubleshooting information, maybe some websites (?), i will try to further specify my problem.

Kind regards :)

mrhpearson, 2023-05-19:
Hi,
Apologies as I'd missed this one. I've flagged it to the FW team (internal ticket is LO-246).
Please confirm if you're still seeing the issue, and I'll let you know any feedback from the FW team.

Thanks
Mark

TWeber132, 2023-05-20:
Hey Mark,
Yes I'm still seeing this issue, unfortunately.

If you need any further information let me know and I'll try to provide them :)

ChiWei-Chen, 2023-06-05:
Hi,
FW team released the latest BIOS 1.21. (It looks some system display wrong BIOS version, v0.1.16 is same as v0.1.22, same BIOS can’t update, it's an expected behavior.)
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1YET.BIOS.firmware

closed 2023-06-07 by TWeber132