fwupd/firmware-lenovo #322: Thinkpad Yoga L13 Gen 3 and Fedora 37 - Can't update anything on firmware level

fabykr, 2023-03-13:
Hi there,

I tried two ways of update firmware software. With terminal I am running at this error:

[image]

With Gnome Software Center a click on either of this buttons does (seemingly) nothing.

[image]

Not a tech guy, but can provide everything you need for a way to solve this.

Thanks in advance!

hughsie, 2023-03-13:
What distro? what fwupd version?

fabykr, 2023-03-13:
Distro: Fedora 37

$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.12
runtime   org.kernel                    6.1.15-200.fc37.x86_64

fabykr, 2023-04-03:
Okay, I don't know, if you did something. But it now worked within the gnome software center. Thanks a lot!