fwupd/firmware-lenovo #323: ThinkPad L14 flashing embedded controller fails

notsim0, 2023-03-14:
I tried to install EC firmware version 0.1.16/R17HT32W and fwupd reported that the update was installed successfully and required a reboot. I confirmed, the computer booted into fwupd-efi and seemed to normally install the update, then started flashing it onto the EC. After a few seconds it rebooted without any indication of error or failure, but BIOS still shows EC version 0.1.10/R17HT26W and the relevant part of the output of fwupdmgr get-history is

```
└─Embedded Controller:
  │   Device ID:          XXX
  │   Previous version:   0.1.10
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2023-03-14 07:12
  │   GUID:               XXX
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad L14 Gen1 / L15 Gen 1:
        New version:      0.1.16
        Remote ID:        lvfs
        Release ID:       21394
        Summary:          Lenovo ThinkPad L14 Gen1 / L15 Gen 1 Embedded Controller Firmware
        Licence:          Proprietary
        Size:             742.8 kB
        Created:          2022-12-08
        Urgency:          Medium
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:
        Embedded controller Firmware Version 1.16, R17HT32W.

        Optional update for charge problem. 
```

This behaviour is exactly the same with secure boot enabled or disabled.

notsim0, 2023-03-14:
In the meantime, I saw that issue #318 describes a similar problem and tried updating from a local cab file, but the one from [fwupd.org](https://fwupd.org) does not work (I did not really expect it to, because that is the same file `fwupdmgr update` uses, but tried anyway) and the [one I got from the Lenovo website](https://download.lenovo.com/pccbbs/mobiles/r17ul16w.zip) is reported to be unsigned. Namely, running
```
fwupdmgr local-install R17HT32W.cab
```
on the file from the linked zip archive gives the error
```
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer
```

I would rather not install unsigned firmware, so it would be nice to have another way of resolving this.

notsim0, 2023-03-20:
Having read that firmware from the Lenovo website is expected to be unsigned because the signature is done by LVFS, I decided to attempt the update with the cab file from the zip archive linked above after setting `OnlyTrusted=false`. It failed the same way as before.

ChiWei-Chen, 2023-04-11:
@notsim0 , @zyzzyxdonta , thanks for reporting this issue.
I've informed the FW team to look into this. (LO-2293 for internal tracking)

ChiWei-Chen, 2023-05-10:
@notsim0 , @zyzzyxdonta , could you let us know does this issue still exist? 
We check this version has been downloaded more than 9K times.

If the issue still there, please provide the whole `fwupmgr get-devices` output for further debugging. Thanks!

zyzzyxdonta, 2023-05-10:
Do the 9000 downloads contain any indication about whether the installation was successful? Maybe the downloads were done automatically and the users didn't notice the installation failing.

Anyway, for me the problem persists.

The output of `fwupdmgr get-devices` is attached. Unfortunately, the output is in German. Setting `LANG=en` breaks the output of the program. Nevertheless, I attached it as well.

Please note that I also activated automatic upload of error reports when prompted by fwupmgr.

[get-devices.txt](https://github.com/fwupd/firmware-lenovo/files/11441351/get-devices.txt)
[get-devices-en.txt](https://github.com/fwupd/firmware-lenovo/files/11441352/get-devices-en.txt)

ChiWei-Chen, 2023-05-11:
@zyzzyxdonta , thanks for the feedback.
The FW team re-uploaded a new [1.16 ECFW](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR17HT.firmware) with the fix to LVFS server few hours ago, could you give it a try?

Thanks.

zyzzyxdonta, 2023-05-11:
Unfortunately, the problem persists. Please let me know if I can provide any other information.

ChiWei-Chen, 2023-05-11:
@zyzzyxdonta , thanks, I assume the `fwupdmgr refresh` is executed at initial.
Could you also provide output `fwupdmgr get-history` ?
I assume the fail is still:
`│`   Update Error:       failed to run update on reboot

zyzzyxdonta, 2023-05-11:
Sorry, I was not aware that a manual refresh was necessary. I force-refreshed (cached data was from yesterday, it seems) and tried the update again. This time, installation succeeded!

Thank you and sorry again for the confusion!

ChiWei-Chen, 2023-05-12:
@zyzzyxdonta great! Thanks for the feedback. 
I'd wait for a few more days to monitor if there is no further question or not.

notsim0, 2023-05-12:
Sorry for the late reply, @ChiWei-Chen. I retried the update as well and it worked for me now, too. Thank you and the FW team.

ChiWei-Chen, 2023-05-15:
@notsim0 , great! Thanks for the feedback.

closed 2023-05-15 by ChiWei-Chen