fwupd/firmware-lenovo #324: Failed to update firmware of ThinkPad T14s Gen 3 | 21BR00EYMX |  (v0.1.11 --> v0.1.25)

ruihildt, 2023-03-14:
When I then proceed with the update, it fails and nothing happens on reboot.
I have disabled secure boot and boot order lock.

When I do `fwupdmgr update`, I get the following:

```
Devices with no available firmware updates:
 • ELAN0676:00 04F3:3195
 • Integrated RGB Camera
 • Intel Management Engine
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.11 to 0.1.25?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Embedded controller Firmware 1.13,                                           ║
║                                                                              ║
║ Improve thermal solution.                                                    ║
║                                                                              ║
║ 21BR00EYMX must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.
```

Here's the result for `fwupdmgr get-history`:

```
LENOVO 21BR00EYMX
│
├─UEFI dbx:
│ │   Device ID:          362301da643102b9f38477387e2193e57abaa590
│ │   Previous version:   190
│ │   Update State:       Success
│ │   Last modified:      2022-11-29 16:14
│ │   GUID:               14503b3d-73ce-5d06-8137-77c68972a341
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─Secure Boot dbx:
│       New version:      217
│       Remote ID:        lvfs
│       Release ID:       15179
│       Summary:          UEFI Secure Boot Forbidden Signature Database
│       Variant:          x64
│       License:          Proprietary
│       Size:             13.8 kB
│       Created:          2020-07-29
│       Urgency:          High
│       Vendor:           Linux Foundation
│       Duration:         1 second
│       Description:
│       This updates the dbx to the latest release from Microsoft which adds insecure versions of grub and
│
│       Before installing the update, fwupd will check for any affected executables in the ESP and will ref
│
│       Once you have installed this dbx update, any DVD or USB installer images signed with the old signat
│       Issues:           309662
│                         CVE-2022-34303
│                         CVE-2022-34302
│                         CVE-2022-34301
│
├─Embedded Controller:
│ │   Device ID:          ae1abd099407b1d95698d69b7273f7fed5c6f35c
│ │   Previous version:   0.1.11
│ │   Update State:       Failed
│ │   Update Error:       failed to run update on reboot
│ │   Last modified:      2023-03-07 13:21
│ │   GUID:               92e31f5c-e4ba-4169-9130-476edc7d25b1
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.25
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:
│       The vendor did not supply any release notes.
│
├─Embedded Controller:
│ │   Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│ │   Previous version:   0.1.11
│ │   Update State:       Failed
│ │   Update Error:       failed to run update on reboot
│ │   Last modified:      2023-03-14 09:29
│ │   GUID:               92e31f5c-e4ba-4169-9130-476edc7d25b1
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.25
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:
│       The vendor did not supply any release notes.
│
└─Embedded Controller:
  │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
  │   Previous version:   0.1.11
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2023-03-14 10:00
  │   GUID:               92e31f5c-e4ba-4169-9130-476edc7d25b1
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad T14s Gen 3 / X13 Gen 3:
        New version:      0.1.25
        Remote ID:        lvfs
        Release ID:       17774
        Summary:          Lenovo ThinkPad T14s Gen 3 / X13 Gen 3 Embedded controller Firmware
        License:          Proprietary
        Size:             1.2 MB
        Created:          2021-07-13
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:
        Embedded controller Firmware 1.13,

        Improve thermal solution.
```

ruihildt, 2023-03-14:
`fwupdmgr get-devices --show-all-devices`

```
LENOVO 21BR00EYMX
│
├─ELAN0676:00 04F3:3195:
│     Device ID:          e0e5d26dd3fa9940f593886e80a7e1eb8f564836
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              4e8cda57-37ba-5dc6-b114-7610e38714e2 ← HIDRAW\VEN_04F3&DEV_3195
│                         85ef74b5-0975-50ae-9971-0781dc54b48c ← HIDRAW\VEN_04F3&DEV_3195&REV_00
│                         cc456594-6f7e-5d1b-82a6-b4558a813152 ← HIDRAW\VEN_04F3&DEV_3195&MOD_0006
│                         4b808d27-c833-50cc-8177-9a401f5c08fe ← ELANTP\ICTYPE_14
│                         d15e804c-570f-511b-803a-b38a4eddae29 ← ELANTP\ICTYPE_14&MOD_0006
│                         296e64f7-9e08-5717-bff2-651e41b0477e ← ELANTP\ICTYPE_14&MOD_0006&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│
├─Unknown Device:
│     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│
├─12th Gen Intel Core™ i7-1260P:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x00000423
│ │   Vendor:             Intel
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│ │                       ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│ │                       3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│ │   Device Flags:       • Internal device
│ │
│ ├─USB4 host controller:
│ │     Device ID:        80a34c86f50f105cd8cb1d90c649eb6fc24ee399
│ │     Summary:          Unmatched performance for high-speed I/O
│ │     Vendor:           INTEL (THUNDERBOLT:0x8087)
│ │     GUIDs:            5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
│ │                       d651f364-21d8-5cc3-9790-dd019a8958ef ← THUNDERBOLT\VEN_8087&DEV_463E&REV_00
│ │                       e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│ │     Device Flags:     • Internal device
│ │                       • System requires external power source
│ │                       • Signed Payload
│ │
│ └─USB4 host controller:
│       Device ID:        e1ad6a1809a2bf6283f030d75d89c1ff36d43e2e
│       Summary:          Unmatched performance for high-speed I/O
│       Vendor:           INTEL (THUNDERBOLT:0x8087)
│       GUIDs:            5041b15d-6e1c-5681-86b3-bb1d56993357 ← THUNDERBOLT\VEN_8087&DEV_463E
│                         d651f364-21d8-5cc3-9790-dd019a8958ef ← THUNDERBOLT\VEN_8087&DEV_463E&REV_00
│                         e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│       Device Flags:     • Internal device
│                         • System requires external power source
│                         • Signed Payload
│
├─Alder Lake-P Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              eaad9970-8e4d-56da-88ab-41a8c1e2811f ← PCI\VEN_8086&DEV_46A6
│                         ed0b9458-c2f1-54c5-9063-dea8f75b4039 ← PCI\VEN_8086&DEV_46A6&REV_0C
│                         374c1c01-6c69-5955-bc2e-a812cec17fe7 ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22EE
│                         fbb5bf35-4726-5eeb-86cf-b106bba70e10 ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22EE&REV_
0C
│                         c4625510-a985-517c-8800-0ecfc6f68c8f ← PCI\VEN_8086&DEV_46A6&REV_00
│                         5a05cb88-27e2-5e4c-8659-4932845d6c3b ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22EE&REV_
00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
[... 261 lines cut ...]
│     GUID:               86c5d3f5-4e59-481f-a567-194eb62c5014
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
└─UEFI Device Firmware:
      Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
      Summary:            UEFI ESRT device
      Current version:    984775
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               faa47346-ec3e-431c-b653-2c09cd893522
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

────────────────────────────────────────────────
Devices that were not updated correctly:
 • Embedded Controller (0.1.11 → 0.1.25)
 ```

kanschat, 2023-03-19:
Same Problem with my Thinkpad X13 Gen3 and Ubuntu 22.10 with Kernel 5.19.0

pelekhBB, 2023-03-23:
I have the same issue

pelekhBB, 2023-03-23:
@ChiWei-Chen Maybe you can help?

ChiWei-Chen, 2023-03-23:
Thanks for reaching out, I already reported to the FW team.
I'll keep you updated once I here any progress. (LO-2352 for my internal tracking)

pelekhBB, 2023-03-23:
@ruihildt We have the same laptops and I'm wondering, when this update is performed, do you also have the screen blink (go out for a second or less) in the middle of the process when you see the percentages?

I also have encryption enabled and it also happens that when entering a password, it may blink, but not every time

ChiWei-Chen, 2023-03-29:
Hi, 
It seems this is a mistake in the metadata, the correct version should be 1.13 instead of 1.25.
The next version 1.14 will resolve the incorrect version of metadata soon.

And there is no much progress on the error "failed to run update on reboot".
The FW wasn't able to duplicate this issue yet.

ChiWei-Chen, 2023-03-29:
Hi @pelekhBB @ruihildt , 
The FW team just replaced the correct 1.13 EC version, it might help with this issue. Could you give it a try? Thanks!
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3CHT.firmware

kanschat, 2023-03-29:
The corrected version fixed my problem (Thinkpad X13 Gen3 and Ubuntu 22.10 with Kernel 5.19.0). Thank you very much

pelekhBB, 2023-03-29:
@ChiWei-Chen my issue is fixed as well. Thank you

```
21BR00DURA
│
└─Embedded Controller:
  │   Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
  │   Previous version:   0.1.11
  │   Update State:       Success
  │   Last modified:      2023-03-29 15:15
  │   GUID:               92e31f5c-e4ba-4169-9130-476edc7d25b1
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14s Gen 3 / X13 Gen 3:
        New version:      0.1.13
        Remote ID:        lvfs
        Release ID:       27963
        Summary:          Lenovo ThinkPad T14s Gen 3 / X13 Gen 3 Embedded controller Firmware
        License:          Proprietary
        Size:             1,2 MB
        Created:          2022-12-28
        Urgency:          Medium
        Vendor:           Lenovo
        Description:      
        Embedded Controller Version 1.13
        
        Update for Improve thermal solution.

```

ruihildt, 2023-03-29:
Also fixed it for me, I will now close this issue.

Thanks @ChiWei-Chen and Lenovo firmware team. :)

closed 2023-03-29 by ruihildt