fwupd/firmware-lenovo #325: failed to update to 0: authentication signing error

personaingrata, 2023-03-14:
today i updated "Lenovo Embedded Controller" via gnome-firmware on my Thinkpad P14s Gen 3 to 0.1.15. i signed fwupdx64.efi with my secure boot keys and the update went through. i thought business like usual but after rebooting, gnome-firmware reports "failed to update to 0: authentication signing error" and "Current Version 0"

`fwupdmgr get-history`                                                    
LENOVO 21AKCTO1WW
...
│     
└─Embedded Controller:
  │   Gerätekennung:      97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
  │   Vorherige Version:  0.1.14
  │   Aktualisierungsstatus:Fehlgeschlagen
  │   Aktualisierungsfehler:failed to update to 0: authentication signing error
  │   Zuletzt geändert:   2023-03-13 15:03
  │   GUID:               7ddc47a5-a390-4571-a943-2636de3504d2
  │   Device Flags:       • Internes Gerät
  │                       • Aktualisierbar
  │                       • System benötigt externe Stromquelle
  │                       • Unterstützt auf dem entfernten Server
  │                       • Benötigt einen Neustart nach der Installation
  │                       • Das Gerät ist während der Dauer der Aktualisierung nutzbar
  │ 
  └─  Neue Version:     0.1.15
        Remote ID:        lvfs
        Lizenz:           Unbekannt
        Beschreibung:     
        Der Anbieter hat keine Versionshinweise zur Verfügung gestellt.
        
i downloaded the file from the fwupd website and tried to manually reinstall v0.1.14 and 0.1.15 but the output is the following
`fwupdmgr install 66473a15f7c7b9056c765ddbb7ca51af6ec9b30d635832138eb3f2ad79654d11-Lenovo-ThinkPad-T14Gen3-P14sGen3-T16-P16s-SystemFirmware-1.15.cab`
Entpacken …              [ -                                     ]
Not compatible with firmware version 0, requires >= 0.1.0

`fwupdmgr downgrade 97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e             [1]`
Keine Herabstufungen für Embedded Controller: No releases found

`fwupdmgr --version                                                      [2]`
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.13
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.15-hardened1-1-hardened
runtime   org.freedesktop.fwupd         1.8.12
`

`efivar -l | grep fw`                                                                                                                                                     
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-7ddc47a5-a390-4571-a943-2636de3504d2-0
`

the bios reports firmware version for the Embedded Controller as 0.1.14.
it seems like the update grilled my notebook and that i`m stuck on this version.
how can i fix this?

hughsie, 2023-03-14:
@mrhpearson the EC ESRT version seems to be zero!

mrhpearson, 2023-03-15:
Yeah - that's weird. But an authentication error as well.

Raised internal ticket LO-2330 for the FW team. Some questions
 
- Can you confirm what FW version is shown in the BIOS setup screen (F1 during early boot)
- The self-signed is curious. I know it should work (and does for the distro's) so that is making me wonder if there's any issues there. Are you running the latest shim which requires a sbat section added to your signed UEFI executable?
- I've asked on recommendations for getting this back to normal.

Mark

personaingrata, 2023-03-15:
- the bios reports v 0.1.14
- i can't think of any issues there, i've updated the EC one time and the UEFI System Firmware several times following the same process, with no issues. secure boot works as expected.
-  the system is up-to-date, shim -signed 15.6+fedora+2 with --sbat /usr/share/grub/sbat.csv
- 
thanks

DaleBCooper, 2023-03-16:
Same problem here with Lenovo X13 Gen 2i after upgrading system + embedded controller firmware, also got the
"Update Error: failed to update to 0: authentication signing error"

20WL003SGE
│
├─Embedded Controller:
│     Device ID:          e35bc9f8a97b4d6a436962aaa2047da2d2186374
│     **Current version:    0**
│     Vendor:             DMI:LENOVO
│     GUIDs:              e6c0d74d-98a2-489b-920d-f8e8c9418cb3
│                         32b2ee96-472f-527d-9486-35075869ff73 ← UEFI\RES_{E6C0D74D-98A2-489B-920D-F8E8C9418CB3}

BIOS setup screen (F1) shows correct Version:

Product Name:ThinkPad X13 Gen 2i
Manufacturer:LENOVO
Bios Version:N35ET51W (1.51 )
Bios Date:2022-12-07
ME Firmware Version:15.0.41.2142
Embedded Controller Version:**N35HT41W (1.41 )**
Machine Type:20WL003SGE

Trying to manual install the firmware fails:

sudo fwupdmgr install N35HT42W.cab
Decompressing…           [***************************************]
Not compatible with firmware version 0, requires >= 0.1.0

Are there any news regarding the recommendations for getting this back to normal?

mrhpearson, 2023-03-17:
Added to notes to the FW team - no updates as yet.

Just to check - are you using your own secure boot keys as well? Is decure boot enabled?

sch1ag, 2023-03-19:
Very similar problem on my ThinkPad T14 gen3 after `fwupdmgr update`:

~~~
# fwupdmgr get-history 
21AHA001CD
│
├─Embedded Controller:
│ │   Device ID:          ae1abd099407b1d95698d69b7273f7fed5c6f35c
│ │   Previous version:   0.1.4
│ │   Update State:       Failed
│ │   Update Error:       failed to run update on reboot
│ │   Last modified:      2023-03-18 23:01
│ │   GUID:               c6fdd612-fdf3-4bff-bc77-347bcdb0219f
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      0.1.08
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
└─System Firmware:
  │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
  │   Previous version:   0.1.3
  │   Update State:       Failed
  │   Update Error:       failed to run update on reboot
  │   Last modified:      2023-03-18 23:02
  │   GUID:               30fbf24a-e1da-4840-84f6-6d7deb630e1d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      0.1.11
        Remote ID:        lvfs
        License:          Unknown
        Description:      
        The vendor did not supply any release notes.
 # 
~~~

BIOS shows:
~~~
Product Name:ThinkPad T14 Gen 3
Manufacturer:LENOVO
Bios Version:N3MET12W (1.11 )
Bios Date:2023-02-09
ME Firmware Version:16.0.15.1810
Embedded Controller Version:N3MHT07W (1.04 )
Machine Type:21AHA001CD
<snip>
Asset Tag:No Asset Tag
Cpu Type:12th Gen Intel(R) Core(TM) i5-1240P
Cpu Speed:1.700GHz
Installed Memory:40960MB
UUID:916694cc-2e29-11b2-a85c-910c3a6ad3d8
~~~

ChiWei-Chen, 2023-03-28:
Hi @personaingrata @DaleBCooper @sch1ag ,
I work with Mark and got some feedback from FW team.
The FW team suspects upgrading EC FW when the battery level lower than 25% may cause an unexpected error which clears ESRT table. 
Do you remember the battery level when the upgrade EC failed on the first try? thanks.

DaleBCooper, 2023-03-28:
Ho @ChiWei-Chen,

> Do you remember the battery level when the upgrade EC failed on the first try? thanks.

My X13 was connected to the power supply and the battery level was 100%.

personaingrata, 2023-03-29:
Hi @ChiWei-Chen,
thanks for getting back to us. my battery charge start stop thresholds is between 50%-55% and i'm always connected to AC. so while updating, my battery level must be something inbetween there.

today fwupd offered me again to update to 0.1.15 "Lenovo-ThinkPad-T14Gen3-P14sGen3-T16-P16s-SystemFirmware-1.15-3.cab" but the update fails immediately with the following error "Message recipient disconnected from mes
[image]
sage bus without replying"

ChiWei-Chen, 2023-03-29:
@DaleBCooper thanks for sharing the information.

@personaingrata , I haven't seen this issue when doing FW update. Not sure if this is fwupd related.
Could you try firmware with `fwupdmgr update` and `fwupdmgr grade` terminal commands?

Please also provide log if it still failed. Thanks!

personaingrata, 2023-03-29:
it still fails with the same error:

`fwupdmgr refresh --force && fwupdmgr get-updates`                           
Wird aktualisiert lvfs-testing
Herunterladen …          [   -                                   ]
Wird aktualisiert lvfs
Herunterladen …          [***************************************]
Neue Metadaten wurden erfolgreich heruntergeladen: 3 lokale Geräte unterstützt
Geräte mit keinen verfügbaren Firmware-Aktualisierungen: 
 • Fingerprint Sensor
 • Integrated Camera
 • Intel Management Engine
 • SSD 980 PRO 2TB
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Geräte mit der neuesten verfügbaren Firmware-Version:
 • System Firmware
 • Unifying Receiver
LENOVO 21AKCTO1WW
│
└─Embedded Controller:
  │   Gerätekennung:      97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
  │   Zusammenfassung:    UEFI ESRT device
  │   Aktuelle Version:   0
  │   Anbieter:           Lenovo (DMI:LENOVO)
  │   Aktualisierungsstatus:Fehlgeschlagen
  │   Aktualisierungsfehler:failed to update to 0: authentication signing error
  │   GUID:               7ddc47a5-a390-4571-a943-2636de3504d2
  │   Device Flags:       • Internes Gerät
  │                       • Aktualisierbar
  │                       • System benötigt externe Stromquelle
  │                       • Unterstützt auf dem entfernten Server
  │                       • Benötigt einen Neustart nach der Installation
  │                       • Das Gerät ist während der Dauer der Aktualisierung nutzbar
  │ 
  └─ThinkPad T14 Gen 3 / P14s Gen 3 / T16 / P16s:
        Neue Version:     0.1.15
        Remote ID:        lvfs
        Release ID:       27928
        Zusammenfassung:  Lenovo ThinkPad T14 Gen 3 / P14s Gen 3 / T16 / P16s Embedded controller Firmware
        Lizenz:           Proprietär
        Größe:            1,2 MB
        Erstellt:         2023-02-09
        Dringlichkeit:    Mittel
        Anbieter:         Lenovo
        Release Flags:    • Vertrauenswürdige Metadaten
                          • Ist Höherstufung
        Beschreibung:     
        Embedded controller version 1.15. Optional update by battery request.

`sudo fwupdmgr update`                                                                                                                                             
Geräte mit keinen verfügbaren Firmware-Aktualisierungen: 
 • Fingerprint Sensor
 • Integrated Camera
 • Intel Management Engine
 • SSD 980 PRO 2TB
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Embedded Controller von 0 auf 0.1.15 aktualisieren?                          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Embedded controller version 1.15. Optional update by battery request.        ║
║                                                                              ║
║ 21AKCTO1WW muss während der gesamten Dauer der Aktualisierung an eine        ║
║ Stromquelle angeschlossen bleiben, um Schäden zu vermeiden.                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Operation durchführen? [Y|n]: y
Authentifizierung …      [***************************************]
Message recipient disconnected from message bus without replying

`sudo fwupdmgr upgrade `                                                                                                                                               [1]
Geräte mit keinen verfügbaren Firmware-Aktualisierungen: 
 • Fingerprint Sensor
 • Integrated Camera
 • Intel Management Engine
 • SSD 980 PRO 2TB
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Embedded Controller von 0 auf 0.1.15 aktualisieren?                          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Embedded controller version 1.15. Optional update by battery request.        ║
║                                                                              ║
║ 21AKCTO1WW muss während der gesamten Dauer der Aktualisierung an eine        ║
║ Stromquelle angeschlossen bleiben, um Schäden zu vermeiden.                  ║
╚══════════════════════════════════════════════════════════════════════════════╝
Operation durchführen? [Y|n]: y
Authentifizierung …      [***************************************]
Message recipient disconnected from message bus without replying

`sudo fwupdmgr install 8dd439c15818f7cf06a3619969d3c22cb6846a6e53012683dabf25f9b0e12e44-Lenovo-ThinkPad-T14Gen3-P14sGen3-T16-P16s-SystemFirmware-1.15-3.cab`
Authentifizierung …      [   -                                   ]
Message recipient disconnected from message bus without replying

sch1ag, 2023-03-30:
> Hi @personaingrata @DaleBCooper @sch1ag , I work with @mrhpearson and got some feedback from FW team. The FW team suspects upgrading EC FW when the battery level lower than 25% may cause an unexpected error which clears ESRT table. Do you remember the battery level when the upgrade EC failed on the first try? thanks.

Hello!

My T14 G3 was connected to AC power. I don’t remember exactly the battery level, but the probability of a level below 25% is extremely small.

Thank you!

ChiWei-Chen, 2023-03-31:
Richard, thanks.

Hi, 
The firmware team suspected there is an issue in the previous EC firmware on LVFS, they re-upload the EC firmware, and it seems could get pass now. 
Could you give it a try? I can see some users can upgrade from zero version to this 1.15.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3BHT.firmware

Thanks!

sch1ag, 2023-04-09:
Hello!

No changes for my T14 G3 yet.

It looks a little strange that the latest versions of System Firmware and Embedded controller on LVFS do not correspond to one "package" from Lenovo.

LVFS:
1. Embedded controller Firmware **N3MHT10W, version 1.07**
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3MHT.firmware
2. System Firmware **N3MET12W, version 1.11**
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3MET.firmware

Lenovo:
https://download.lenovo.com/pccbbs/mobiles/n3bul09w.txt

  ~~~
  <snip>
  Nuvoton EC chip version
  -----------------------------------
  Package ID     UEFI BIOS (BIOS ID)     ECP    (ECP ID)        Rev.   Issue Date
  -------------  ---------------------   --------------------   ----   ------------
  N3BUL09W       1.11 (N3MET12W)         1.08 (N3MHT11W)        01     2023/02/22
  N3BUL08W       1.10 (N3MET11W)         1.07 (N3MHT10W)        01     2022/12/21
  <snip>
  ~~~

Thanks!

ChiWei-Chen, 2023-04-10:
Hi @sch1ag 
N3M this is a different model T14 gen3, the FW team needs to check with it. I'll give you update if there is any progress.

ChiWei-Chen, 2023-04-12:
@sch1ag , @personaingrata , thanks for the feedback. Can we close this task?

sch1ag, 2023-04-12:
@ChiWei-Chen ,

in my opinion yes.

Thank you very much again!

closed 2023-04-12 by ChiWei-Chen

Lucas-Florin, 2023-08-25:
I have a similar problem on my ThinkPad L14 Gen4 AMD
`fwupdmgr get-devices`:
```
LENOVO 21H5CTO1WW
[...]
│   
├─UEFI Device Firmware:
│     Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               d6fb7af8-7862-42a3-b442-8f990f194941
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Failed
│     Update Error:       failed to update to 0: authentication signing error
│     GUID:               f601b368-963d-43a6-ae78-c4dfff74231d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    532
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               88523ddf-3aac-41f6-b15f-dc7bea434b39
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Minimum Version:    15
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               f8637dbe-5762-0000-873a-910aa71c49e4
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
[...]
Devices that were not updated correctly:
 • UEFI Device Firmware (0 → 0.1.17)
Devices that have been updated successfully:
 • Prometheus (10.01.3478575 → 10.01.3478575)

```

`fwupdmgr get-history`:
```
LENOVO 21H5CTO1WW
│
├─Prometheus:
│ │   Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
│ │   Previous version:   10.01.3478575
│ │   Update State:       Success
│ │   Last modified:      2023-08-25 15:42
│ │   GUID:               659f7e45-8d45-528d-b3c7-0695eed055f6
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
│ │ 
│ └─Prometheus:
│       New version:      10.01.3478575
│       Remote ID:        lvfs
│       Release ID:       10696
│       Summary:          Firmware for the Synaptics Prometheus Fingerprint Reader device
│       License:          Proprietary
│       Size:             425,5 kB
│       Created:          2021-06-09
│       Urgency:          Medium
│       Vendor:           Synaptics
│       Duration:         2 seconds
│       Description:      
│       New features and enhancements:
│       
│       • Fix an authentication issue on FOP project.
│     
└─UEFI Device Firmware:
  │   Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
  │   Previous version:   0
  │   Update State:       Failed
  │   Update Error:       failed to update to 0: authentication signing error
  │   Last modified:      2023-08-25 15:44
  │   GUID:               f601b368-963d-43a6-ae78-c4dfff74231d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad L14 L15 Gen3 AMD:
        New version:      0.1.17
        Remote ID:        lvfs
        Release ID:       27500
        Summary:          Lenovo ThinkPad L14 L15 Gen3 AMD Embedded controller Firmware
        License:          Proprietary
        Size:             1,1 MB
        Created:          2023-03-17
        Urgency:          Critical
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad L14 L15 Gen3 AMD Embedded controller Firmware
        
        • Fix "=", "(", ") "3 keys, actually output error characters on Numpad Keyboard for CR (hrvatski)
        • Add ECRAM[0x7F] bit5 GPIO control for TP4 FW update

```
Which LVFS package would I need to install? 
Here 0.1.17 is the newest stable version: 
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1YHT.firmware
should I just wait until 0.1.18 becomes stable?

Lucas-Florin, 2023-08-25:
I tried installing the following: 
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR25HT.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR25ET.BIOS.firmware
But I get the following messages: 
```
Decompressing…           [ -                                     ]
Specified firmware is older than installed '0.1.03 < 65539'

```
and 
```
Decompressing…           [             /                         ]
Specified firmware is older than installed '0.1.08 < 65544'
```
respectively.

ChiWei-Chen, 2023-08-28:
@Lucas-Florin , thanks for reporting the issue, this needs to be addressed by separate FW team (L14 Gen4 AMD).
Could you create a separate ticket? I'll ask L14 Gen4 AMD team to look at this.