fwupd/firmware-lenovo #329: Battery capacity reported as 0% after firmware upgrade on Lenovp T14s 2nd Gen

mlechner, 2023-04-16:


mlechner, 2023-04-16:
I guess the update itself did not fail. So you mean, I have to report that to Lenovo, because there firmware seems to be broken? Is there some QM within fwupd? I mean is there a way to blacklist firmware upgrades in fwupd if they are known to have a serious bug?

maierd, 2023-04-17:
I think I am facing the same issue. Ran the update this morning (nothing seems to have failed) and now the battery is stuck at 0% while it says "Estimating...". I am running Ubuntu 20.04

oshmoun, 2023-04-17:
Confirming the same issue here, on Archlinux.
I ran the diagnostics uefi tool. That reported a health test failure, suggesting the firmware is buggy (since the battery was just fine before the update)

hughsie, 2023-04-17:
Agreed, I've moved the update back to testing.

oshmoun, 2023-04-17:
> If you don't know, Can you please share upgrade history ?

Sure, [here](https://github.com/fwupd/firmware-lenovo/files/11250016/fwupd_history.txt) is the output of `fwupdmgr get-history`

mlechner, 2023-04-17:
And mine is here: 
[20230417_fwupdmgr_history.txt](https://github.com/fwupd/firmware-lenovo/files/11255415/20230417_fwupdmgr_history.txt)

superm1, 2023-04-18:
So all of those reports show the battery firmware upgraded and system firmware upgraded.  Which one caused the problem?  Can you perhaps downgrade the battery firmware?

@mrhpearson ^

maierd, 2023-04-18:
I am not able to downgrade or install another firmware, as a battery level of at least 25% is required to do that. I am not aware of a way to "force" the downgrade regardless.

mrhpearson, 2023-04-18:
Raised internal ticket as high priority: LO-2416

@hughsie or @superm1 - it looks like there are four battery updates and they're all in stable (15977, 15978, 15980, 15981). I don't have permissions to move them but I'm suspicious of these as they were only released to stable recently. Should we demote these?

superm1, 2023-04-18:
Until we know what's going on I think that's a good idea.

hughsie, 2023-04-18:
> I don't have permissions to move them

I think you do, if you change your vendor group. If not, yell.

> Should we demote these?

I've moved them all to embargo.

oshmoun, 2023-04-18:
> Can you perhaps downgrade the battery firmware?

I could not find a way to downgrade, since `fwupdmgr downgrade <battery-device-id>` simply reports the following:
`No downgrades for UEFI Device Firmware: No releases found`

Though I'm probably just not knowledgeable enough about this stuff, and would appreciate it if someone were to provide a rollback method :sweat_smile:

helios2k6, 2023-04-19:
Also I should add that if I unplug my A/C adapter, my laptop shuts off completely. The battery is entirely useless, making my Lenovo nearly useless as well.

helios2k6, 2023-04-19:
Also to note, the firmware update of my battery *failed*, yet the bug is present, meaning something might be going on with the update process.

maierd, 2023-04-23:
Did someone find a way to downgrade the firmware?

mlechner, 2023-04-24:
Am I right that it is unclear if the battery firmware upgrade to 1.1.9482 or the System Firmware uprade from 0.1.24 to 0.1.26 is causing the error? If the battery firmware is causing the bug, could it be a solution to replace the battery like described in the lenovo hardware maintenance manual (is the firmware stored on the battery-part itself or is it on a controller that stays in the machine (on the board?) even when replacing the battery?

mlechner, 2023-04-24:
> Did someone find a way to downgrade the firmware?

No success yet. Always stopped by battery capacity being reported below 25%.

superm1, 2023-04-24:
> > Did someone find a way to downgrade the firmware?
> 
> No success yet. Always stopped by battery capacity being reported below 25%.

You may be able to use `--force` to override the fwupd prevention of the update, but it won't stop the firmware doing a similar check.

mrhpearson, 2023-04-25:
We are discussing with the FW team but don't have any guidance yet. At the moment I'm assuming it's the battery FW update that failed - the symptoms match what would happen if the system was unplugged mid-update.
As yet I don't have a solution to recover the system - we've asked for it.

drilorin, 2023-04-27:
Looking forward getting a way to downgrade / roll back the firmware. I'll help in any way I can.

mlechner, 2023-04-28:
Lenovo will displace my battery and my board based on an existing support contract. I will see what that means regarding the firmware. Worst case may be that the new battery gets bricked immediately because the broken firmware is flashed to the new battery after replacement. I will share my experience here, if needed. Just for record here the offer (in german) from Lenovo:

"bezüglich Ihrer Supportanfrage werden wir einen Technikereinsatz unter der Nummer 0000000000 veranlassen. Wir gehen davon aus, dass das Firmwareupdate das Systemboard beschädigt hat. Ihr Gerät wird bei Ihnen vor Ort zeitnah instandgesetzt. Sie bekommen vorher noch einen Anruf zur Terminabsprache.
Der Techniker wird folgende Bauteile zur Verfügung haben:  Systemboard und Batterie ..."

translated: "regarding your support request, we will arrange for a technician to call the number 0000000000. We assume that the firmware update has damaged the system board. Your device will be repaired promptly at your site. You will receive a phone call beforehand to make an appointment.
The technician will have the following components available: system board and battery ..."

ieure, 2023-04-29:
Seeing similar problems with my daily driver X13 Gen 2 AMD after updating to 1.26.  The X13 Gen 2 AMD has the same motherboard / uses the same BIOS as the T14s Gen 2 AMD.

My symptoms are:

- Won't charge at all via the main port.
- Still charges through the second port in the dock connector.
- Battery thresholds don't work, it always charges to 100%.

It was working fine prior to 1.26, and there's no physical damage to the port.  My system is out of warranty. :(

cd-dr, 2023-05-02:
I have the same problem with my AMD T14S gen 2 as well. In my case the update was performed from windows. I was using bios ver. 1.26 for over a month without any issues. 
Yesterday windows installed the following two updates,
1. Lenovo Ltd. - Firmware - 257.0.0.9482
2. Lenovo Ltd. - Firmware - 1.0.26.0
Since then both in windows and gentoo the battery is stuck at 0% and shows not to be charging further. 

But it seems to be actually charging, because I am able to use the computer on battery after starting the system on AC. Without AC charger the system refuses to boot showing *0190 critical low-battery error*.

Eagerly waiting for some help on fixing this, since this is my only system.

superm1, 2023-05-02:
Those who are having this issue, if you feel comfortable can you try to open up your chassis and disconnect the battery, wait a minute or two and then reconnect it and see if it helps?

cd-dr, 2023-05-02:
I tried both disabling from bios and later physically disconnecting it. It didn't do anything. I even tried draining the capacitors holding the power down and resetting. None of those worked.

mrhpearson, 2023-05-02:
I've escalated this to the FW team as a critical issue.

Please do call support if you have hit this problem - even if you are out of warranty. It's important they know that people are being affected - I have no way of pulling FW updates from Windows.

Afraid I've not had any guidance on how to recover a system yet :(

cd-dr, 2023-05-02:
Thanks @mrhpearson. Seems like there's already a few folks facing the same issue [here](https://forums.lenovo.com/t5/ThinkPad-T400-T500-and-newer-T-series-Laptops/how-to-uninstall-Lenovo-Ltd-Firmware-257-9482/m-p/5221798?page=1#5970436) at Lenovo forums.

ChiWei-Chen, 2023-05-08:
@ieure could you check if the battery firmware version is also 257.0.0.9482 ?

mrhpearson, 2023-05-10:
As a note - there is a battery FW update coming soon with the fix. We expect it in the next couple of weeks as it's under internal testing right now (and it will be delivered on LVFS of course)
It will be apparently be possible to update to it using fwupdate - will publish details once they are confirmed by the battery team. This should recover any impacted systems and hopefully avoid the need to ship batteries to support for recovery.

helios2k6, 2023-05-10:
> > Am I right that it is unclear if the battery firmware upgrade to 1.1.9482 or the System Firmware uprade from 0.1.24 to 0.1.26 is causing the error? If the battery firmware is causing the bug, could it be a solution to replace the battery like described in the lenovo hardware maintenance manual (is the firmware stored on the battery-part itself or is it on a controller that stays in the machine (on the board?) even when replacing the battery?
> 
> I should have an answer for you in about a week on that. I'm sending my laptop in for repairs. We'll see what gets replaced.

I can confirm that replacing the battery will fix the issue on a Thinkpad X1 Carbon Gen 9. Obviously, not an ideal scenario, as the battery will go out of warranty after 1 year and require you to buy a replacement directly.

ChiWei-Chen, 2023-05-16:
Hi, I can see the FW is in embargo and going through the internal process, it would be moved to stable once it's completed.

WarTP, 2023-05-21:
There is a firmware patch available in this thread, which seems to work fine, for anyone who does not want to wait on the official rollout.

https://forums.lenovo.com/t5/ThinkPad-T400-T500-and-newer-T-series-Laptops/how-to-uninstall-Lenovo-Ltd-Firmware-257-9482/m-p/5221798?page=6#5986468

oshmoun, 2023-05-21:
Thanks for the hint. Assuming that fw is legit, then hopefully this makes Lenovo expedite the process and make the fw available sooner rather than later.

YounesOMK, 2023-05-23:
the exact same problem, i can't downgrade since the battery is below 25%, Any updates on this?

mrhpearson, 2023-05-24:
FW team are preparing the battery FW update for release - it should be available soon. Once that's available the battery operation should return to normal and unblock this.

mrhpearson, 2023-05-30:
Just to note that the FW has been released on LVFS. Fwupd should find it - let me know if it's not showing up.

We're clarifying with the FW team exactly how to install it if it won't go through the regular install. The notes we have say
```
fwpudate -a <Device ID> firmware.cab
fwupdmgr install --force XXX
```

But that was before release so I'd like to confirm it's correct still (I think it will be)
Mark

oshmoun, 2023-05-30:
> Just to note that the FW has been released on LVFS. Fwupd should find it - let me know if it's not showing up.
>
> We're clarifying with the FW team exactly how to install it if it won't go through the [...]

Thanks for the update, but `fwupdmgr` is not having much success finding the updated FW for me. I'm assuming that the firmware needs to be downloaded and installed manually, judging by the commands you provided. In which case could you please confirm, just to be extra extra cautious, that [this](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN35BL_000100C9.firmware) is the firmware that needs to be downloaded and installed?

mrhpearson, 2023-05-30:
For extra extra caution I'm going to recommend holding off until either Chi-Wei or I get the thumbs up from the FW team on updating uncharged batteries.
But yes - that looks like the right FW to me. Does fwupdate not find it if you use the battery device-id?

oshmoun, 2023-05-30:
> But yes - that looks like the right FW to me. Does fwupdate not find it if you use the battery device-id?

Thanks for the confirmation. I went ahead and attempted a manual install, but that didn't work either, here's what I tried, alongside the output:
```
$ fwupdmgr download https://fwupd.org/downloads/b25fbb18e9dab0cc28a811ffc03a44ce68bf41992996c45c598d886a5ce9396a-Lenovo-ThinkPad-T14s-Gen-2i-BatteryFirmware-258.0.0.9482.cab
Downloading…             [**********************************     ]
$ fwupdmgr install --force b25fbb18e9dab0cc28a811ffc03a44ce68bf41992996c45c598d886a5ce9396a-Lenovo-ThinkPad-T14s-Gen-2i-BatteryFirmware-258.0.0.9482.cab
Decompressing…           [ -                                     ]
No supported devices found
```

oshmoun, 2023-05-30:
> Can you let me know what fwupdmgr get-devices shows for the battery device please? Mark

gladly :smile: 

```
├─UEFI Device Firmware:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    16852234
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     Problems:           • System power is too low to perform the update
│     GUID:               8f7a3473-a9a7-4aef-b4e0-256976c6816a
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
```

It is not reported as a battery for some reason, probably because of the bad fw. However, this is definitely the battery device, at least according to `fwupdmgr get-history`:
```
├─Battery:
│ │   Device ID:          f95c9218acd12697af946874bfe4239587209232
│ │   Previous version:   1.0.9478
│ │   Update State:       Success
│ │   Last modified:      2023-04-17 07:49
│ │   GUID:               8f7a3473-a9a7-4aef-b4e0-256976c6816a
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─(null) Update:
│       New version:      1.1.9482
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
```

mrhpearson, 2023-05-30:
OK - your device has a different GUID than the FW posted on LVFS - so the FW is correctly being rejected.

Can you confirm you have a T14s G2 Intel - and if so I'll have to check with the FW team why your GUID isn't being used. I checked my fwupd metadata cache and I can't find that GUID listed so right now I don't think there is FW for it available

Definitely don't recommend forcing any updates until we have it clarified.

maierd, 2023-05-30:
Same for me, also T14s G2 AMD.

ChiWei-Chen, 2023-05-31:
> fwpudate -a <Device ID> firmware.cab
Hi, 
Please use this command to upgrade [FW](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN35BL_000100C9.firmware), this should work even battery level is less than 20%. 
Note. **Do-not-turn-off-your-computer-or-remove-the-AC-adaptor-while-update-is-in-progress.**

Regarding the different GUID, FW team is under checking.

mrhpearson, 2023-05-31:
For @oshmoun and @maierd OK - that's the problem :)
The FW hasn't been released for the AMD platform yet. It will be landing soon. The link above is for the Intel platform.

Mark

YounesOMK, 2023-05-31:
any  guess on the approximate time for the AMD firmware update release?

oshmoun, 2023-05-31:
> The FW hasn't been released for the AMD platform yet. It will be landing soon.

And indeed it has 😄 
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1NBL_000100C9.firmware

Manual installation worked, and the battery is back in working order. Many thanks! So the AMD firmware is confirmed ok, at least on my laptop.

Oh and sorry for the Intel/AMD confusion. It was not brought up in this issue prior, so didn't think it was noteworthy 😅

maierd, 2023-05-31:
@oshmoun which command did you use? with the `fwupdate -a` command it tells me `missing or invalid embedded capsule header`

oshmoun, 2023-05-31:
> @oshmoun which command did you use? with the `fwupdate -a` command it tells me `missing or invalid embedded capsule header`

I don't believe you need any special command, just the usual `fwupdmgr update`. You can refresh beforehand just to make sure `fwupdmgr refresh --force`. I did it manually out of impatience 😅

YounesOMK, 2023-05-31:
for me
`sudo fwupdate -a` gives
(fwupdate:25178): FuPluginUefiCapsule-WARNING 16:47:48.266: missing or invalid embedded capsule header
failed: Invalid ESRT GUID: GUID is not valid format

where `fwupdmgr update` after `fwupdmgr refresh --force` gives **No updatable devices**

mrhpearson, 2023-05-31:
I've checked the FW metainfo and the GUID in there looks correct:
`<firmware type="flashed">8f7a3473-a9a7-4aef-b4e0-256976c6816a</firmware>`

What does your fwupdmgr get-devices show as the battery GUID?
I don't think it will matter - but which version of fwupdmgr are you running?

oshmoun, 2023-05-31:
> gives Device Battery [f95c9218acd12697af946874bfe4239587209232] does not currently allow updates: Cannot install update when system battery is not at least 25%

While I didn't have to deal with such warning, it should be quite simple to ignore it. In fact this was discussed in a [linked issue](https://github.com/fwupd/firmware-lenovo/issues/342)

mrhpearson, 2023-05-31:
Thanks @YounesOMK 

> runtime   org.freedesktop.fwupd         1.7.9

Would you mind trying with the flatpak version to see if this is just an old version issue please? I can't see anything wrong with the GUIDs.
(I am guessing wildly though :( )

YounesOMK, 2023-05-31:
@mrhpearson i tried  but ran into a few issues which I suspect are due to the sandboxing characteristic of Flatpak packages. Consequently, I've followed @oshmoun's suggestion from this [issue](https://github.com/fwupd/firmware-lenovo/issues/342) and set the IgnorePower property to true in the /etc/fwupd/daemon.conf file.

these are steps that worked for me:

setting `IgnorePower=true` to either `/etc/fwupd/daemon.conf` or `/etc/fwupd/fwupd.conf`
```
> sudo systemctl restart fwupd

> fwupdmgr download https://fwupd.org/downloads/68c25e060da764f9608c4dfaea14135f7d930066f9fc1a6c66f44c6b0d42e96e-Lenovo-ThinkPad--T14s-Gen-2-AMD-BatteryFirmware-258.0.0.9482.cab

> fwupdmgr install 68c25e060da764f9608c4dfaea14135f7d930066f9fc1a6c66f44c6b0d42e96e-Lenovo-ThinkPad--T14s-Gen-2-AMD-BatteryFirmware-258.0.0.9482.cab

```

mrhpearson, 2023-05-31:
Awesome - so you're back up and running?

woodypl, 2023-05-31:
@mrhpearson Could you please also confirm whether there will be updated firmware released for other battery models on T14s Gen2i laptops? (https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN35BL_00060100.firmware)

ChiWei-Chen, 2023-06-01:
> Could you please also confirm whether there will be updated firmware released for other battery models on T14s Gen2i laptops? (https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN35BL_00060100.firmware)
@woodypl could you let us know what your `fwupdmgr get-devices` show as the battery GUID?

The one you posted is a different battery supplier, it should work well.

OGPayet, 2023-06-01:
Hi,

I have the same problem on my ThinkPad T15 Gen 1, is there a firmware update compatible with my laptop ?

```bash
├─Battery:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    1.1.8343
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Error:       Cannot install update when system battery is not at least 25%
│     GUID:               c8aa69a5-16dd-4be0-a860-374a6e23c77f
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
```

```bash
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.19.0-43-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

ChiWei-Chen, 2023-06-02:
Hi @OGPayet , this task was used to track issue of T14s and X13 Gen2, the problem on your machine could be different. 
Could you create a new task and provide more detail (Steps to Reproduce, failure screenshot, logs) about the issue? Thanks.

adwk67, 2023-06-04:
@OGPayet I have the same problem on a T15 and have opened an issue here: https://github.com/fwupd/firmware-lenovo/issues/346

OGPayet, 2023-06-05:
@adwk67 I already created an issue here: #344, and today I managed to make it work by setting **IgnorePower** to true in **/etc/fwupd/daemon.conf**.

[29 more replies]