fwupd/firmware-lenovo #330: T490s - Intel Management Engine Version invalid?

Sash2208, 2023-04-17:
In my Fedora Security Dashboard:
[image]

[image]

Anyone have an idea why its telling me to install a new version but according to fwupdmgr, the latest version is already installed..

hughsie, 2023-04-17:
Hmm, unexpected. Can you attach the output of `fwupdmgr get-devices` please?

Sash2208, 2023-04-17:
Yes, no problem:

LENOVO 20NX000EGE
│
├─USB2.0 Hub:
│ │   Gerätekennung:      b6bcdb236926a2452b472d3dad455c414145b52e
│ │   Zusammenfassung:    USB 3.x hub
│ │   Aktuelle Version:   50.33
│ │   Anbieter:           VIA Labs, Inc. (USB:0x17EF)
│ │   Installationsdauer: 15 Sekunden
│ │   GUIDs:              e2c96871-8be3-530e-a796-5652cee64eeb ← USB\VID_17EF&PID_3071
│ │                       dea51bff-550e-54e7-9b88-d05519771a15 ← USB\VID_17EF&PID_3071&REV_5033
│ │                       6f3f32bd-ac98-523e-9128-ef67941fb614 ← USB\VID_17EF&PID_3071&HUB_20
│ │   Device Flags:       • Aktualisierbar
│ │                       • Kryptografische Hash-Verifizierung ist verfügbar
│ │                       • Gerät führt Aktualisierungen in Etappen durch
│ │                       • Das Gerät kann sich nach Fehlern beim Aufspielen wiederherstellen
│ │                       • Unsignierte Nutzdaten
│ │ 
│ └─USB2.0 Hub:
│       Gerätekennung:    63331ea6a67c1085fbdfda1a9b630896d9fc7ec0
│       Zusammenfassung:  USB 3.x hub
│       Aktuelle Version: 50.34
│       Anbieter:         VIA Labs, Inc. (USB:0x17EF)
│       Installationsdauer:15 Sekunden
│       GUIDs:            e2c96871-8be3-530e-a796-5652cee64eeb ← USB\VID_17EF&PID_3071
│                         f6540363-8288-5dce-86dd-be3656cee148 ← USB\VID_17EF&PID_3071&REV_5034
│                         3545c1f3-4d81-59f7-b9d9-d25d5811e4be ← USB\VID_17EF&PID_3071&HUB_10
│       Device Flags:     • Aktualisierbar
│                         • Kryptografische Hash-Verifizierung ist verfügbar
│                         • Gerät führt Aktualisierungen in Etappen durch
│                         • Das Gerät kann sich nach Fehlern beim Aufspielen wiederherstellen
│                         • Unsignierte Nutzdaten
│     
├─USB3.1 Hub:
│ │   Gerätekennung:      da77984c82b59c6fc69516431f467fd9a8d39a7f
│ │   Zusammenfassung:    USB 3.x hub
│ │   Aktuelle Version:   50.33
│ │   Anbieter:           VIA Labs, Inc. (USB:0x17EF)
│ │   Installationsdauer: 15 Sekunden
│ │   GUIDs:              e2c96871-8be3-530e-a796-5652cee64eeb
│ │                       2c08e9d9-1f25-5564-b4db-49867dd2c593 ← USB\VID_17EF&PID_3070
│ │                       804c86fc-4fc2-5426-8600-14a8691f1083 ← USB\VID_17EF&PID_3070&REV_5033
│ │                       6d51c3aa-707f-5d41-812d-5cb5c8a4a25d ← USB\VID_17EF&PID_3070&HUB_0004
│ │   Device Flags:       • Aktualisierbar
│ │                       • Kryptografische Hash-Verifizierung ist verfügbar
│ │                       • Gerät führt Aktualisierungen in Etappen durch
│ │                       • Das Gerät kann sich nach Fehlern beim Aufspielen wiederherstellen
│ │                       • Unsignierte Nutzdaten
│ │ 
│ └─USB3.1 Hub:
│       Gerätekennung:    d68b5d74453efa336214e42d39ec96008a3fed64
│       Zusammenfassung:  USB 3.x hub
│       Aktuelle Version: 50.34
│       Anbieter:         VIA Labs, Inc. (USB:0x17EF)
│       Installationsdauer:15 Sekunden
│       GUIDs:            e2c96871-8be3-530e-a796-5652cee64eeb
│                         2c08e9d9-1f25-5564-b4db-49867dd2c593 ← USB\VID_17EF&PID_3070
│                         bf3a0660-00f1-52fc-9b18-2cfa0cbf4dc7 ← USB\VID_17EF&PID_3070&REV_5034
│                         816fc838-78d7-5deb-85ef-a95935318198 ← USB\VID_17EF&PID_3070&HUB_0010
│       Device Flags:     • Aktualisierbar
│                         • Kryptografische Hash-Verifizierung ist verfügbar
│                         • Gerät führt Aktualisierungen in Etappen durch
│                         • Das Gerät kann sich nach Fehlern beim Aufspielen wiederherstellen
│                         • Unsignierte Nutzdaten
│     
├─Core™ i7-8565U CPU @ 1.80GHz:
│     Gerätekennung:      4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Aktuelle Version:   0x000000f0
│     Anbieter:           Intel
[... 176 lines cut ...]
│     Gerätekennung:      f37fb01122dd62c773f4e84ec89737e059712d59
│     Zusammenfassung:    UEFI ESRT device
│     Aktuelle Version:   26918
│     Minimale Version:   1
│     Anbieter:           DMI:LENOVO
│     Aktualisierungsstatus:Erfolg
│     GUID:               80c2e1e4-0a6a-42d0-b188-5a218e57c8de
│     Device Flags:       • Internes Gerät
│                         • Aktualisierbar
│                         • System benötigt externe Stromquelle
│                         • Benötigt einen Neustart nach der Installation
│                         • Das Gerät ist während der Dauer der Aktualisierung nutzbar
│   
└─WhiskeyLake-U GT2 [UHD Graphics 620]:
      Gerätekennung:      5792b48846ce271fab11c4a545f7a3df0d36e00a
      Aktuelle Version:   02
      Anbieter:           Intel Corporation (PCI:0x8086)
      GUIDs:              52aa8ac7-c837-561f-af32-b7979cd2f6ed ← PCI\VEN_8086&DEV_3EA0
                          6986c21a-0f3f-5538-bcef-fe4e8cda7ab7 ← PCI\VEN_8086&DEV_3EA0&REV_02
                          07616327-80e8-5a8a-9f8d-3cb40661abc4 ← PCI\VEN_8086&DEV_3EA0&SUBSYS_17AA2286
                          b04f3e30-a317-5349-bbdd-c3c25153a5bd ← PCI\VEN_8086&DEV_3EA0&SUBSYS_17AA2286&REV_02
                          1bb467a3-4af9-53a9-ab48-f588ee6e8066 ← PCI\VEN_8086&DEV_3EA0&REV_00
                          781afc31-5405-5966-8e99-9919e869e38c ← PCI\VEN_8086&DEV_3EA0&SUBSYS_17AA2286&REV_00
      Device Flags:       • Internes Gerät
                          • Kryptografische Hash-Verifizierung ist verfügbar

hughsie, 2023-04-17:
@mrhpearson the ME with GUID ef5cdc85-9cf6-469d-9cb7-920b7dd6672b has the last release on the LVFS over 1 year ago. Has this model been forgotten, or is it now EoL?

Sash2208, 2023-04-17:
EOL can not be, then it was apparently forgotten to add to LVFS. What can I do?

hughsie, 2023-04-17:
@Sash2208 @mrhpearson is able to look internally and find out what's happened.

mrhpearson, 2023-04-17:
Following up with the FW team - internal ticket is LO-2412

kmauleon, 2023-04-20:
hi ... it seems ODM missed this... ongoing testing now... will let you know once firmware is put to stable... thank you very much

kmauleon, 2023-04-25:
latest has been put to stable... please check... thank you very much
[Consumer](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRN.firmware)
[Corporate](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRM.firmware)

Sash2208, 2023-04-25:
Many thank, version 192.92.2145 is now installed

kmauleon, 2023-04-25:
thank you very much for the confirmation... will close this issue

closed 2023-04-25 by kmauleon