fwupd/firmware-lenovo #332: Failed ME Firmware on Lenovo P1 Gen4i

Jiyangson, 2023-04-22:
I'm having the same problem of failing to recover ME Firmware as  #226. I was using Windows 10 but there was no available solution so I decided to install Ubuntu, and tried to use `fwupdmgr reinstall` or `fwupdmgr --allow-older --allow-reinstall install n35rk02w.cab`. But none of them worked, and it showed no supported devices found
[image]
It could mean that I need to update the package so I tried `fwupdmgr refresh`, and it showed verification failed
[image]
I'm stuck on this step. What should I do now? Thanks!
Here are some info
```
$ sudo dmidecode | grep SKU
		Corporate SKU
	SKU Number: LENOVO_MT_20Y3_BU_Think_FM_ThinkPad P1 Gen 4i
	SKU Number: Not Specified
```
```
$ fwupdmgr get-devices
20Y3004JUS
│
├─MZVL21T0HCLR-00BL7:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    CL2QGXA7
│     Vendor:             Samsung (NVME:0x144D)
│     GUIDs:              4d7a2791-106b-5e72-9cfb-8ea3d89f5421
│                         310f81b5-6fce-501e-acfb-487d10501e78
│                         60c89aac-f321-515b-b419-3cf02aa9d375
│                         bec63ed7-a95f-54fe-b8cc-8e9fee64ba5a
│                         c0e40d86-e47a-57fe-8ed1-453e6d83a586
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─Prometheus:
│ │   Device ID:          4295296d98b3ba38c72f6baa33d24f03a1d428f6
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3333223
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   GUIDs:              d67c4925-e30f-53e6-abc6-e7af0e4c5884
│ │                       659f7e45-8d45-528d-b3c7-0695eed055f6
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
│ │ 
│ └─Prometheus IOTA Config:
│       Device ID:        ea324f9d78d801d381c77ba6947fb8112bebc3a8
│       Summary:          Fingerprint reader config
│       Current version:  0009
│       Minimum Version:  0009
│       Vendor:           Synaptics (USB:0x06CB)
│       GUIDs:            49a675dd-c0d7-5c08-b19a-78fdfb9ea399
│                         d2e397c9-fe28-5506-bf0d-b5860baa33ba
│       Device Flags:     • Updatable
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    65558
│ │   Minimum Version:    65536
│ │   Vendor:             LENOVO (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              f1918178-1ab6-4d4e-b82c-26e4d0e3869c
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
[... 119 lines cut ...]
│     Summary:            UEFI ESRT device
│     Current version:    4115
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               7c8c8f35-30c9-0032-5e44-5a045511746b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
      Summary:            UEFI ESRT device
      Current version:    10551296
      Minimum Version:    57374
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               09f77c9f-1c5d-4616-bafb-bbb19f557480
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

kmauleon, 2023-04-25:
asking ODM... please wait...

hughsie, 2023-04-25:
Looks like your SSL is failing. Are you using a corporate proxy or behind a firewall perhaps?

Jiyangson, 2023-05-13:
sorry for the late response. It was the proxy issue and I have fixed it. But new problem occurs. When I updated the system firmware using `fwupdmgr reinstall`, It asked me to restart. After I typed yes, the screen went black and fans were still running for more than an hour. I had to manually turn off and turn back on for three times, and the failed to recover problem was still there. 

BTW This issue has wasted me a lot of time, so is there a way to totally reset my computer?

kmauleon, 2023-05-17:
hi @Jiyangson let's close this one and please open another issue for proper tracking.... this maybe BIOS system firmware issue and handled by different ODM owner.... thank you very much

closed 2023-05-17 by kmauleon