fwupd/firmware-lenovo #333: firmware signature missing or not trusted

stevelarocque, 2023-04-23:
Hello.  I am trying to upgrade the firmware on my linux t14s using the r22ul61w.zip file downloaded from https://support.lenovo.com/ca/en/downloads/ds557680-bios-update-utility-bootable-cd-for-windows-10-64-bit-thinkpad-x13-gen-3-type-21cm-21cn-thinkpad-t14s-gen-3-type-21cq-21cr

readme instructions say to execute `fwupdmgr install Lenovo-Thinkpad-X13-T14s-AMD-SystemFirmware-1.31.cab` but the response is:
```
Decompressing…           [      \                                ]
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/daemon.conf ONLY if you are a firmware developer
```

Is the file not corrected built?  How do I resolve this, if I am doing something wrong or making a logical error?

Thanks

hughsie, 2023-04-23:
I don't think you should be installing the unsigned package from Lenovo. Can you use the signed one from the LVFS instead?

stevelarocque, 2023-04-23:
Sorry I am kinda noob on this.  What's LVFS and where is it?  Seems odd that I get notifications from Lenovo to download the updates from their site but that they don't install as expected.  (I mean, why wouldn't Lenovo sign them?)

Anyway, thanks for the tip.

hughsie, 2023-04-23:
`fwupdmgr update` should probably do the right thing.

closed 2023-04-23 by stevelarocque