fwupd/firmware-lenovo #334: Failed ME Firmware on Lenovo Thinkpag P15 Gen 2

arduinotech, 2023-04-23:
I update all before update Fedora 37 to Fedora 38 using GUI utility.
After that and reboot I see
(1) "Reading ME Firmware... Please do not power off! 48% Completed"
(2) "Recovering ME Firmware...Please do not power off! 10% Completed"
(3) "Failed to recover ME Firmware..."

I try #fwupdmgr --allow-reinstall --allow-older install ./N37HT28W.cab
I try #fwupdmgr reinstall
I try fwupdmgr upgrade
- no effect - there is no black screen 3-5 minutes after the reboot - nothing is updated and the error remains.

In BIOS I see ME Firmware version - 15.0.42.2235

I try #fwupdmgr reinstall with all 1-8 - no effect.

In my Fedora 38 no sounds.

Please help.

hughsie, 2023-04-23:
What's the output of `fwupdmgr get-devices` and `fwupdmgr get-history` please

arduinotech, 2023-04-23:
Surprisingly, after updating Fedora from 37 to 38 and after closing the laptop lid, the update went away and the error disappeared and the sound appeared! I'm very glad, because I've been trying to fix the problem all day. I'll keep watching and post later!

kmauleon, 2023-04-28:
hi @arduinotech any additional update?  can we close this one for now?

closed 2023-04-28 by arduinotech