fwupd/firmware-lenovo #336: Lenovo ThinkPad P16s firmware update fails on `invalid checksum`

PyroDevil, 2023-05-07:
Here is what I currently (2023-05-07 20:31:00+00:00) have tried:

```
❯ sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [*******************************        ]
Successfully downloaded new metadata: 5 local devices supported
❯ sudo fwupdmgr get-updates
Devices with no available firmware updates:
 • ELAN0689:00 04F3:320C
 • Integrated Camera
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • MZVL22T0HBLB-00BL7
 • Prometheus
LENOVO 21CKCTO1WW
│
├─Battery:
│ │   Device ID:          11a7fbdd09ce583e58b899660fe10f147fd155a2
│ │   Summary:            UEFI ESRT device
│ │   Current version:    1.0.4
│ │   Minimum Version:    0.0.1
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               5a0bfd1d-e221-4536-93dc-7efb56d2b90d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │
│ └─ThinkPad T16 Gen 1 P16s Gen 1:
│       New version:      1.3.7
│       Remote ID:        lvfs
│       Release ID:       18534
│       Summary:          Lenovo ThinkPad  T16 Gen 1 P16s Gen 1 Battery Firmware
│       License:          Proprietary
│       Size:             123.0 kB
│       Created:          2023-04-14
│       Urgency:          High
│       Tested by Lenovo:
│         Tested:         2023-02-13
│         Distribution:   ubuntu 20.04
│         Old version:    0.1.1
│         Version[fwupd]: 1.3.9
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=R23BL01W
│       Vendor:           Lenovo
│       Release Flags:    • Trusted metadata
│                         • Is upgrade
│       Description:
│       Lenovo-ThinkPad--T16-Gen-1-P16s-Gen-1-BatteryFirmware-259.0.0.7
│
│       Added support for
│
│       ThinkPad T16 Gen 1      (Machine Types: 21CH),ThinkPad T16 Gen 1  LTE1  (Machine Types:  21CJ),ThinkPad P16s Gen 1      (Machine Types:  21CK),ThinkPad P16s Gen 1  LTE1        (Machine Types:  21CL)
│
│       Improved charging performance for the following battery:FRU P/N : 5B10W51872  (ASM P/N : SB10W51973)
│
└─System Firmware:
  │   Device ID:          d96de5c124b60ed6241ebcb6bb2c839cb5580786
  │   Summary:            UEFI ESRT device
  │   Current version:    0.1.38
[... 228 lines cut ...]
Downloading…             [*******************************        ]Fwupd-INFO: 22:28:38.993: download progress: 83%
Downloading…             [********************************       ]Fwupd-INFO: 22:28:39.049: download progress: 84%
Downloading…             [********************************       ]Fwupd-INFO: 22:28:39.102: download progress: 85%
Downloading…             [*********************************      ]Fwupd-INFO: 22:28:39.156: download progress: 86%
Downloading…             [*********************************      ]Fwupd-INFO: 22:28:39.211: download progress: 87%
Downloading…             [*********************************      ]Fwupd-INFO: 22:28:39.266: download progress: 88%
Downloading…             [**********************************     ]Fwupd-INFO: 22:28:39.319: download progress: 89%
Downloading…             [**********************************     ]Fwupd-INFO: 22:28:39.373: download progress: 90%
Downloading…             [***********************************    ]Fwupd-INFO: 22:28:39.427: download progress: 91%
Downloading…             [***********************************    ]Fwupd-INFO: 22:28:39.479: download progress: 92%
Downloading…             [***********************************    ]Fwupd-INFO: 22:28:39.533: download progress: 93%
Downloading…             [************************************   ]Fwupd-INFO: 22:28:39.588: download progress: 94%
Downloading…             [************************************   ]Fwupd-INFO: 22:28:39.641: download progress: 95%
Downloading…             [*************************************  ]Fwupd-INFO: 22:28:39.695: download progress: 96%
Downloading…             [*************************************  ]Fwupd-INFO: 22:28:39.750: download progress: 97%
Downloading…             [*************************************  ]Fwupd-INFO: 22:28:39.804: download progress: 98%
Downloading…             [************************************** ]Fwupd-INFO: 22:28:39.856: download progress: 99%
Downloading…             [************************************** ]Fwupd-INFO: 22:28:39.909: download progress: 100%
(fwupdmgr:354023): Fwupd-DEBUG: 22:28:39.909: Emitting ::status-changed() [idle]
Fwupd-INFO: 22:28:39.909: status-code was 200

checksum invalid, expected c122a2063571f4b700ed7dca11f8e1d76ffd46e5fee46997a3b1ae6e7672a0f9 got 32496bd6343b6077ec308cef1b2a3fdcf212da968a49a66a8f82ede890598401
```

So both updating the battery as well as system firmware fails.

PyroDevil, 2023-05-07:
I downloaded the uefi capsules manually and installed them successfully.

The download filename prefix matched the sha256sum of those, so maybe some server metadata issue?

hughsie, 2023-05-08:
Mea culpa; https://gitlab.com/fwupd/lvfs-website/-/commit/f98ebc4da3801b8a2cf49c159dc6219c789eaecd

Just deploying now.

hughsie, 2023-05-08:
Okay, can you try `fwupdmgr refresh --force` please.

hughsie, 2023-05-08:
fwupdmgr reinstall should do the trick.

PyroDevil, 2023-05-08:
Thanks! `reinstall` worked.

closed 2023-05-08 by PyroDevil

Berghopper, 2023-05-09:
I had this same issue, a day later it seems to be gone however.. odd.