fwupd/firmware-lenovo #338: Failed UEFI ESRT device firmware: P15 Gen 2i

acarlton, 2023-05-09:
`fwupdmgr get-devices` is reporting a failed UEFI ESRT device firmware. No available firmware updates reported. Is this something to be concerned about? Not super fluent with managing the system firmware, but willing to provide more information with some guidance. Thanks!

```
LENOVO 20YQCTO1WW
│
├─Battery:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    1.3.10246
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               a5aea3ab-1b50-4943-b6c9-04c041980f50
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.10
│     Minimum Version:    0.1.10
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               e8dddfb5-175b-4ca5-b77a-ddb98c30f0c0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─GA104GLM [RTX A4000 Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE, PCI:0x8086)
│     GUIDs:              27fac98c-fd14-5cab-91ec-9e8608cd66c3 ← PCI\VEN_10DE&DEV_24B7
│                         44d0f2cd-a579-58e2-a0f1-80008342d4e3 ← PCI\VEN_10DE&DEV_24B7&REV_A1
│                         3b61d91d-d429-5da9-b98c-5463b0d83c8f ← PCI\VEN_10DE&DEV_24B7&SUBSYS_17AA22DD
│                         e1ed92df-18d5-59d0-8622-9fe758f1ed97 ← PCI\VEN_10DE&DEV_24B7&SUBSYS_17AA22DD&REV_A1
│                         c7dac7da-12ca-57be-ba1d-8617ec295689 ← PCI\VEN_8086&DEV_9A01
│                         7305c01c-2139-594e-ace6-16ab469a5d85 ← PCI\VEN_8086&DEV_9A01&REV_05
│                         7fb4d435-3c86-5696-9909-4b1490b280fc ← PCI\VEN_8086&DEV_9A01&SUBSYS_17AA22D8
│                         848f553c-eb2d-5a8d-af50-9469cc7a25b3 ← PCI\VEN_8086&DEV_9A01&SUBSYS_17AA22D8&REV_05
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Integrated Camera:
│     Device ID:          23d940eac81a886d3523a7d731f2b68a2bedb0f7
│     Current version:    61.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Serial Number:      0001
│     GUIDs:              5e9afdd3-1159-5da5-9b47-113b031f5409 ← USB\VID_04F2&PID_B6BE
│                         63fb795d-b37a-5ade-a07e-0d3df89769e3 ← USB\VID_04F2&PID_B6BE&REV_6118
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    240.42.2235
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               666705ce-0624-4943-8738-09829a26c8ee
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
[... 233 lines cut ...]
│   
├─UEFI Device Firmware:
│     Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
│     Summary:            UEFI ESRT device
│     Current version:    24856
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               2bece6a9-392e-48de-9cd1-67dc82dcfead
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Xeon W-11955M CPU @ 2.60GHz:
      Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
      Current version:    0x00000042
      Vendor:             Intel
      GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
                          30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
                          a6bd4ca5-75a6-5796-b564-66b5cab1b11b ← CPUID\PRO_0&FAM_06&MOD_8D
                          d9dd5e77-df17-5bab-b5ec-22827598bfed ← CPUID\PRO_0&FAM_06&MOD_8D&STP_1
      Device Flags:       • Internal device
```

mrhpearson, 2023-05-19:
That's strange....
As far as I can tell - that particular device does not have any FW on LVFS (I scanned the cached metadata for the GUID).
I'm assuming the system is working fine? Do you have Windows dual booted on it, or had Windows installed on it before and done any FW updates from there?

@hughsie - how does fwupd determine the Update state? Is this something I should chase up with the FW team or is it just a benign oddness?

mrhpearson, 2023-05-19:
OK - that makes sense. I am curious what that device is - but it must have been something Windows messed up the update for. We cover most devices from Linux but panels still aren't covered - maybe it was that (though that seems unlikely as panel updates after shipping are exceedingly rare)

I suggest checking from Vantage under Windows and seeing if you can figure out what it is and hopefully completing the update there. If you figure out what the device is let me know :)

For Nvidia issues - suggest trying xorg. The Wayland support with Nvidia is still spotty - in our 22.04 preloads Canonical are still using xorg when the Nvidia driver is used (and Wayland if no Nvidia) because of issues during certification.

For managing FW and dual boot - I don't think I have any recommendations really. We work to make sure the FW updates are available in Linux so I think do whatever works best for your workflow. I personally avoid booting Windows whenever possible - but that's just me :)

Mark

acarlton, 2023-05-22:
@mrhpearson I booted into Windows 11 and installed Vantage, but all system diagnostics checkout and there are no pending system updates. I was unaware of Vantage previously and had been using Lenovo System Update to manage firmware, which also has been reporting no updates. I'll keep an eye when I (rarely ;) boot into Windows and let you know if I see the issue resolve itself on the Linux side.

Thank you so must for the kind and informative reply. I've been trying to narrow down the hardware issues with Nvidia and Wayland, but I think a combination of the 6.2 kernel, recent Nvidia drivers with IBT support and Gnome 44.0 are a bad mix at the moment. I'm looking forward to getting this combination ironed out on my P15, so let me know if there's somewhere else I could help out with testing as you work toward the next Lenovo-certified stack!

Kind regards, Aaron

mrhpearson, 2023-05-23:
Thanks for the update. I'm pretty clueless about how things are done in Windows but I think Vantage and System update all end up doing the same thing - guessing though....Windows world is weird.

Nividia drivers have been going through some instability recently but I think 5.25 (hope I have that right) has fixes in. My understanding is that wayland support is still iffy though :(

Mark

hughsie, 2023-05-24:
> how does fwupd determine the Update state

We're reading out the status from the ESRT table -- so it could have been the last factory preload failed and then fwupd is just reporting that. I'd be included to ignore it if there are no pending updates in either windows or linux.

acarlton, 2023-05-24:
Ok, thanks. Good enough for me at the moment! If I ever figure out if there's a component responsible on the Windows side, I'll update here.