fwupd/firmware-lenovo #339: Missing IME updates (to 12.0.92.2145) for ThinkPad X390 Yoga (Machine types: 20NN, 20NQ)

Kjili, 2023-05-11:
I noticed that the IME on the ThinkPad X390 Yoga is on version 192.81.1753 but there have been three subsequent releases by Lenovo (according to https://download.lenovo.com/pccbbs/mobiles/n2lrl07w.txt). The current version should be 12.0.92.2145.

Sorry if this is not the correct place to post this.

kmauleon, 2023-05-12:
the latest have already been released and posted to LVFS as below... please check.  thank you very much.
[Corporate](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRM.firmware)
[Consumer](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2JRN.firmware)

kmauleon, 2023-05-17:
hi @Kjili can we close this one already?  Please advise.  Thank you very much...

Kjili, 2023-05-17:
Sorry for the delay @kmauleon and thanks for pointing me to the LVFS page. I just checked and you're right about the X390. It seems that it is only the X390 Yoga that is not on it's current version: [Consumer](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2LRN.firmware), [Corporate](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2LRM.firmware)?

kmauleon, 2023-05-18:
hi @Kjili sorry but let me check again with ODM owner... will get back to you soon... thank you very much...

kmauleon, 2023-05-18:
hi @Kjili here are the links for X390 Yoga... please check... thank you very much.
[X390 Yoga Corporate](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2LRG.firmware)
[X390 Yoga Consumer](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2NRG.firmware)

Kjili, 2023-05-18:
Hi @kmauleon. That solved the issue, thank you very much. :)

closed 2023-05-18 by Kjili