fwupd/firmware-lenovo #340: Lenovo Thinkpad z13/z16 bios update is available on lenovo.com but not through lvfs

brittyazel, 2023-05-11:
On my Lenovo Thinkpad z13 I noticed that I had stopped receiving bios around the end of 2022, whereas they were arriving fairly steadily up until then. Looking online I see that my model of Thinkpad Z13 gen 1 has new bios updates as of March 27th, version 0.1.55,  whereas LVFS still has the latest release as 0.1.27. 

Is there a reason the bios updates are not available? I.e. are they being held back for a particular reason? Thanks!

Likewise, when downloading the firmware directly from their site and using their included instructions to install the two .cab files, I am getting this error:
`firmware signature missing or not trusted`

brittyazel, 2023-05-11:
@mrhpearson Any ideas here? Afaik you use this laptop yourself, so maybe you know what's going on.

mrhpearson, 2023-05-12:
There were a few release issues - in particular the last release (1.55) which fixed a critical issue on Windows but caused a regression on Linux - it really messed up suspend and resume for us, so we pulled it from LVFS (and I would avoid it)

That issue has now been fixed and I have been testing 1.56. The FW team wanted to make sure it also fixed the profile and backlight issues that were being seen on Linux - there's still some testing ongoing there (actually I owe the FW team a reply but have been swamped this week)

I'll check in and see when 1.56 (or 1.57) will be released. Afraid there's been just a bunch of churn recently which has been the reason for the delays.

Mark

brittyazel, 2023-05-12:
Oh ok, as long as this is intentional I'm totally fine waiting. I just didn't know. I appreciate you letting me know!

You might want to check with the folks who run the [pcsupport](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-z-series-laptops/thinkpad-z13-type-21d2-21d3/downloads/driver-list/component?name=BIOS%2FUEFI) page though, as they're still hosting the 1.55 (buggy) linux version of the firmware, with versions explicitly for Linux. Fortunately these version don't have signatures included in the cab files so they're unable to install, but it's still somewhat confusing.

mrhpearson, 2023-05-12:
Agreed. I wish they didn't put the Linux versions on the support page at all :( Trying to unwind that from their release process but it was done at the request of a large enterprise customer (quite a few years ago) so it is taking a bit of untangling to stop it.
Thanks for letting me know

brittyazel, 2023-05-12:
@mrhpearson do you have any idea why the camera driver isn't working on my end though? GNOME Software doesn't list any updates past or future for the camera, even though there appears to be 4-5 versions. And when I tried to manually install the .cab is said I had unsupported hardware.

aleasto, 2023-06-01:
1.57 was released May 29th, for Windows only 🥲

mrhpearson, 2023-06-01:
Yeah - that's normal. It always takes the FW team a little bit longer to complete the testing and review process for Linux I'm afraid.
I am checking what the status is with the FW team though - I'm not sure where we are with the fixes and in which version.

brittyazel, 2023-06-12:
Just a quick update, it appears Bios update 1.57 showed up today on LVFS. That said, it looks like we're still 1 version behind on Embedded Controller Update (1.55 vs 1.54 in LVFS).

Likewise, I still cannot get the camera firmware to install for some reason. The versions included on LVFS just says "no supported device found" when trying to manually install the .cab, which seems super odd.

mrhpearson, 2023-06-12:
Ack. I've asked for a status update on the EC firmware.

As a note - the profile stuck in low-power mode issue is not fixed until BIOS 1.58, but I have tested that fix and it is coming.

For the camera - can you let me know the GUID of your camera please (fwupdmgr get-devices)? There are different camera vendors used in the same platform so it might be just that your camera doesn't have an update.

brittyazel, 2023-06-12:
Oh for real? I had no idea that there were multiple camera vendors for the same models.

My GUID's report as:
```
9b693d83-fbd6-5809-9018-9f6720cfb220 ← USB\VID_174F&PID_1813
10f6ceb4-aad1-5d30-b41b-e208faed0182 ← USB\VID_174F&PID_1813&REV_1020

```

mrhpearson, 2023-06-12:
Yep. If you check the Windows site there are four vendors in two pairs:
https://pcsupport.lenovo.com/ca/en/products/laptops-and-netbooks/thinkpad-z-series-laptops/thinkpad-z13-type-21d2-21d3/downloads/driver-list/component?name=Camera%20and%20Card%20Reader

The updates on LVFS are for the Chicony/JYT devices. There aren't any updates I can see for the Realtek/Sunplus devices - I don't know if that's by design or missed. I'll see if I can find out (and I have the Sunplus device on mine if it's any consolation)

Mark

brittyazel, 2023-06-12:
Awesome thanks. This is good info to know.

The weird thing is that the webcam has reported firmware updates in the past, and I remember performing at least one upgrade shortly after buying it, but at some point it just  stopped showing the releases. Within GNOME-Firmware, other devices show the currently installed firmware and past firmwares in a list, and the camera used to as well, but at some point it just stopped showing anything at all.

Thaodan, 2023-08-16:
> Yeah - that's normal. It always takes the FW team a little bit longer to complete the testing and review process for Linux I'm afraid. I am checking what the status is with the FW team though - I'm not sure where we are with the fixes and in which version.

The [changelog](https://download.lenovo.com/pccbbs/mobiles/n3guj14w.txt) says that some Linux specific issues have been fixed would it make sense to expidite the process or release the updates on lvfs-testing?

Thaodan, 2023-08-17:
> Just a quick update, it appears Bios update 1.57 showed up today on LVFS. That said, it looks like we're still 1 version behind on Embedded Controller Update (1.55 vs 1.54 in LVFS).

Where did you found 0.1.57? On the Z13 I only have 0.1.155.

Jaeson813, 2023-08-17:
According to the [release notes of N3GUJ14W](https://download.lenovo.com/pccbbs/mobiles/n3guj14w.txt), firmware updates for BIOS 1.57 were withdrawn.
```
Package ID  UEFI BIOS (BIOS ID)  ECP (ECP ID)      Rev.  Issue Date
----------  -------------------  ---------------   ----  ----------
N3GUJ14W                         1.61 (N3GHT61W)   01    2023/07/27
N3GUJ13W                         1.59 (N3GHT59W)   01    2023/07/12
N3GUJ12W    1.57 (N3GET57W)      1.59 (N3GHT59W)   Withdrawn
N3GUJ11W    1.57 (N3GET57W)      1.57 (N3GHT57W)   Withdrawn
N3GUJ10W    1.55 (N3GET55W)      1.56 (N3GHT56W)   01    2023/03/23
```
As I primarily use Linux and currently run 1.55 unable to downgrade, I have tried looking for those versions even though they apparently have Windows regressions, but they have seemingly been completely removed from LVFS and Lenovo's website.
Some issues people encountered with 1.57 and 1.59 (which doesn't even appear in the docs?) are [here](https://forums.lenovo.com/t5/ThinkPad-Z-series-Laptops/External-USB-C-display-stopped-working-1-57-BIOS-Update-Z16-AMD-Gen-1/m-p/5228342) and [here](https://forums.lenovo.com/t5/ThinkPad-Z-series-Laptops/What%E2%80%99s-the-deal-with-the-current-BIOS-Versions-rolled-back/m-p/5242844).

Thaodan, 2023-08-17:
Ah, I assumed the later Windows  BIOS update still included the update.

I read that the BIOS update fix  the L860 WWAN device lost to be lost.
I noticed that it doesn't power up correctly if the device is started on power, then suspend and then resumes.

cmikovits, 2023-08-28:
Anyone knows why there is only an .exe for Windows an no ISO available for the latest update?

gnunn1, 2023-10-01:
Looks like a new update, 1.63, was released on Lenovo's site on Sept 28th. Will this be available in fwupd at some point or is it a Windows only update?

mrhpearson, 2023-10-02:
Yes, it should be on LVFS - it always takes a little big longer just as they check there are no issues with the fwupd update. 

I know the FW updates have been a bit of a mess recently and am trying to find out what is going on. I know there's a bunch of Linux fixes that _didn't_ make it into this release, but should be in the next one.

Mark

Thaodan, 2023-10-02:
Mark Pearson ***@***.***> writes:

> I know the FW updates have been a bit of a mess recently and am trying
> to find out what is going on. I know there's a bunch of Linux fixes
> that _didn't_ make it into this release, but should be in the next
> one.

What Linux fixes do you mean?

When should the next one be available?

mrhpearson, 2023-10-02:
I have updated my z13 to 63W (1.63). I haven't done the z16 yet as I'm using it, will do that later. So far no issues, albeit with pretty limited use.
Issues I'm tracking:
 - slow wakeup from BIOS 1.55 - I'm not seeing any issues on the Z13, it's waking quickly.
 - trackpoint not working after suspend - all working on my Z13.
 - Unable to configure VRAM - this one is not fixed in this version
 - TSC unstable - this was fixed previously, but I did check that it was still working with this version
I did notice the boot time seems slower from power on to when the screen comes on.

If you're on older than 56W I recommend waiting for our testing to be complete as you can't revert back to previous to 56W. 

Also updated to EC 64W (1.64) - with same results as above. So far looking good

Mark

Thaodan, 2023-10-02:
Did anyone notice that the Intel XMM 7560 doesn't wakup correctly from
suspend or when the system is started in battery mode? (z13)

I read that newer on Linux unreleased BIOS updates had a workaround for
that.
I'm pretty sure it's a firmware rather than a Linux issue.

kvaster, 2023-10-04:
I've tried to upgrade to latest 1.63/1.64 and Z16 was stuck a bit after ECP update. A had to completely reset BIOS in order to make it starting up again, but it seems that now I have BIOS 1.63 and ECP still 1.62... Is there any way to force upgrade ECP via iso ? (have only linux...)

Thaodan, 2023-10-08:
> Yes, it should be on LVFS - it always takes a little big longer just as they check there are no issues with the fwupd update.

What is a  bit in this instance?

Thaodan, 2023-10-08:
> Yes, it should be on LVFS - it always takes a little big longer just as they check there are no issues with the fwupd update.

What is a bit in this instance? I'm looking forward to update because the LTE modem doesn't resume from suspend correctly (doesn't respond timeout when communicating with iosm).

mrhpearson, 2023-10-10:
> What is a bit in this instance? I'm looking forward to update because the LTE modem doesn't resume from suspend correctly (doesn't respond timeout when communicating with iosm).

It's usually a week or too, but with holidays in China last week there's an additional delay.
I did ping them to check what the ETA is.

Mark

Extra note:
Turns out the VRAM fix is in this version too.

gnunn1, 2023-10-16:
Sorry to be a pest but any update on the ETA? I'm having an issue with my z16 and TB4 dock (HP G4) not displaying on external monitors on initial boot making typing the password in for encrypt challenging. I'm hoping this item from the README for 1.63 might fix my issue:

```
Fixed can't ouput dispaly with external monitor by specific docks.
```

mrhpearson, 2023-10-16:
It's on LVFS now, but under embargo. Waiting for the FW team to finish their testing and move it to the stable channel.

gnunn1, 2023-10-18:
I see the update was released today and I went ahead and upgraded. The upgrade process worked fine unfortunately it did not seem to fix my issue (not sending signals to external monitors via TB4 dock on initial boot) but I appreciate the update in the BIOS and the updates from @mrhpearson, so thank you for that.

Thaodan, 2023-10-18:
I was thinking something the same. I'm wondering why for Windows the
update CD is used instead of the UEFI Firmware update capsules that are
used for Linux and also supported by Windows[1]

[1] https://learn.microsoft.com/en-us/windows-hardware/drivers/bringup/windows-uefi-firmware-update-platform

brittyazel, 2023-10-18:
> I've tried to upgrade to latest 1.63/1.64 and Z16 was stuck a bit after ECP update. A had to completely reset BIOS in order to make it starting up again, but it seems that now I have BIOS 1.63 and ECP still 1.62... Is there any way to force upgrade ECP via iso ? (have only linux...)

For science, I tried the same thing. I wanted to see if the bootable firmware update .iso would update the ECP, as I've been stuck on 1.54 for well over a year now. After booting the .iso (via a usb stick) everything seemed to work, but the update process indefinitely stalled at a black screen for many tens of minutes with no progress being made. In this state the power button was completely unresponsive, and the only path forward was to use the bios reset button on the underside of the laptop. After pressing the button, the system lights started flashing, and the first boot said that the bios was restored from a backup. Nevertheless, upon booting up I now see the "Lenovo Embedded Controller" is now at 1.62, so it appears an upgrade did work, though it might have been a failsafe bios restore (rather than the .iso itself) that did it. The update .iso seems to consistently stall out after the first reboot, requiring a hard reset, which is somewhat sketchy. 

So, I'm now at Bios 1.63 and ECP 1.62, which is great and better than ECP 1.54, but the update notes would seem to indicate that ECP is at 1.64, so I'm still out of date. Likewise, fwupd is no longer showing any past releases for the "Lenovo Embedded Controller" as it used to, it just says 'No Releases Available'

Also, I would **not recommend** doing what @kvaster and I did, as it seemed SUPER sketchy.

mrhpearson, 2023-10-18:
@gnunn1 - I assume the display is working once the kernel boots? I'm using the TBT4 dock on my Z16 and it's working (but I haven't noticed the boot piece - I think you're right but don't want to reboot right now to check :))

@brittyazel - I'll have to check. My system is on EC 1.64....but I updated pre the LVFS release (using an internal release) as I was checking that fixes had really been implemented. I won't close this bug until we've confirmed why the EC is not being updated.

gnunn1, 2023-10-18:
@mrhpearson Yes once the kernel boots and I'm at the GDM (gnome login) screen to login the displays work fine. Basically neither the systemd-boot selector nor the Plymouth screen with the encrypt hook which prompts for my password to unlock my disk appear on the external displays, they remain powered off until it gets to GDM. 

Since my laptop is sitting on lower shelf away from my desk it makes dealing with the boot process difficult (along with that annoying "Dock has insufficient power, press escape" message).

brittyazel, 2023-10-18:
@gnunn1 This is somewhat off topic for this thread, but do you have Early KMS enabled for your GPU? I.e. have you added amdgpu to your list of initcpio modules?

vincentbernat, 2023-10-18:
After update, my system also stays at 1.54 for ECP.

gnunn1, 2023-10-18:
@brittyazel I do have amdgpu added to modules and early kms is the default now in Arch Linux so there should not be any need to include the kms hook these days though I did try that as well. You are right this is offtopic here (unless it turns out to be BIOS related), I started a thread on the Arch Linux forum:

https://bbs.archlinux.org/viewtopic.php?id=289648

Thaodan, 2023-10-18:
Mark Pearson ***@***.***> writes:

> @brittyazel - I'll have to check. My system is on EC 1.64....but I
> updated pre the LVFS release (using an internal release) as I was
> checking that fixes had really been implemented. I won't close this
> bug until we've confirmed why the EC is not being updated.

Is it related that related to the Lenovo Embedded Controller not longer
showing as releases available in fwupd?

brittyazel, 2023-10-18:
In the release notes it has all of the ECP versions between 1.54 and the new 1.64 as "Withdrawn due to system might hang up after update ECP". I'm guessing that's why they're all pulled from lvfs @Thaodan. In theory, as long as they don't also pull version 1.64, it should show up on lvfs eventually

gnunn1, 2023-10-18:
I solved my display issue, needed to add "thunderbolt" to the list of modules in mkinitcpio which wasn't needed on Intel.

cmikovits, 2023-10-19:
Hi.

As I'm also on arch (garuda) I'm wondering if you also have audio
issues after going to suspend and wakeup again. Most of the time I do
not have sound from the onboard speakers anymore.

Best wishes, C

gnunn1, 2023-10-19:
@cmikovits I've only had the laptop for a few weeks and it's been docked the entire time, no issues with audio when it suspends though in that scenario. I'm using the pipewire stack for audio.

brittyazel, 2023-10-19:
Fantastic, I grabbed the Embedded Controller 1.64 .cab file from lvfs and installed it from the cli and it worked beautifully, and it's reporting as the correct version in GNOME Firmware. For the first time in (ever), I'm finally on the latest Bios and ECP!

mrhpearson, 2023-10-19:
EC FW 1.64 is on LVFS in the testing channel. It will be released to stable soon - it's just finishing it's testing round internally.
Mark

brittyazel, 2023-10-19:
@mrhpearson You're the man, Mark! Thank you so much for sticking with us all through this and pushing it through. People like you are the reason I stick with Lenovo. Much appreciated.

Thaodan, 2023-10-19:
> Is the update now available for all devices? I can't get it from fwupd.

Just noticed that the reason the updates where not shown was that I hibernated my computer.
After rebooting the updates where visible, I think the devices did not show correctly because they were "unknown device".

closed 2023-10-21 by brittyazel