fwupd/firmware-lenovo #341: ThinkPad neo 14 - failed to update to 117474352: authentication signing error

moosyeen, 2023-05-15:
I execute fwupdmgt update for several times, but never succeed. The Update Error is "failed to update to 117474352: authentication signing error". Here are some details below.

# get-history
```shell
fwupdmgr get-history
```
21DNA000CD
│
├─KBG5AZNV512G LA KIOXIA:
│ │   Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│ │   Previous version:   1107ANLA
│ │   Update State:       Success
│ │   Last modified:      2023-04-27 00:58
│ │   GUID:               e4a0d6e9-5a83-500c-9726-065b8c6c15c3
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─KBG5AZNV256G LA / KBG5AZNT256G LA / KBG5AZNV512G LA / KBG5AZNT512G LA / KBG5AZNV1T02 LA / KBG5AZNT1T02 LA:
│       New version:      1108ANLA
│       Remote ID:        lvfs
│       Release ID:       18995
│       Summary:          KIOXIA BG5 NVMe SSD Firmwre for Lenovo PC
│       License:          Proprietary
│       Size:             1.3 MB
│       Created:          2023-02-06
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       Do NOT turn off your computer or remove the AC adapter while update is in progress.
│       
│       The computer shall be restarted after updating firmware completely. The device may not properly function until you shut down or reboot PC.
│       
│       [Support devices and Firmware version] KBG5AZNV256G LA 1108ANLA KBG5AZNT256G LA 1108ANLA, KBG5AZNV512G LA 1108ANLA KBG5AZNT512G LA 1108ANLA, KBG5AZNV1T02 LA 1108ANLA KBG5AZNT1T02 LA 1108ANLA
│       
│       [Problem fixes]1. Task conflict issue on OS Aging test (Fix BSoD issue)/2. Thermal shutdown issue on SMBus enabled system./3. Risk of SSD dead due to LUT overwrite. /4. Change the SSD mode to Read-only when internal LUT clash occurred /5. Drive no response issue on specific modern standby system.
│       
│       [Support Product Scope]Lenovo ThinkPad, ThinkCentre, ThinkStation, IdeaCentre
│     
├─System Firmware:
│ │   Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│ │   Previous version:   0.1.17
│ │   Update State:       Failed
│ │   Update Error:       failed to run update on reboot
│ │   Last modified:      2023-05-12 13:50
│ │   GUID:               34d84f45-4685-4019-b7e3-dba67b96ef7d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      0.1.37
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     
└─System Firmware:
  │   Device ID:          f95c9218acd12697af946874bfe4239587209232
  │   Previous version:   0.1.17
  │   Update State:       Failed
  │   Update Error:       failed to update to 117474352: authentication signing error
[... 158 lines cut ...]
│   
└─UEFI Device Firmware:
      Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
      Summary:            UEFI ESRT device
      Current version:    0.0.0.1
      Minimum Version:    0.0.0.1
      Vendor:             Lenovo (DMI:LENOVO)
      Update State:       Success
      GUID:               7e5534a3-2069-414b-90b6-3e365d2ccd09
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
Idle…                    [***************************************]
Successfully uploaded 1 report

# Additional questions

- Operating system: ubuntu 22.04.
- rebooting has been tried.
- I'm using an NVME disk.
- Secure boot is disabled.
- Boot Order Lock is turned off.

SlurmMcKenzie, 2023-06-04:
I have a similar issue but me existing fw is older:

System Firmware:
  │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
  │   Previous version:   0.1.9
  │   Update State:       Failed
  │   Update Error:       failed to update to 117474352: authentication signing error
  │   Last modified:      2023-06-04 20:13
  │   GUID:               34d84f45-4685-4019-b7e3-dba67b96ef7d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 10th / X1 Yoga 7th:
        New version:      0.1.37
        Remote ID:        lvfs
        Release ID:       22912
        Summary:          Lenovo ThinkPad X1 Carbon 10th / X1 Yoga 7th System Firmware
        License:          Proprietary
        Size:             22,2 MB
        Created:          2023-03-02
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo System Firmware Version 1.37
        
        New functions or enhancements
        
        • Supported WiFi 6E in Japan
        • Displays TCO logo at startup screen with Lenovo Logo for Windows model.
        
        Problem fixes
        
        • Fixed an issue where abnormal shutdown may be logged when Thunderbolt device is attached or detached while system is in shutdown state.
        • Fixed an issue where LCD screen goes black after unplugging docking station in BIOS Setup.

mrhpearson, 2023-06-12:
Hi @hughsie - I think I need your insight as to what might be needed here as next steps.

There are 12K downloads of this FW with 10 error reports - of which a few are authentication issues. 
Is this potentially related to version of fwupd installed and some authentication checks that are catching some people out? I'm not sure what I can flag to the FW team as an issue here so any suggestions appreciated :)

Reporters - please provide fwupdmgr version.
```shell
fwupdmgr --version
```
Mark

mrhpearson, 2023-06-12:
Oh - well that's confusing. And looks like @moosyeen has similar - his device has the same GUID...urgh

You E14 is trying to use the same GUID as the X1C10 (34d84f45-4685-4019-b7e3-dba67b96ef7d)....that's not right.

Your FW is at the location below (and should be using GUID bcbad27d-b2fd-423b-8d81-6f81208eaa96 )
https://fwupd.org/downloads/d8a464ec62efc89b586849056b12bf7255dc8089169a264ffe36ed40a347304f-R1SET45W.cab

Can you check what is in your ESRT tables by doing

`sudo grep . /sys/firmware/efi/esrt/entries/*/*`

And also confirm what BIOS version is shown in the BIOS setup screen (F1 during early boot).

Then I think I need to find out why everyone is using the X1C10 GUID for everything....surreal.

SlurmMcKenzie, 2023-06-12:
Here you go:

/sys/firmware/efi/esrt/entries/entry0/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry0/fw_class:23192307-d667-4bdf-af1a-6059db171246
/sys/firmware/efi/esrt/entries/entry0/fw_type:2
/sys/firmware/efi/esrt/entries/entry0/fw_version:984775
/sys/firmware/efi/esrt/entries/entry0/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry1/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry1/fw_class:34d84f45-4685-4019-b7e3-dba67b96ef7d
/sys/firmware/efi/esrt/entries/entry1/fw_type:1
/sys/firmware/efi/esrt/entries/entry1/fw_version:65545
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status:5
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version:117474352
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry2/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry2/fw_class:677ee0c5-bedd-4116-ac65-056912e78a79
/sys/firmware/efi/esrt/entries/entry2/fw_type:2
/sys/firmware/efi/esrt/entries/entry2/fw_version:1869177961
/sys/firmware/efi/esrt/entries/entry2/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry3/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry3/fw_class:62ebd2ea-0dbc-4156-9018-795c6eded30d
/sys/firmware/efi/esrt/entries/entry3/fw_type:2
/sys/firmware/efi/esrt/entries/entry3/fw_version:16777220
/sys/firmware/efi/esrt/entries/entry3/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry3/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry3/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry4/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry4/fw_class:7e5534a3-2069-414b-90b6-3e365d2ccd09
/sys/firmware/efi/esrt/entries/entry4/fw_type:2
/sys/firmware/efi/esrt/entries/entry4/fw_version:1
/sys/firmware/efi/esrt/entries/entry4/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry4/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry4/lowest_supported_fw_version:1

mrhpearson, 2023-06-12:
Just to double check - E14 G4 Intel or AMD?

SlurmMcKenzie, 2023-06-12:
Intel.

mrhpearson, 2023-06-12:
And just to confirm - E14 G4 Intel or AMD?

mrhpearson, 2023-06-12:
Internal ticket created: LO-2507

@moosyeen - we'll need to do the same for you if you can confirm your ESRT table details.

_Note - for anybody else seeing this issue: Unless you are E14/E16 G4 Intel or Neo 14 - please create a separate ticket._

moosyeen, 2023-06-13:
@mrhpearson 

Here are my `ESRT table detail` and `fwupdmgr version`.

```
ESRT table detail
/sys/firmware/efi/esrt/entries/entry0/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry0/fw_class:23192307-d667-4bdf-af1a-6059db171246
/sys/firmware/efi/esrt/entries/entry0/fw_type:2
/sys/firmware/efi/esrt/entries/entry0/fw_version:984775
/sys/firmware/efi/esrt/entries/entry0/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry1/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry1/fw_class:54cb734c-975c-4a74-9556-57ae6d9a2229
/sys/firmware/efi/esrt/entries/entry1/fw_type:2
/sys/firmware/efi/esrt/entries/entry1/fw_version:196608
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version:57374
/sys/firmware/efi/esrt/entries/entry2/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry2/fw_class:88523ddf-3aac-41f6-b15f-dc7bea434b39
/sys/firmware/efi/esrt/entries/entry2/fw_type:2
/sys/firmware/efi/esrt/entries/entry2/fw_version:532
/sys/firmware/efi/esrt/entries/entry2/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry3/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry3/fw_class:34d84f45-4685-4019-b7e3-dba67b96ef7d
/sys/firmware/efi/esrt/entries/entry3/fw_type:1
/sys/firmware/efi/esrt/entries/entry3/fw_version:65553
/sys/firmware/efi/esrt/entries/entry3/last_attempt_status:5
/sys/firmware/efi/esrt/entries/entry3/last_attempt_version:117474352
/sys/firmware/efi/esrt/entries/entry3/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry4/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry4/fw_class:677ee0c5-bedd-4116-ac65-056912e78a79
/sys/firmware/efi/esrt/entries/entry4/fw_type:2
/sys/firmware/efi/esrt/entries/entry4/fw_version:1869177961
/sys/firmware/efi/esrt/entries/entry4/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry4/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry4/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry5/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry5/fw_class:d43d6b1a-26c8-41a2-b7f4-d2b1ccffce2e
/sys/firmware/efi/esrt/entries/entry5/fw_type:2
/sys/firmware/efi/esrt/entries/entry5/fw_version:16777221
/sys/firmware/efi/esrt/entries/entry5/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry5/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry5/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry6/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry6/fw_class:7e5534a3-2069-414b-90b6-3e365d2ccd09
/sys/firmware/efi/esrt/entries/entry6/fw_type:2
/sys/firmware/efi/esrt/entries/entry6/fw_version:1
/sys/firmware/efi/esrt/entries/entry6/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry6/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry6/lowest_supported_fw_version:1
```

```
fwupdmgr version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.19.0-43-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

mrhpearson, 2023-06-16:
Thanks.
I confirmed that this has already been fixed in R1SET40W (which I think is 1.11). Afraid that I think you'll have to manually update to that first - suggest using the bootable CD method on the support site with a USB stick?

Mark

moosyeen, 2023-06-18:
> 

Hi Mark, I'm pretty noob about Linux, would you mind giving me a link about how to do this?

SlurmMcKenzie, 2023-06-18:
From what I understand, Mark is proposing to download: https://fwupd.org/downloads/d8a464ec62efc89b586849056b12bf7255dc8089169a264ffe36ed40a347304f-R1SET45W.cab and write it on a USB-Stick. You will probably have to rename it or you will have to unpack the CAB-Archive first and write it on the USB stick. I'm not sure about that. 

@Mark: There is another way of installing firmware: 

`sudo fwupdmgr install R1SET45W.cab
`
This won't work because of the bug, right?

mrhpearson, 2023-06-19:
Not quite :) I was suggesting going to the Lenovo support site and to the manual downloads and there is an iso file there.

For the E14 for example there is a bootable ISO here:
https://pcsupport.lenovo.com/ca/en/products/laptops-and-netbooks/thinkpad-edge-laptops/thinkpad-e14-gen-4-type-21e3-21e4/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C

I think you should be able to download that and write it to a USB stick with you preferred SW (I personally tend to use Fedora Media Writer).

Only caveat is I'm not sure what a Neo 14 actually is - it's not in the linux program and I couldn't find it on the Lenovo support site. The 'fixed in R1SET40W' comment above is specifically for the E14. So see if you can find your system on the Lenovo support site and get it running the latest somehow (may involve Windows I'm afraid).  If you're already running the latest on your Neo14 I'll have to flag it to them and ask them to fix. As this seems wrong I think they should address it - but I have to add the caveat that as it's not a Linux certified platform I don't have any levers to pull - I can only ask nicely.

Hope that helps - and I'm definitely suffering Monday morning brain so if anything is confusing let me know
Mark

kghose, 2024-02-09:
Sorry to necro this thread, but I have the same error (same machine, gen, processor (i5) etc.)

```
─System Firmware:
  │   Device ID:          
  │   Previous version:   0.1.9
  │   Update State:       Failed
  │   Update Error:       failed to update to 117474352: error-auth-error
  │   Last modified:      2024-02-09 16:03

```

```
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.12
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.6.0-14-generic
runtime   org.freedesktop.fwupd         1.9.12
```

```
sudo grep . /sys/firmware/efi/esrt/entries/*/*

/sys/firmware/efi/esrt/entries/entry0/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry0/fw_class:23192307-d667-4bdf-af1a-6059db171246
/sys/firmware/efi/esrt/entries/entry0/fw_type:2
/sys/firmware/efi/esrt/entries/entry0/fw_version:984775
/sys/firmware/efi/esrt/entries/entry0/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry1/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry1/fw_class:34d84f45-4685-4019-b7e3-dba67b96ef7d
/sys/firmware/efi/esrt/entries/entry1/fw_type:1
/sys/firmware/efi/esrt/entries/entry1/fw_version:65545
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status:5
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version:117474352
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry2/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry2/fw_class:677ee0c5-bedd-4116-ac65-056912e78a79
/sys/firmware/efi/esrt/entries/entry2/fw_type:2
/sys/firmware/efi/esrt/entries/entry2/fw_version:1869177961
/sys/firmware/efi/esrt/entries/entry2/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry3/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry3/fw_class:e36e4405-4a1c-4733-8a3a-40dd2011733a
/sys/firmware/efi/esrt/entries/entry3/fw_type:2
/sys/firmware/efi/esrt/entries/entry3/fw_version:16785731
/sys/firmware/efi/esrt/entries/entry3/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry3/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry3/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry4/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry4/fw_class:7e5534a3-2069-414b-90b6-3e365d2ccd09
/sys/firmware/efi/esrt/entries/entry4/fw_type:2
/sys/firmware/efi/esrt/entries/entry4/fw_version:1
/sys/firmware/efi/esrt/entries/entry4/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry4/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry4/lowest_supported_fw_version:1
```

mrhpearson, 2024-02-12:
> Ok, so is this an issue on Lenovo's part, and updating to a new version of `fwupd` is _not_ expected to resolve it and we should manually install the firmware update from that link?**
> 

Yes. You need to update to R1SET40W (either via USB stick using the iso image, or from Windows if that is an option)