fwupd/firmware-lenovo #343: Thinkpad T14s gen3 (amd) - Missing firmwares

dal00, 2023-05-30:
* Bios is not at the latest 1.31 but at 0.1.30
* NVMe Firmware for MZVL21T0HCLR-00BL7 is over a year old and latest update from this month is missing.

mrhpearson, 2023-06-12:
Agreed - not good :(
Internal ticket LO-2505 created. Will check what is happening with the FW team.

dal00, 2023-06-12:
There was an nvme update yesterday btw.

dal00, 2023-06-28:
Does that also include the update for the Quectel WWAN which seems to be supported by fwupd/lvfs but the latest update is not there for the T14s g3 (amd).

EC 1.25 update applied correctly on my end btw.

closed 2023-07-18 by mrhpearson

dal00, 2023-07-20:
@mrhpearson I'm afraid something might be wrong. As EC 1.27 does not seem to be released even for windows yet. Is this supposed to be like this? Also no mention of 1.31.

```
$ fwupdmgr update
Devices with no available firmware updates: 
 • ELAN0678:00 04F3:3195
 • EM05-G
 • Fingerprint Sensor
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.25 to 0.1.27?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Modify the initialization sequence of expander.                            ║
║ • Add password protection for MPS2845A.                                      ║
║                                                                              ║
║ 21CQCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
n
```

ChiWei-Chen, 2023-07-20:
@dal00 , could you try with `fwupdmgr refresh`, we saw hundreds of downloads of 1.31 BIOS.

dal00, 2023-07-20:
@ChiWei-Chen I actually did `fwupdmgr refresh --force` before.

The thing was I didn't do `fwupdmgr get-updates` because that would have showed that 1.31 was there.

I was just worried that 1.27 went out as a mistake because I could not see it on lenovo.com even.

ChiWei-Chen, 2023-07-21:
@dal00 , just confirmed with FW team. 1.27 is an official EC version.
The Windows process may take more time, it would go to Lenovo Support site a few days later.