fwupd/firmware-lenovo #345: missing hwid for p14s bios update ( 7d75287704a454afef9808855fd78cc57d625515c3979fa6c46800490c00a7e7 )

prometheanfire, 2023-06-03:


prometheanfire, 2023-06-13:
suppose there's nothing else needed from me on this for now?

mrhpearson, 2023-06-22:
Hi @prometheanfire - confirmed there is something wrong with your system. The HW ID is incorrect and unexpected.

The FW team are investigating what might have happened, but it's a holiday in China this week (dragon boat festival) so I won't have an update until next week. I'm guessing they will need your serial number to investigate what happened - would you mind emailing that to me (don't post it publicly). send to markpearson at lenovo dot com.

prometheanfire, 2023-06-22:
sent, I'm not in a huge rush and can update manually if I really need to.  Would be nice to get this fixed for anyone else encountering it though :D

ChiWei-Chen, 2023-07-25:
@prometheanfire would you mind emailing your serial number to me as well ( cchen50 at lenovo dot com )? Since Mark is on leave this week.

ChiWei-Chen, 2023-08-08:
@prometheanfire , thank you for your patience.
As FW team confirmed internally, they cannot find anything suspicious which cause HW ID mismatch.
We will keep monitoring this and see if there are other users who reproduce the same problem.