fwupd/firmware-lenovo #346: Cannot upgrade battery firmware (Thinkpad T15

adwk67, 2023-06-04:
This is a very similar issue to https://github.com/fwupd/firmware-lenovo/issues/342, but the fix described there has not worked for me.

<details>
<summary>fwupdmgr get-devices</summary>
20W4CTO1WW
│
├─Battery:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Transient failure
│     Update Error:       failed to update to 0: AC power required
│     GUID:               6d8e9931-55aa-4ff3-bb93-a7cd1a1bc5d7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
...
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.52
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Transient failure
│ │   Update Error:       failed to update to 0: battery level is too low
│ │   GUIDs:              973a10fa-79c0-453e-b78c-27b55cc0e6eb
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
</details>

<details>
<summary>fwupdmgr get-history</summary>
─System Firmware:
...
│     
└─Battery:
  │   Device ID:          f95c9218acd12697af946874bfe4239587209232
  │   Previous version:   0
  │   Update State:       Transient failure
  │   Update Error:       failed to update to 0: AC power required
  │   Last modified:      2023-06-01 15:41
  │   GUID:               6d8e9931-55aa-4ff3-bb93-a7cd1a1bc5d7
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T15 Gen 2 P15s Gen 2:
        New version:      1.3.2
        Remote ID:        lvfs
        Release ID:       18734
        Summary:          Lenovo ThinkPad T15 Gen 2 P15s Gen 2 Battery Firmware
        License:          Proprietary
        Size:             136,0 kB
        Created:          2023-05-10
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N34BL04W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
[... 86 lines cut ...]
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.

An update requires a reboot to complete. Restart now? [y|N]: y
```

On reboot, the update is not updated as the power supply is not detected, and the updates are re-presented for download:
[image]

adwk67, 2023-06-05:
Having tried this every day for the past week without success, it has suddenly worked this morning. Just the battery update was done:

```
08:20 $ fwupdmgr get-devices
20W4CTO1WW
│
├─Battery:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    1.3.2
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               6d8e9931-55aa-4ff3-bb93-a7cd1a1bc5d7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
```

I will carry out the other outstanding update (firmare 0.1.55) and update this issue accordingly.

ChiWei-Chen, 2023-06-05:
Thanks, I'll keep you update if any progress from FW team. (LO-2491 for my internal tracking)

ChiWei-Chen, 2023-06-05:
Hi @adwk67, the Battery team replied follows:
0% symptom is same as #329 case but we're not sure if root cause is same or not, because those batteries do not have same issue with T14s Gen2.
And we found this comment. (power supply is not detected)
On reboot, the update is not updated as the power supply is not detected, and the updates are re-presented for download:

If power supply is stopped during battery FW update, that will cause abnormal system shutdown and battery FW corruption. 
Once battery FW corruption occurs, the battery icon will be 0% and the output is OFF until the retry FW update.

Is it possible to check if the system can work in DC mode(without adapter plugged) or not during 0% symptom?
Thanks!

adwk67, 2023-06-06:
The updates have worked (and are now no longer being re-presented on each re-boot) but the battery does not appear to be charging. It was on 96% after the updates (whilst plugged in to the power source) but then I worked with it for a while and saw it drop from 96% to 57% (as expected), and then plugged in the power source again. Since then, the display remains at 57% (laptop currently attached to power source):

[image]

adwk67, 2023-06-16:
@ChiWei-Chen Is there any update on this? My laptop has remained plugged in since my last comment: it is now on 53% and is still not re-charging.

ChiWei-Chen, 2023-06-16:
@adwk67 could you try this 1.56 BIOS and see how it works? Thanks.
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN34ET.firmware

adwk67, 2023-06-16:
I've already updated to 0.1.56 a day or two ago and the issue remains.

adwk67, 2023-06-16:
```
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Previous version:   0.1.55
  │   Update State:       Success
  │   Last modified:      2023-06-15 08:44
  │   GUID:               973a10fa-79c0-453e-b78c-27b55cc0e6eb
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2:
        New version:      0.1.56
        Remote ID:        lvfs
        Release ID:       35528
        Summary:          Lenovo ThinkPad T14 Gen 2 / P14s Gen 2 / T15 Gen 2 / P15s Gen 2 System Firmware
        License:          Proprietary
        Size:             18,3 MB
        Created:          2021-06-15
        Urgency:          Critical
        Vendor:           Lenovo
        Description:      
        This stable release fixes the following issues:
        
        • Updated includes a security fix.
        
        Some new functionality has also been added:
        
        • Updated the Diagnostics module to version 04.29.000.
```

adwk67, 2023-06-19:
Closing this as - as of today - the issue seems to have fixed itself (possibly after 0.1.56, although that update was a few days ago): I unplugged the laptop while it was turned off, and when re-connecting the power supply and re-starting it, the battery has started to charge again.

closed 2023-06-19 by adwk67

ChiWei-Chen, 2023-06-19:
@adwk67 it's great to know this works for you!