fwupd/firmware-lenovo #348: fwupd pushed broken system firmware 1.23 for P14s

rubin55, 2023-06-12:
Yesterday I used `fwupd` to upgrade my P14s to system firmware version 1.23. It turns out that that firmware is broken (see #282 for the exact same issue). Lenovo released version 1.24 to address this, which seems to not be available through `fwupd`. 

Attempting to manually install the firmware, either through `fwupd install` or `fwupdate -a` fails due to the battery charge erronuously being detected as empty. Attempting to install through the official Lenovo firmware update ISO fails with a message "Make sure that a charged battery pack is installed in the system".

How do I force-force installation of firmware 1.24 so the system can correctly identify the battery again (note: it does see the battery, it refuses to charge)?

hughsie, 2023-06-12:
@mrhpearson i thought I untagged all the battery firmware? has more been pushed?

rubin55, 2023-06-12:
I see that the battery id is identical to that mentioned in #315 (f95c9218acd12697af946874bfe4239587209232). Following the steps:

1. download [mentioned](https://github.com/fwupd/firmware-lenovo/issues/315#issuecomment-1425377179) battery [firmware](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN34BL_00060109.firmware)
2. Unpack cab using `7z x`
3. Running `fwupdate -a <my-guid> firmware.bin`

Gives an error:

```
[root@THINK ~]# fwupdate -a 7a9bff9b-9aab-445b-a6c7-21f5ce12e95e firmware.bin

(fwupdate:5725): FuPluginUefiCapsule-WARNING **: 11:22:17.024: missing or invalid embedded capsule header

```

hughsie, 2023-06-12:
> Gives an error:
(fwupdate:5725): FuPluginUefiCapsule-WARNING **: 11:22:17.024: missing or invalid embedded capsule header

That's a warning -- don't panic.

> Update: I just updated to 1.9.1, but that seems to not have the fwupdate tool anymore. What to do?

`fwupdmgr install-blob firmware.bin` should do exactly the same steps.

rubin55, 2023-06-12:
Running `fwupdtool install-blob firmware.bin` gives me:

```
[root@THINK ~]# fwupdtool install-blob firmware.bin
Loading…                 [                                       ]X11 connection rejected because of wrong authentication.
Writing…                 [************                           ]
0.	Cancel
1.	f95c9218acd12697af946874bfe4239587209232 (Battery)
2.	71b677ca0f1bc2c5b804fa1d59e52064ce589293 (MZVL22T0HBLB-00BL7)
3.	349bb341230b1a86e5effe7dfe4337e1590227bd (System Firmware)
4.	a45df35ac0e948ee180fe216a5f703f32dda163f (UEFI Device Firmware)
5.	2292ae5236790b47884e37cf162dcf23bfcd1c60 (UEFI Device Firmware)
6.	d96de5c124b60ed6241ebcb6bb2c839cb5580786 (UEFI Device Firmware)
7.	f37fb01122dd62c773f4e84ec89737e059712d59 (UEFI Device Firmware)
8.	36efb79c255f402f619fa9eb53cd659db51f2a04 (UEFI Device Firmware)
9.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
Choose device [0-9]: 1
Waiting…                 [***************************************]
An update requires a reboot to complete. Restart now? [y|N]: y
```

After update I get a quick message mentioning: `Device Error`. After that it reboots into the regular OS.

hughsie, 2023-06-12:
What does `sudo fwupdtool get-devices --plugins uefi_capsule` say?

rubin55, 2023-06-12:
After the previous steps trying to install the firmware manually, I retried as follows:

1. fwupdmgr refresh --force
2. fwupdmgr get-updates (notice that the battery says that a version 1.2.10 is available)
3. fwupdate update
4. reboot, don't touch black screen. just wait for 10 minutes
5. now have working battery with system firmware 1.23
6. download version 1.24 from support.lenovo.com
7. use fwupdmgr install r1mul54w.cab (to upgrade to 1.24)
8. reboot, don't touch black screen. just wait for 10 minutes
9. now have working battery with system firmware 1.24

rubin55, 2023-06-12:
> What does `sudo fwupdtool get-devices --plugins uefi_capsule` say?
[out.txt](https://github.com/fwupd/firmware-lenovo/files/11722886/out.txt)

rubin55, 2023-06-12:
I managed to rescue this from my terminal history:

```
[root@THINK ~]# fwupdmgr get-devices
LENOVO 21A0CTO1WW
│
├─Battery:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    1.1.9
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               7a9bff9b-9aab-445b-a6c7-21f5ce12e95e
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─MZVL22T0HBLB-00BL7:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    AL2QGXB7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S64SNF0R401564
│     GUIDs:              bec63ed7-a95f-54fe-b8cc-8e9fee64ba5a ← NVME\VEN_144D&DEV_A80A
│                         60c89aac-f321-515b-b419-3cf02aa9d375 ← NVME\VEN_144D&DEV_A80A&REV_00
│                         310f81b5-6fce-501e-acfb-487d10501e78 ← NVME\VEN_144D&DEV_A80A&SUBSYS_144DA801
│                         4d7a2791-106b-5e72-9cfb-8ea3d89f5421 ← NVME\VEN_144D&DEV_A80A&SUBSYS_144DA801&REV_00
│                         75877a5d-2070-57b7-ba80-feb6c937df57 ← SAMSUNG MZVL22T0HBLB-00BL7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─System Firmware:
│ │   Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.23
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Transient failure
│ │   Update Error:       failed to update to 117474352: battery level is too low
│ │   GUIDs:              8fb7e1f9-447b-41fb-8832-ae5ef955665f
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  190
│       Minimum Version:  190
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            1e9c3d61-4bd9-514e-ad87-0a9b61b9e77d ← UEFI\CRT_156645FC005F8617669DF76E1DE6FAE02B3856D24B624E47D97BD2B8ADFFB6E0
│                         304a9ede-aff2-58f6-b5e5-0e651b065f1a ← UEFI\CRT_156645FC005F8617669DF76E1DE6FAE02B3856D24B624E47D97BD2B8ADFFB6E0&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
[... 61 lines cut ...]
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               4bea12df-56e3-4cdb-97dd-f133768c9051
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
      Summary:            UEFI ESRT device
      Current version:    0
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               3954e118-d997-4499-b917-d4c454e4b124
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
Idle…                    [***************************************]
Successfully uploaded 1 report
```

hughsie, 2023-06-12:
> After update I get a quick message mentioning: Device Error

Can you describe that a bit more please -- anything would be helpful.

mrhpearson, 2023-06-12:
> @mrhpearson i thought I untagged all the battery firmware? has more been pushed?

There are new fixed versions released...there shouldn't be any broken versions. Going through to double check though.

@rubin55 - I don't think this was system FW update related. If you check 'fwupdmgr get-history' does it show updates done for the battery FW?

mrhpearson, 2023-06-12:
I think this is a different issue to the ones being pointed at. 
The battery FW 1.2.10 was released on LVFS 11 months ago - with a fair few downloads and success report.

Only thought is if there is something in the system FW that relies on that battery FW update having been done? I'll see if I can see anything in the internal release notes.

As a note - it's generally recommended to wait for FW to be released on LVFS as that means that it has been tested on Linux by the FW team. I know it's frustrating that it takes us longer to get the LVFS releases done (seems an endless task to improve that....but it is something being worked on) but we've had a couple of cases where a FW update broke something under Linux and we had to skip a release to get that addressed.

You're welcome to nag me for status if it's been more than a couple of weeks for a FW to be releases (and in this case I will note that it's been in embargo for 26 days....so I do need to go chase them).

rubin55, 2023-06-13:
@mrhpearson here's the output of `fwupdmgr get-history`:

```
├─Battery:
│ │   Device ID:          f95c9218acd12697af946874bfe4239587209232
│ │   Previous version:   1.1.9
│ │   Update State:       Success
│ │   Last modified:      2023-06-12 11:38
│ │   GUID:               7a9bff9b-9aab-445b-a6c7-21f5ce12e95e
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14 Gen 2/P14s Gen 2 Battery Update:
│       New version:      1.2.10
│       Remote ID:        lvfs
│       Release ID:       14110
│       Summary:          Lenovo ThinkPad  T14 Gen 2/P14s Gen 2 Battery Firmware
│       License:          Proprietary
│       Size:             98.9 kB
│       Created:          2016-07-08
│       Urgency:          High
│       Tested by Lenovo:
│         Tested:         2022-06-30
│         Distribution:   ubuntu 22.04
│         Old version:    0.1.1
│         Version[fwupd]: 1.7.5
│       Details:          https://pcsupport.lenovo.com/de/en/search?query=R1MBL03W
│       Vendor:           Lenovo
│       Release Flags:    • Trusted metadata
│                         • Tested by trusted vendor
│       Description:      
│       Lenovo Thinkpad T14 Gen 2/P14s Gen 2 Battery Firmware Version 258.0.0.10
│       
│       change in this release Added support for ThinkPad T14 Gen 2 AMD (Machine Types: 20XK, 20XL)
│       
│       Added support for ThinkPad T14 Gen 2 AMD (Machine Types: 20XK, 20XL).Added support for ThinkPad P14s Gen 2 AMD	(Machine Types: 21A0, 21A1).Problem fixes:Improved charging performance for the following battery:FRU P/N: 5B10W51829 (ASM P/N: SB10W51928)
```

mrhpearson, 2023-06-14:
Thanks - confirms a battery FW update happened and probably triggered this.
I've created LO-2512 to flag this to the FW team for review.

Just to confirm - you're all up and running now without issues right?

rubin55, 2023-06-15:
@mrhpearson yes, up + running, haven't had any particular issues since!

Genues, 2023-06-19:
Hi.
On my system last battery firmware update (up to version 1.1.10246)  broke information about  capability 
`
+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 5B10W51826
/sys/class/power_supply/BAT0/cycle_count                    =     13
/sys/class/power_supply/BAT0/energy_full_design             =  50500 [mWh]
/sys/class/power_supply/BAT0/energy_full                    = 121210 [mWh]
/sys/class/power_supply/BAT0/energy_now                     = 107000 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  22658 [mW]
...
Charge                                                      =   90.5 [%]
Capacity                                                    =  240.0 [%]
`

Abdul-Karim-Jahan, 2023-10-16:
Firmware 1.23 is causing several issue ,in my thinkpad e14 gen 5 amd, like after sleep its not booting up so i had to force shutdown the laptop, ( i am a win 11 user) after performing system restore the os got completely crashed and only showing blue screen of death resulting me to reinstall os. To fix sleep issue i tried to reset the bios to default which leads to bitlocker issue in which i had to enter bitlocker recovery key each and every time i bootup my laptop. In the end i turned off bitlocker(device encryption) and now my bios is on factory default state and the sleep issue is also fixed.

ChiWei-Chen, 2023-10-17:
@Abdul-Karim-Jahan , thanks for reporting this.
Could also report it to forums.lenovo.com , since this issue is found in Windows environment, I believe you'll get immediate support there.