fwupd/firmware-lenovo #349: lvfs-testing: Embedded Controller update for T15p/P15v Gen 2 offered on T15g Gen 2

DerDreschner, 2023-06-16:
I have a ThinkPad T15g Gen 2 using the latest Linux Mint together with the following versions of fwupd:
```
➜  ~ sudo fwupdmgr --version                              
compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.3.7-060307-generic
runtime   org.freedesktop.fwupd         1.8.12
```

When I activate lvfs-testing, fwupd wants to install the following update: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN38HT.firmware

[image]

This is wrong, as this is only for T15p Gen 2 / P15v Gen 2 and not for my T15g Gen 2. I installed it once before recognize this obstacle, but the update itself got aborted by the UEFI before being flashed to the embedded controller (thankfully).

Could someone forward that issue to the correct team at Lenovo? Thanks in advance!

mrhpearson, 2023-06-19:
Hi @DerDreschner 
Would you mind collecting the contents of your ESRT table please:

`sudo grep . /sys/firmware/efi/esrt/entries/*/*`

I want to confirm if the FW on LVFS is wrong, or if the BIOS is for some reason using the wrong GUID in the ESRT table.
Thanks
Mark

DerDreschner, 2023-06-19:
Hey @mrhpearson,

I don't mind providing any information that can help in that issue. Here is the output:
```
➜  ~ sudo grep . /sys/firmware/efi/esrt/entries/*/*       
/sys/firmware/efi/esrt/entries/entry0/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry0/fw_class:a1392d82-62d5-4e24-863a-0f682993408f
/sys/firmware/efi/esrt/entries/entry0/fw_type:1
/sys/firmware/efi/esrt/entries/entry0/fw_version:65562
/sys/firmware/efi/esrt/entries/entry0/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry0/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry0/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry1/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry1/fw_class:666705ce-0624-4943-8738-09829a26c8ee
/sys/firmware/efi/esrt/entries/entry1/fw_type:2
/sys/firmware/efi/esrt/entries/entry1/fw_version:4029286587
/sys/firmware/efi/esrt/entries/entry1/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry1/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry1/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry2/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry2/fw_class:e8dddfb5-175b-4ca5-b77a-ddb98c30f0c0
/sys/firmware/efi/esrt/entries/entry2/fw_type:2
/sys/firmware/efi/esrt/entries/entry2/fw_version:65546
/sys/firmware/efi/esrt/entries/entry2/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry2/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry2/lowest_supported_fw_version:65546
/sys/firmware/efi/esrt/entries/entry3/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry3/fw_class:a5aea3ab-1b50-4943-b6c9-04c041980f50
/sys/firmware/efi/esrt/entries/entry3/fw_type:2
/sys/firmware/efi/esrt/entries/entry3/fw_version:16984070
/sys/firmware/efi/esrt/entries/entry3/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry3/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry3/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry4/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry4/fw_class:52c9711e-7a5d-4e8d-ac4a-ba4d920a1008
/sys/firmware/efi/esrt/entries/entry4/fw_type:2
/sys/firmware/efi/esrt/entries/entry4/fw_version:0
/sys/firmware/efi/esrt/entries/entry4/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry4/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry4/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry5/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry5/fw_class:e97c0e4d-defe-4d92-a079-24ebc5168046
/sys/firmware/efi/esrt/entries/entry5/fw_type:2
/sys/firmware/efi/esrt/entries/entry5/fw_version:28
/sys/firmware/efi/esrt/entries/entry5/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry5/last_attempt_version:28
/sys/firmware/efi/esrt/entries/entry5/lowest_supported_fw_version:1
/sys/firmware/efi/esrt/entries/entry6/capsule_flags:0x0
/sys/firmware/efi/esrt/entries/entry6/fw_class:2f45f824-5964-4cf7-a20e-6b2a1e3e69f0
/sys/firmware/efi/esrt/entries/entry6/fw_type:2
/sys/firmware/efi/esrt/entries/entry6/fw_version:65545
/sys/firmware/efi/esrt/entries/entry6/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry6/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry6/lowest_supported_fw_version:65545
/sys/firmware/efi/esrt/entries/entry7/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry7/fw_class:072c0411-31b3-4e4f-85bc-1e1b5ed55223
/sys/firmware/efi/esrt/entries/entry7/fw_type:2
/sys/firmware/efi/esrt/entries/entry7/fw_version:2754747
/sys/firmware/efi/esrt/entries/entry7/last_attempt_status:1
/sys/firmware/efi/esrt/entries/entry7/last_attempt_version:2754747
/sys/firmware/efi/esrt/entries/entry7/lowest_supported_fw_version:0
/sys/firmware/efi/esrt/entries/entry8/capsule_flags:0x8010
/sys/firmware/efi/esrt/entries/entry8/fw_class:2bece6a9-392e-48de-9cd1-67dc82dcfead
/sys/firmware/efi/esrt/entries/entry8/fw_type:2
/sys/firmware/efi/esrt/entries/entry8/fw_version:22808
/sys/firmware/efi/esrt/entries/entry8/last_attempt_status:0
/sys/firmware/efi/esrt/entries/entry8/last_attempt_version:0
/sys/firmware/efi/esrt/entries/entry8/lowest_supported_fw_version:1
➜  ~
```

Please keep in mind that - for whatever reason - the update is marked as successful in fwupd. This is the output from `fwupdmgr get-devices` for the embedded controllers:

```
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.10
│     Minimum Version:    0.1.10
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               e8dddfb5-175b-4ca5-b77a-ddb98c30f0c0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Embedded Controller:
│     Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
│     Summary:            UEFI ESRT device
│     Current version:    0.1.9
│     Minimum Version:    0.1.9
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               2f45f824-5964-4cf7-a20e-6b2a1e3e69f0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
```
If you need any additional information, just let me know.

David

mrhpearson, 2023-06-19:
Thanks - that confirms the ESRT table GUID matches the T15p/P15v FW being offered. So this either means the FW is wrong; or the FW metadata on LVFS needs updating.

Created internal ticket LO-2521 and asking the FW team for their input on what is wrong

Mark

DerDreschner, 2023-06-19:
Alright, thanks for your effort and good work! Looking forward for any updates.

David

DerDreschner, 2023-07-20:
@ChiWei-Chen : Sorry for my late response due to personal circumstances.

Yes, I use the latest version 1.26. I flashed it via fwupd. Every previous version (including 1.24) was flashed using the appropriate Windows update file from lenovo.com.

David

ChiWei-Chen, 2023-07-21:
@DerDreschner , thanks.
Could you provide whole output of `fwupdmgr get-devices` for the analysis? Thanks!

DerDreschner, 2023-07-21:
@ChiWei-Chen: Yeah, of course I can. Here it is:

```
  ~ sudo fwupdmgr get-devices
LENOVO 20YS000CGE
│
├─11th Gen Intel Core™ i9-11950H @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000042
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         a6bd4ca5-75a6-5796-b564-66b5cab1b11b ← CPUID\PRO_0&FAM_06&MOD_8D
│                         d9dd5e77-df17-5bab-b5ec-22827598bfed ← CPUID\PRO_0&FAM_06&MOD_8D&STP_1
│     Device Flags:       • Internal device
│   
├─Battery:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    1.3.10246
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               a5aea3ab-1b50-4943-b6c9-04c041980f50
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.10
│     Minimum Version:    0.1.10
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               e8dddfb5-175b-4ca5-b77a-ddb98c30f0c0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Embedded Controller:
│     Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
│     Summary:            UEFI ESRT device
│     Current version:    0.1.9
│     Minimum Version:    0.1.9
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               2f45f824-5964-4cf7-a20e-6b2a1e3e69f0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─GA104M [GeForce RTX 3070 Mobile / Max-Q]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE, PCI:0x8086)
│     GUIDs:              402f12fa-651c-57c6-b5c5-cf4e59c4cce7 ← PCI\VEN_10DE&DEV_249D
│                         f32dbf96-ef8c-5105-b143-8755f08f93da ← PCI\VEN_10DE&DEV_249D&REV_A1
│                         fa3b0f74-def0-5478-9d5d-ed2383f36046 ← PCI\VEN_10DE&DEV_249D&SUBSYS_17AA22E1
│                         7f3cd266-1c69-52dc-9a87-639ca9c13c02 ← PCI\VEN_10DE&DEV_249D&SUBSYS_17AA22E1&REV_A1
│                         c7dac7da-12ca-57be-ba1d-8617ec295689 ← PCI\VEN_8086&DEV_9A01
[... 233 lines cut ...]
│     Summary:            UEFI ESRT device
│     Current version:    22808
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               2bece6a9-392e-48de-9cd1-67dc82dcfead
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─USB3.0 Hub:
      Device ID:          cf294bf55b333004beb7c41f952c1838c23e1f4a
      Current version:    38.0
      Vendor:             Intel Corporation. (USB:0x8087)
      GUIDs:              b3f8b518-ebdd-5969-9b44-6b096eff5b9b ← USB\VID_8087&PID_0B40
                          27f4ed60-ed53-52a7-ae28-ea87d9a23302 ← USB\VID_8087&PID_0B40&REV_1234
                          c6e7a14f-f13c-55e5-99de-d59461db1d27 ← USB\VID_8087&PID_0B40&HUB_0000
                          350ec5bb-25c0-543e-8d4c-72c30d2b8df0 ← TBT-01082031
      Device Flags:       • Updatable
                          • Signed Payload
    
➜  ~
```

ChiWei-Chen, 2023-07-25:
@DerDreschner , I've shared the output to FW team, I'll keep you updated if any progress. Thanks!

ChiWei-Chen, 2023-08-01:
@DerDreschner , the FW team wants to have a video (or detail steps) how did you try to update the EC on this system.

They used the commands below to update and the result is pass: 
```
fwupdmgr --force refresh
fwupdmgr --force update  
```

DerDreschner, 2023-08-01:
@ChiWei-Chen : Sure, here is a video with the steps. But it's not a lot. Just activating lvfs-testing and then use the commands you posted.

https://github.com/fwupd/firmware-lenovo/assets/34408442/7c172613-592c-4397-b8a0-25c72ffb8693

In case you need any more information, just let me know!

ChiWei-Chen, 2023-08-02:
@DerDreschner , thanks! I'll give you back if any progress from the team.

DerDreschner, 2023-08-12:
Hey @ChiWei-Chen, the firmware update team released a new version of the [problematic embedded controller firmware](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN38HT.firmware) to the stable branch. fwupd wants to install it on my T15g Gen 2, although it's clearly compatible. As it's marked as stable, there are no changes to any config files necessary to trigger that behavior on my machine.

[image]

Let me know if you need any more informations.

ChiWei-Chen, 2023-08-14:
@DerDreschner , thanks for the feedback.
The FW team still investigating this issue, but they want to know if "load default setting in BIOS" can help with this, could you give it a try?
Thanks!

DerDreschner, 2023-08-14:
@ChiWei-Chen: Negative. I loaded the default setup settings without any change. I also tried using the OS Optimized Defaults - which are disabled on my machine as I don't want secure boot etc. enabled - without any luck. fwupd still wants to install this firmware update on my machine.

Let my know if you need anything else.

ChiWei-Chen, 2023-08-15:
@DerDreschner , thanks, I shared this with FW team. I'll give you update if any progress from the team.

DerDreschner, 2023-08-15:
@ChiWei-Chen : FYI, I installed the new update (UEFI: 1.27 / ECP: 1.12) today. Still no change.

ChiWei-Chen, 2023-08-16:
@DerDreschner , thanks, I've shared this with the FW team. They're checking this internally.

ChiWei-Chen, 2023-12-13:
@DerDreschner , great! Thanks!

closed 2023-12-13 by ChiWei-Chen