fwupd/firmware-lenovo #350: can't update to new BIOS update to fix not charging battery on Thinkpad X1 Carbon Gen 9

ClubSaumon07, 2023-06-18:
System : Fedora 38

Describe the bug

Battery at 0% status on not charging even when AC plugged in. Have seen some issues about it but can't manage to udate the BIOS to the latest version without at least 25% battery level.d

Steps to Reproduce
Trigger a firmware update using fwupdmgr update, reboot.

Expected behavior
The firmware should have been updated.

fwupd version information :

    fwupdmgr --version
    compile   org.freedesktop.fwupd         1.9.2
    compile   com.hughsie.libxmlb           0.3.11
    compile   com.hughsie.libjcat           0.1.13
    runtime   org.freedesktop.fwupd-efi     1.4
    compile   org.freedesktop.gusb          0.4.5
    runtime   com.hughsie.libjcat           0.1.13
    runtime   org.freedesktop.gusb          0.4.5
    runtime   org.freedesktop.fwupd         1.9.2
    runtime   org.kernel                    6.3.8-200.fc38.x86_64

Even with the setting `ignorePower=True` in `/etc/fwupd/daemon.conf `, the reboot screen indicate the following error :` Check power failure. Please insert power source.` exactly like in this issue ([here](https://github.com/fwupd/firmware-lenovo/issues/348#issue-1752232606)).

How do I proceed to update my firmware ?

l-vincent-l, 2023-06-19:
I had the same  problem with ubuntu.

A patch was sent yesterday and I was able to upgrade the firmware.

However, I now have battery but I can't charge it.

mrhpearson, 2023-06-20:
I've added a note to the open case we have on this (LO-2416).
Could you please confirm which BIOS, EC and Battery version you have on these systems please. Should be displayed with `fwupdmgr get-devices`

ChiWei-Chen, 2023-06-21:
@l-vincent-l , @ClubSaumon07 ,
Could you try to update battery firmware with following instructions?
```
download the correct firmware.bin to the <home> directory
fwupdate -a <GUID> firmware.bin
reboot
```
*note: there are multiple batteries of X1C9, please download correct one is using: https://fwupd.org/lvfs/search?value=Carbon+Gen+9+Battery

Thanks.

ClubSaumon07, 2023-06-21:
> I've added a note to the open case we have on this (LO-2416). Could you please confirm which BIOS, EC and Battery version you have on these systems please. Should be displayed with `fwupdmgr get-devices`

    Battery:
    │     Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
    │     Résumé:             UEFI ESRT device
    │     Version actuelle:   1.9.2
    │     Version minimum:    0.0.1
    │     Fournisseur:        Lenovo (DMI:LENOVO)
    │     État de mise à jour:Success
    │     GUID:               7fdc5607-979e-4b5c-899d-39afd61f4512
    │     Drapeaux de périphérique:• Périphérique interne
    │                         • Mise à jour possible
    │                         • Le système nécessite une source d'alimentation externe
    │                         • Supported on remote server
    │                         • Needs a reboot after installation
    │                         • Device is usable for the duration of the update

    System Firmware:
    │ │   Device ID:          11a7fbdd09ce583e58b899660fe10f147fd155a2
    │ │   Résumé:             UEFI ESRT device
    │ │   Version actuelle:   0.1.52
    │ │   Version minimum:    0.1.39
    │ │   Fournisseur:        Lenovo (DMI:LENOVO)
    │ │   État de mise à jour:Transient failure
    │ │   Problems:           • System power is too low to perform the update
    │ │   GUIDs:              14f3350e-cf63-4e68-a0d9-0af1d5389a17
    │ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
    │ │   Drapeaux de périphérique:• Périphérique interne
    │ │                       • Le système nécessite une source d'alimentation externe
    │ │                       • Supported on remote server
    │ │                       • Needs a reboot after installation
    │ │                       • Cryptographic hash verification is available
    │ │                       • Device is usable for the duration of the update
    │ │                       • Mise à jour possible

    ├─Embedded Controller:
    │     Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
    │     Résumé:             UEFI ESRT device
    │     Version actuelle:   0
    │     Fournisseur:        Lenovo (DMI:LENOVO)
    │     État de mise à jour:Transient failure
    │     Problems:           • System power is too low to perform the update
    │     GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
    │     Drapeaux de périphérique:• Périphérique interne
    │                         • Le système nécessite une source d'alimentation externe
    │                         • Supported on remote server
    │                         • Needs a reboot after installation
    │                         • Device is usable for the duration of the update
    │                         • Mise à jour possible

ClubSaumon07, 2023-06-21:
> @l-vincent-l , @ClubSaumon07 , Could you try to update battery firmware with following instructions?
> 
> ```
> 1. download the correct firmware.bin to the <home> directory
> 2. fwupdate -a <GUID> firmware.bin
> 3. reboot
> ```
> 
> *note: there are multiple batteries of X1C9, please download correct one is using: https://fwupd.org/lvfs/search?value=Carbon+Gen+9+Battery
> 
> Thanks.

This is not working, when rebooting I stil have the  `Check power failure. Please insert power source.` when the lenovo logo show up.

ChiWei-Chen, 2023-06-23:
@ClubSaumon07 , the FW team suspects this no charging could be caused by Battery Firmware Update failure.
This is also new to the team, they will try to duplicate this. But please expect a delay since the team is on leave until Sunday.

ChiWei-Chen, 2023-06-26:
hi @ClubSaumon07 ,
The team needs the following out put for further debugging, could you help with this? 
`fwupdmgr get-devices `
`lsb_release -a`

If the Embedded Controller Firmware still shows zero, could you check it in BIOS Menu? Thanks!

l-vincent-l, 2023-06-26:
> @l-vincent-l , @ClubSaumon07 , Could you try to update battery firmware with following instructions?
>
> ```
> 1. download the correct firmware.bin to the <home> directory
> 2. fwupdate -a <GUID> [...]

I was able to update my firmware thanks to an update, then after a reset of the battery and a reboot everything is now working great.

ClubSaumon07, 2023-06-26:
> hi @ClubSaumon07 , The team needs the following out put for further debugging, could you help with this? `fwupdmgr get-devices ` `lsb_release -a`
> 
> If the Embedded Controller Firmware still shows zero, could you check it in BIOS Menu? Thanks!

From the bios : 
UEFI BIOS Version : N32ET76W (1.52)
Controller Version : N32HT52W (1.32)

ChiWei-Chen, 2023-06-28:
Hi @ClubSaumon07 , the FW team verified that they can upgrade battery firmware successfully even the battery is 0%, could you try with the following instructions?
1. Download the [N32BL04W_L00070206.cab](https://drive.google.com/file/d/1s0NlGaL31LispYpFvbMJEZkLq4-J1U3D/view?usp=drive_link)
2. Double-click the cab file
[image]
3. Press the Install button
[image]
4. Then start the restart and install
[image]
After the system reboot, it should be updated.
The machine is verified in following environment:
```
BIOS N32ET84W(1.60)
EC N32HT53W(1.33)
ME 15.0.42.2235
OS fedora 38
```
Please let us know if this doesn't work for you.

ClubSaumon07, 2023-06-29:
> Hi @ClubSaumon07 , the FW team verified that they can upgrade battery firmware successfully even the battery is 0%, could you try with the following instructions?
>
>     1. Download the [...]

I am not able to find this firmware on fwupd.org and I can't download the file you sent, because of google drive restriction

ClubSaumon07, 2023-06-29:
> Hi @ClubSaumon07 , the FW team verified that they can upgrade battery firmware successfully even the battery is 0%, could you try with the following instructions?
>
>     1. Download the [...]

It did not change anything, the battery still won't charge

ChiWei-Chen, 2023-06-30:
@ClubSaumon07 , thanks for the update, could you provide the `fwupdmgr get-history`?

ClubSaumon07, 2023-06-30:
> @ClubSaumon07 , thanks for the update, could you provide the `fwupdmgr get-history`? (LO-2525 for my internal traking)

    Prometheus:
    │ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
    │ │   Version précédente: 10.01.3333223
    │ │   GUID:               896bb9a6-a8be-5727-8a3a-43ca249f6933
    │ │   Drapeaux de périphérique:• Mise à jour possible
    │ │                       • Cryptographic hash verification is available
    │ │                       • Signed Payload
    │ │ 
    │ └─Mise à jour de (null):
    │       Nouvelle version: 10.01.3478575
    │       Licence:          Inconnu
    │       Description:      
    │       The vendor did not supply any release notes.
    │     
    ├─Battery:
    │ │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
    │ │   Version précédente: 1.1.1
    │ │   État de mise à jour:Success
    │ │   GUID:               7fdc5607-979e-4b5c-899d-39afd61f4512
    │ │   Drapeaux de périphérique:• Périphérique interne
    │ │                       • Mise à jour possible
    │ │                       • Le système nécessite une source d'alimentation externe
    │ │                       • Needs a reboot after installation
    │ │                       • Device is usable for the duration of the update
    │ │ 
    │ └─Mise à jour de (null):
    │       Nouvelle version: 1.9.2
    │       Licence:          Inconnu
    │       Description:      
    │       The vendor did not supply any release notes.
    │     
    ├─KXG7AZNV512G LA KIOXIA:
    │ │   Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
    │ │   Version précédente: 5118AJLA
    │ │   GUID:               21978846-c3a3-54f1-a92d-78dea928c1c9
    │ │   Drapeaux de périphérique:• Périphérique interne
    │ │                       • Mise à jour possible
    │ │                       • Le système nécessite une source d'alimentation externe
    │ │                       • Supported on remote server
    │ │                       • Needs a reboot after installation
    │ │                       • Device is usable for the duration of the update
    │ │                       • Signed Payload
    │ │ 
    │ └─KXG7AZNV256G LA / KXG7AZNV512G LA / KXG7AZNV1T02 LA SSD Update:
    │       Nouvelle version: 5123AJLA
    │       Remote ID:        lvfs
    │       Release ID:       28588
    │       Résumé:           KIOXIA XG7 NVMe SSD Firmwre for Lenovo PC
    │       Licence:          Propriétaire
    │       Taille:           1,5 Mo
    │       Created:          2023-03-31
    │       Urgence:          Haute
    │       Fournisseur:      Lenovo
    │       Release Flags:    • Trusted metadata
    │       Description:      
    │       [Support devices and Firmware version] KXG7AZNV256G 5123AJLA, KXG7AZNV512G 5123AJLA, KXG7AZNV1T02 5123AJLA
    │       
    │       [Problem fixes]1.Fixed an issue where BSoD might occur during heavy load work./ 2.Fixed an issue where BSoD issue and M.2 SSD Detection error might occur.
    │       
    │       [Support Product Scope]Lenovo ThinkPad, ThinkCentre, ThinkStation, IdeaCentre
    │     
    ├─UEFI Device Firmware:
    │ │   Device ID:          ae1abd099407b1d95698d69b7273f7fed5c6f35c
    │ │   Version précédente: 1.2.22.0
    │ │   État de mise à jour:Success
    │ │   GUID:               aa096a98-94e6-479b-92f7-5771f6f2d96f
    │ │   Drapeaux de périphérique:• Périphérique interne
    │ │                       • Mise à jour possible
[... 196 lines cut ...]
              Version[fwupd]: 1.7.4
            Tested by Lenovo:
              Tested:         2023-04-12
              Distribution:   sled 15.3
              Old version:    0.1.59
              Version[fwupd]: 1.5.8
            Fournisseur:      Lenovo
            Release Flags:    • Trusted metadata
                              • Is upgrade
                              • Tested by trusted vendor
            Description:      
            Lenovo System Firmware Version 1.60
            
            Important updates
            
            • Update includes a security fix.
            
            Problem fixes
            
            • Fixed an issue where NVMe passwords could not be erased properly.
            • Fixed an issue where abnormal shutdown may be logged when Thunderbolt device is attached or detached while system is in shutdown state.
      

Interestingly the System firmware is flagged with the error "An update is in progress". 
Yesterday I tried to update the BIOS as recommended [here](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-9th-gen-type-20xw-20xx/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C) and once I restart the computer after the update I have a black screen and it is not rebooting. I waited for some hours and still no reboot.

ChiWei-Chen, 2023-07-03:
> Interestingly the System firmware is flagged with the error "An update is in progress".
> Yesterday I tried to update the BIOS as recommended [here](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-9th-gen-type-20xw-20xx/downloads/driver-list/component?name=BIOS%2FUEFI&id=5AC6A815-321D-440E-8833-B07A93E0428C) and once I restarted the computer after the update I had a black screen. I waited for some hours still the computer wouldn't turn on after the install, I had to manually shut it down...

@ClubSaumon07 , I've shared with FW team.
Is it bootable after manually shut it down?

ClubSaumon07, 2023-07-03:
> @ClubSaumon07 , I've shared with FW team. Is it bootable after manually shut it down?

After clicking on reboot the computer turn off and then a it still on because I can hear fans and a function key still has a light on but the screen is black. I have to press the on/off button for some times to force the computer to shut down and then turn it on manually after.
Its like the computer is stuck after the reboot and can't proceed the installations.

On [this issue](https://github.com/fwupd/firmware-lenovo/issues/348#issuecomment-1587217265) the user said that he has to wait 10 minutes after black screen but I waited much longer.

ClubSaumon07, 2023-07-03:
I just saw [this comment ](https://github.com/fwupd/firmware-lenovo/issues/315#issuecomment-1431491697) and was wondering with using fwupdate instead of fwupdmgr ad I thought fwupdate was depreciated.

I'll try to update with fwupdate with the firmware I downloaded from your google drive. I'll keep you updated thank you for the help.

ClubSaumon07, 2023-07-03:
@ChiWei-Chen I finally managed to update everything !

I updated the battery firmware manually from the file like you told me. Then when the computer reboot is detached the power supply and waited for the battery to die (I waited 2 hours to be sure). Then I plugged back the power supply and when I clicked on ON it was installing the updates with the update screen. 
Now my battery is charging. 
I think this is the way to deal with this issue since all the other methods didn't work.

Many thanks for your help @mrhpearson & @ChiWei-Chen

ClubSaumon07, 2023-07-03:
I managed to update the battery firmware, to do so I manually selected the .cab file and when the black screen showed up after reboot I unplugged the AC, waited for the battery to die and re-plugged the AC. 
After hitting ON the update was in progress.

      ─Battery:
      │ │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
      │ │   Version précédente: 1.1.1
      │ │   État de mise à jour:Success
      │ │   GUID:               7fdc5607-979e-4b5c-899d-39afd61f4512
      │ │   Drapeaux de périphérique:• Périphérique interne
      │ │                       • Mise à jour possible
      │ │                       • Le système nécessite une source d'alimentation externe
      │ │                       • Needs a reboot after installation
      │ │                       • Device is usable for the duration of the update
      │ │ 
      │ └─Mise à jour de (null):
      │       Nouvelle version: 1.9.2
      │       Licence:          Inconnu
      │       Description:      
      │       The vendor did not supply any release notes.

Now I have to update the firmware, I tried to do the same manually but it won't update. The battery status is now "charging" but the battery still is at 0%. I waited but it seems that it stuck on 0%.

    └─Embedded Controller:
      │   Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
      │   Version précédente: 0
      │   État de mise à jour:Transient failure
      │   Erreur de mise à jour:Cannot install update when system battery is not at least 25% unless forced
      │   Dernière modification:2023-07-03 16:04
      │   GUID:               61b65ccc-0116-4b62-80ed-ec5f089ae523
      │   Drapeaux de périphérique:• Périphérique interne
      │                       • Le système nécessite une source d'alimentation externe
      │                       • Supported on remote server
      │                       • Needs a reboot after installation
      │                       • Device is usable for the duration of the update
      │                       • Mise à jour possible
      │ 
      └─ThinkPad X1 Carbon 9th / X1 Yoga 6th Embedded Controller Update:
            Nouvelle version: 0.1.33
            Remote ID:        lvfs
            Release ID:       15079
            Résumé:           Lenovo ThinkPad X1 Carbon 9th / X1 Yoga 6th Embedded Controller Firmware
    ...

ChiWei-Chen, 2023-07-04:
> Now I have to update the firmware, I tried to do the same manually but it won't update. The battery status is now "charging" but the battery still is at 0%. I waited but it seems that it stuck on 0%.

@ClubSaumon07 , thanks for the updated information.
The FW team suspected this might be a battery issue, there is not much FW can do about this phenomenon. 
Could you reach out the customer support in your region?  https://www.lenovo.com/fr/fr/contact/
Thank you for your understanding.

ClubSaumon07, 2023-07-04:
> @ClubSaumon07 , thanks for the updated information. The FW team suspected this might be a battery issue, there is not much FW can do about this phenomenon. Could you reach out the customer support [...]

Thank you, I'll get in touch with them but I am not sure this is a hardware issue. 
On the support page for my laptop there is this alert concerning the Embedded Controller update.
[image]

This is the firmware (1.6.1) update I am manually trying to install. Moreover before the update the battery was working perfectly.

Is there a way to bypass upower checks and force the update despite having a sub 25% battery level ?

ChiWei-Chen, 2023-07-05:
@ClubSaumon07 before reaching out the customer support, could you help to get this information on the battery? Thanks!
`$ upower -i /org/freedesktop/UPower/devices/battery_BAT0`

ClubSaumon07, 2023-07-05:
> @ClubSaumon07 before reaching out the customer support, could you help to get this information on the battery? Thanks! `$ upower -i /org/freedesktop/UPower/devices/battery_BAT0`

    upower -i /org/freedesktop/UPower/devices/battery_BAT0
      native-path:          BAT0
      vendor:               Sunwoda
      model:                5B10W13975
      serial:               xxxxx
      power supply:         yes
      updated:              mer. 05 juil. 2023 17:06:15 (1 seconds ago)
      has history:          yes
      has statistics:       yes
      battery
        present:             yes
        rechargeable:        yes
        state:               pending-charge
        warning-level:       none
        energy:              0 Wh
        energy-empty:        0 Wh
        energy-full:         0 Wh
        energy-full-design:  0 Wh
        energy-rate:         0 W
        voltage:             17,64 V
        charge-cycles:       179
        percentage:          0%
        capacity:            100%
        technology:          lithium-polymer
        icon-name:          'battery-caution-charging-symbolic'

ChiWei-Chen, 2023-07-06:
@ClubSaumon07 Thanks, I shared this with the battery team.

ClubSaumon07, 2023-07-06:
> @ClubSaumon07 the battery team checked the battery information and looks abnormal. We would like to get the battery pack back for further analysis. Could you reach out the customer support to get a replacement? Thanks!

I created a ticket, I'll send my computer as soon as I proceed to do a backup of my datas. 

Thank you for the help :)

adggit, 2023-07-17:
Can this be a motherboard issue?  In Lenovo forums, there are hundreds of reports that the motherboard of these models (X1 Carbon Gens 9 and 10, and X1 Yoga Gens 6 and 7) cause charging issues with similar symptoms:
- [X1 Carbon Gen 9 charging issue](https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/X1-Carbon-Gen-9-charging-issue/m-p/5127796) ~500 posts
- [X1 Carbon Gen9 cannot charge through original power port](https://forums.lenovo.com/t5/ThinkPad-X-Series-Laptops/X1-Carbon-Gen9-cannot-charge-through-original-power-port/m-p/5103170/) 100+ posts

Plus many other threads can be found on this issue via search.

ChiWei-Chen, 2023-07-21:
@adggit , this seems to be a known critical issue that is addressed for some models: [Critical BIOS Update for USB-C Port Charging - ThinkPad](https://support.lenovo.com/us/en/solutions/ht513369)