fwupd/firmware-lenovo #352: Firmware update fails to run on Thinkpad z13

kcroker, 2023-06-28:
I was having issues using devuan fwupdmgr to install the BIOS updates for my laptop.  So I spent some time upgrading fwupd from debian unstable and fwupd-efi from github, as some of the issues logs seemed to show that old versions were problematic with newer firmware updates.

I'm now using fwupd 1.9.1 and just built an fwupdx64.efi following the README.md.  I positioned it in the right place for my distribution to install it when running upgrade-grub.  I run fwupdmgr and select the first update.  That works, but when I reboot, the firmware update still fails to run.  

An error message flashes briefly, and I can see (when making a video of the boot and freezeframing with my phone):

```
fwupd-efi version 1.5
WARNING: Could not update variable status for 'fwupd- <lots of hex>': Out of Resources
WARNING: Could not set update status: Out of Resources
```

I have 186MB available on my EFI partition.  The new firmware cab are under 20MB.

Did I miss any steps here?
How can I provide diagnostic information to help diagnose and correct this issue?

superm1, 2023-06-28:
> Later 6.3.x and 6.4 kernel have real issues after waking from sleep on the z13.

JFYI - I'm using 6.3.8 on a Z13 and have no problems with this.  But again I'm on the latest firmware.

hughsie, 2023-06-28:
> Out of Resources

Could the EFI NVRAM be full with crash dumps or something?

kcroker, 2023-06-28:
@superm1 That's very encouraging, at least I know I can get things working! xD
@hughsie Let me try to figure out what that would look like.  I'm a bit scared of bricking my device by just deleting entries in the EFI NVRAM >_<

kcroker, 2023-06-28:
@hughsie That was indeed the issue, tons of dump files in efivars.  Removed them, things worked fine with the v1.5 fwupd-efi built natively from the yesterday's git clone.  Thanks guys, really appreciate the responsive feedback!  I also disabled the pstore logging in the kernel options, so hopefully this won't happen again.  

If you're receptive, maybe add a hint message to the error output (e.g. "efivars full of dump files") to future versions?

closed 2023-06-28 by kcroker