fwupd/firmware-lenovo #353: Cannot update TPM from 15.21.16430 to 15.22.16832 on X13 Gen 3 AMD

gustafla, 2023-07-07:
My device is a 21CMCTO1WW laptop (ThinkPad X13 Gen 3 AMD). I have SecureBoot enabled with my own keys. Microsoft Pluton is completely Disabled in firmware settings. TPM update from 15.21.16430 to 15.22.16832 won't work.

get-results shows this result:
```
TPM:
  Device ID:            a083ebc5138e5e071ef7270cc9a8280722cc7adf
  Previous version:     15.21.16430
  Update State:         Failed
  Update Error:         failed to update to 0: error-unsuccessful
  Last modified:        2023-07-07 16:59
  GUID:                 0717eeac-f27a-4e2a-a95d-fecf6c6cc345
  Device Flags:         • Internal device
                        • Updatable
                        • System requires external power source
                        • Needs a reboot after installation
                        • Device is usable for the duration of the update
```

I'm running following version:
```
compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.1.38-hardened1-1-hardened
runtime   org.freedesktop.fwupd         1.9.2
```

hughsie, 2023-07-07:
Out of interest, does it work if you enrol the default MS keys instead?

kmauleon, 2023-08-02:
as per ODM dev, can you please provide the following information...
BIOS version
Linux OS version
fwupdmgr get-devices results

gustafla, 2023-09-21:
I was able to install this update. The solution is a bit cumbersome, but very much doable when you find a bit of extra time for it. First of all, disable SecureBoot and check that your PC still boots. Then reboot back to firmware setup and clear the security device. After that, run `fwupdmgr update`, which now will finish successfully. Then, re-do SecureBoot setup again (for this, I use [`sbctl`](https://github.com/Foxboron/sbctl)).

Here is my `get-devices` after the update.
```
LENOVO 21CMCTO1WW
│
├─Embedded Controller:
│     Device ID:          36efb79c255f402f619fa9eb53cd659db51f2a04
│     Summary:            UEFI ESRT device
│     Current version:    0.1.27
│     Minimum Version:    0.1.27
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               66d6a3ef-a771-4302-9cd0-d062c79c5ef2
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          91c9a317b30acf7c0cd967f2ce6925d518ef19a7
│     Current version:    10.22
│     Vendor:              (USB:0x174F)
│     Serial Number:      0001
│     GUID:               0cf3aef8-c29e-5cb8-997c-d498ed52b666 ← USB\VID_174F&PID_1812
│     Device Flags:       • Updatable
│   
├─MZVL21T0HCLR-00BL7:
│     Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
│     Summary:            NVM Express solid state drive
│     Current version:    EL2QGXA7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S64PNX0TC96079
│     GUIDs:              bec63ed7-a95f-54fe-b8cc-8e9fee64ba5a ← NVME\VEN_144D&DEV_A80A
│                         310f81b5-6fce-501e-acfb-487d10501e78 ← NVME\VEN_144D&DEV_A80A&SUBSYS_144DA801
│                         c0e40d86-e47a-57fe-8ed1-453e6d83a586 ← SAMSUNG MZVL21T0HCLR-00BL7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─System Firmware:
│     Device ID:          d96de5c124b60ed6241ebcb6bb2c839cb5580786
│     Summary:            UEFI ESRT device
│     Current version:    0.1.35
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              6ab943b7-f4d4-aaa1-2f40-cb03a0c8cf3c
│                         230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│   
├─TPM:
│     Device ID:          a083ebc5138e5e071ef7270cc9a8280722cc7adf
│     Summary:            UEFI ESRT device
│     Current version:    15.22.16832
│     Minimum Version:    15.22.16832
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               0717eeac-f27a-4e2a-a95d-fecf6c6cc345
[... 81 lines cut ...]
│     Minimum Version:    1
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               75d78aef-5e7f-0afe-b68a-aef4c52ccddd
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          11a7fbdd09ce583e58b899660fe10f147fd155a2
      Summary:            UEFI ESRT device
      Current version:    16859233
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               98700de8-2296-4294-97a6-04b0b6b385eb
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
```

gustafla, 2023-09-21:
I think this issue is probably a non-issue and may be closed now, since there is a reasonable way to get this update installed. What do you think?

kmauleon, 2023-11-03:
@VeldoraTheDragon can you please show complete results of fwupdmgr get-devices

kmauleon, 2023-11-07:
@VeldoraTheDragon forwarded to ODM... thank you very much...will let you know once i get an update

kmauleon, 2023-11-07:
@gustafla let's close this one... thank you very much!
@VeldoraTheDragon let's track this here #378 since this is handled by different TPM owner.  thank you very much!