fwupd/firmware-lenovo #354: ThinkPad L480 failed to run update on reboot 0.1.57 -> 0.1.58

Grafcube, 2023-07-09:
**Describe the bug**
I'm trying to install a firmware update for my Thinkpad L480 but it doesn't apply. It downloads the update correctly and prompts me to reboot. After that it boots into fwupd-efi. Then it shows a progress bar and says that updates are being installed. And then finally it reboots back into Linux. There is no indication that anything went wrong until I check and it shows that the update is still available to install. I tried again after disabling secure boot but nothing changed. I'm using `DisableShimForSecureBoot=true` and sbctl to sign `fwupdx64.efi` if that's relevant.

**Steps to Reproduce**
- Start update through either `doas fwupdmgr update` or Gnome Software or Gnome Firmware.
- Restart when prompted.
- Wait for it to complete and reboot.
- Find that update was not applied.

**Expected behavior**
Update is installed.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.2
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.4.2-arch1-1
runtime   org.freedesktop.fwupd         1.9.2
```

```sh
pacman -S fwupd
```

<details>

<summary>**fwupd device information**</summary>

```shell
$ fwupdmgr get-devices --show-all-devices
LENOVO 20LTS3C742
?
??256GB NVME EVM:
?     Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
?     Summary:            NVM Express solid state drive
?     Current version:    V0513A0
?     Vendor:             Silicon Motion, Inc. (NVME:0x126F)
?     Serial Number:      NVSI256G09202256768
?     GUIDs:              eb4c6074-9dc2-57ae-bd43-1119ad6080f5 ? NVME\VEN_126F&DEV_2263
?                         7660210c-6f7e-50ad-847c-f1d9b83bf29b ? NVME\VEN_126F&DEV_2263&REV_03
?                         165a89d1-acc2-51c7-92b9-300207a5409c ? NVME\VEN_126F&DEV_2263&SUBSYS_126F2263
?                         1e96d448-72c0-527c-9236-cd710d2119af ? NVME\VEN_126F&DEV_2263&SUBSYS_126F2263&REV_03
?                         3f786d05-414c-5cd5-8920-80171fca2ed5 ? 256GB NVME EVM
?     Device Flags:       ? Internal device
?                         ? Updatable
?                         ? System requires external power source
?                         ? Needs a reboot after installation
?                         ? Device is usable for the duration of the update
?
??Core? i5-8350U CPU @ 1.70GHz:
?     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
?     Current version:    0x000000f2
?     Vendor:             Intel
?     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ? cpu
?                         30249f37-d140-5d3e-9319-186b1bd5cac3 ? CPUID\PRO_0&FAM_06
?                         561403e8-143a-5071-ab09-bf5e4c146983 ? CPUID\PRO_0&FAM_06&MOD_8E
?                         54bdc429-0179-5019-9cb6-8a343324c97c ? CPUID\PRO_0&FAM_06&MOD_8E&STP_A
?     Device Flags:       ? Internal device
?
??Intel Management Engine:
?     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
?     Summary:            UEFI ESRT device
[... 204 lines cut ...]
├── intel-ucode.img
├── loader
│   ├── entries
│   ├── entries.srel
│   ├── loader.conf
│   └── random-seed
└── vmlinuz-linux

9 directories, 14 files
```

**Additional questions**

- Operating system and version:
  - Arch Linux
- Have you tried rebooting?
  - Yes
- Is this a regression?
  - This is the first time I've gotten a firmware update on this device.
- Are you using an NVMe disk?
  - Yes
- Is secure boot enabled?
  - Yes, but I also tried to update after disabling it. There was no change.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
  - Boot Order Lock is turned off

hughsie, 2023-07-09:
Sounds like the firmware isn't updating the ESRT table. @mrhpearson any ideas for the L480?

mrhpearson, 2023-07-09:
@Grafcube - can you confirm from the BIOS setup screen (F1 during early boot) which version the FW itself thinks is installed please?

Thanks
Mark

Grafcube, 2023-07-10:
I'm not sure where to look. This is what I see in the F1 screen. There's nothing showing 0.1.57 or 0.1.58 anywhere.

[image]

mrhpearson, 2023-07-10:
Right - so your update failed as you're still on BIOS 1.45 :) This isn't an ESRT table update - it s a BIOS update failure.

1.45 to 1.57 is a pretty big jump - I wonder if that's the problem and we should try and do it in smaller steps.

Unfortunately my password to the FW server just expired and I need to get it reset so I can go check the update notes. I'll get that done and have a look - can you nag me in a couple of days if I don't update please? 
(As a note - if you still have Windows on the machine then worth trying the update from there too). This platform is old and it doesn't look like it's had good FW update support sadly.

Grafcube, 2023-07-10:
> Right - so your update failed as you're still on BIOS 1.45 :) This isn't an ESRT table update - it s a BIOS update failure.

That's strange. According to Gnome Firmware, I'm on 0.1.57 and the available update is 0.1.58. That's why I was confused with the BIOS screen, since I didn't see 1.45 anywhere else.

[image]

> 1.45 to 1.57 is a pretty big jump - I wonder if that's the problem and we should try and do it in smaller steps.
>
> Unfortunately my password to the FW server just expired and I need to get it reset [...]

I don't have Windows on the laptop so I'll just wait for the fix. Thanks for the help!

Grafcube, 2023-07-13:
@mrhpearson any updates on this?

mrhpearson, 2023-07-13:
Thanks for the nudge - I'm got my account reset and then forgot to go look for FW. I'm out today so not on the VPN but I'll aim to do it this evening. Ping me if I forget...inbox is a bit crazy.

Grafcube, 2023-07-14:
@mrhpearson Here's another ping like you requested (I don't know when your evening is).

mrhpearson, 2023-07-14:
Thanks - I need the nudge! Bad news is I can't find this platform on the FW server that I have access to - which is strange :(

@ChiWei-Chen - is this something you can help with? I think we need to get an older version of BIOS for the L480 so @Grafcube can do an incremental update as jumping to 1.57 isn't working. I'm not sure where to find it - it's not in the normal places that I can find.

Mark

mrhpearson, 2023-07-18:
Hi 
Got access to the internal server where this is stored, and it looks like the metainfo is wrong. The FW you are trying to install is actually 1.46 (not 1.57). So this should be a normal update.
Checking the stats there are a lot of successful updates - so not sure what is going on. I've asked the FW team to review for any insight. Sorry for the distraction - the whole incorrect numbering threw me for a loop.
Potentially could try the iso file on the Lenovo support site (from a USB) - but the fwupdmgr update _should_ work
Mark

Grafcube, 2023-07-19:
> but the fwupdmgr update _should_ work

I'm not sure what you mean by that. The update still doesn't work though.

mrhpearson, 2023-07-19:
FW team the mismatch in numbers (1.46 vs 1.58) is expected - apparently it's been like that for a while.

I've asked for details on debugging the issue - having the progress bar but not the update seems peculiar.

mrhpearson, 2023-07-20:
Hi @Grafcube 

FW team updated that version 1.46 and 1.45 are the same. I'm asking for clarification on why they did a new release with just a version number update because it's most peculiar (and confusing); but from a system point of view I think you are safe to ignore this update (though I'm assuming it keeps getting offered by fwupd? That will be annoying).

My guess is this happened as part of getting the FW up on LVFS and just wasn't done very cleanly. With the different version in the metainfo it's a bit of a mess really :(

Mark

Grafcube, 2023-07-20:
If it's safe to ignore then I guess it's fine. That little indicator in Gnome Software isn't going to kill me or anything.

Thanks for all the help!

Grafcube, 2023-07-29:
I got a firmware update to 0.1.64 and it seems to have been installed without any problems. It didn't unlock my disk automatically on restart and I had to manually enter the passphrase but I just used systemd-cryptenroll to fix that. It seems like this has been resolved.

closed 2023-07-29 by Grafcube