fwupd/firmware-lenovo #356: Lenovo Thinkpad P15s - Bettery Firmware Update 1.0.8523 -> 1.1.10246 - Battery gets too hot while charging

ethaning, 2023-07-21:
Hi, I updated my system firmware a few days ago. Since the update, my laptop gets super hot while charging, especially around the battery area. 

When I run the [`sensors`](https://github.com/lm-sensors/lm-sensors) command, some of the temperatures reported are between 80 and 100 degrees celsius while the power supply is plugged in. In general, the laptop is noticeably and concerningly hotter than before these upgrades.  

Any advice?

---

<details>

<summary> fwupdmgr get-devices </summary>

```
20W6CTO1WW
│
├─Battery:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    1.1.10246
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               8ac4c4f4-963c-40c4-a102-59779bbd17ac
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.42
│     Minimum Version:    0.1.42
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               2a5d161c-0ba2-4136-94b3-c027f0a7e5fc
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          3fa281ddf80d8a06b8ee5d8beb48d38ee95a9627
│     Current version:    59.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:              17d214a8-3e31-5f50-94b5-45bf35278459
│                         f8d483f9-9284-51b6-9363-7e6d59c40a97
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    240.42.2235
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               5d19c74d-6a93-489a-a5ed-65cb7f1e13bf
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─MZVLB1T0HBLR-000L7:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
[... 132 lines cut ...]

<summary>fwupdmgr update</summary>

### There appear to be no futher updates available...

```
Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Battery
 • Embedded Controller
 • Intel Management Engine
 • MZVLB1T0HBLR-000L7
 • System Firmware
```

</details>

---

hughsie, 2023-07-21:
If you downgrade the firmware does the battery behaviour return to normal? @mrhpearson any ideas?

ethaning, 2023-07-24:
I am not able to downgrade the battery firmware using `fwupdmgr`. Any ideas on what else I could try?

The issue is still present; battery is extremely hot while computer is charging...

```
$ fwupdmgr get-history 

# ...

├─Battery:
│ │   Device ID:          f95c9218acd12697af946874bfe4239587209232
│ │   Previous version:   1.0.8523
│ │   Update State:       Success
│ │   GUID:               8ac4c4f4-963c-40c4-a102-59779bbd17ac
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─  New version:      1.1.10246
│       Remote ID:        lvfs
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│     

# ...

$fwupdmgr downgrade f95c9218acd12697af946874bfe4239587209232
No downgrades for Battery: current version is 1.1.10246: 1.1.10246=same

$ fwupdmgr downgrade 
Choose a device:
0.	Cancel
1.	f95c9218acd12697af946874bfe4239587209232 (Battery)
2.	2292ae5236790b47884e37cf162dcf23bfcd1c60 (Embedded Controller)
3.	349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
4.	04e17fcf7d3de91da49a163ffe4907855c3648be (MZVLB1T0HBLR-000L7)
5.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
1
No downgrades for Battery: current version is 1.1.10246: 1.1.10246=same
```

ChiWei-Chen, 2023-07-26:
hi @ethaning , @hughsie , @mrhpearson is on leave this week, I'll ask FW team to help with this. (LO-2571 for my tracking)
Could you try using the pinhole reset and see how it works?

ethaning, 2023-08-01:
Hey, Thanks for the response. 

The situation seems to have slowly improved over the last days. I can't say for certain whether my laptop is still much hotter than it was before the updates... But it still is the case that the laptop heats up noticeably more when charging / plugged in, compared with decharging / not plugged in. This is not something I observed before the firmware update and I believe it is new behaviour since the firmware update. 

Here is some output from a few days ago. Sorry for the delay on my side - I've been on holidays lately.

<details>

<summary>sensors</summary>

$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +58.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +58.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +56.0°C  (high = +100.0°C, crit = +100.0°C)

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         3.00 A  (max =  +0.00 A)

thinkpad-isa-0000
Adapter: ISA adapter
fan1:        6134 RPM
CPU:          +58.0°C  
GPU:          +60.0°C  
temp3:        +42.0°C  
temp4:         +1.0°C  
temp5:        +49.0°C  
temp6:        +58.0°C  
temp7:        +58.0°C  
temp8:            N/A  

BAT0-acpi-0
Adapter: ACPI interface
in0:          12.89 V  

ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max = +13.20 V)
curr1:         3.00 A  (max =  +3.21 A)

nouveau-pci-0100
Adapter: PCI adapter
temp1:            N/A  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +67.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +66.8°C  (low  = -273.1°C, high = +83.8°C)
                       (crit = +84.8°C)
Sensor 1:     +66.8°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +76.8°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +58.0°C  (crit = +128.0°C)

</details>

<details>

<summary>upower</summary>

# I unplugged the charging cable, then plugged it in again a few seconds later.

$ upower --monitor
Monitoring activity from the power daemon. Press Ctrl+C to cancel.
[14:16:43.234]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:43.234]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:43.234]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:43.234]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:43.234]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:43.234]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:43.234]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:43.694]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:43.694]  device changed:     /org/freedesktop/UPower/devices/line_power_AC
[14:16:43.694]  device changed:     /org/freedesktop/UPower/devices/line_power_AC
[14:16:44.148]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:44.148]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:44.148]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:44.148]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:44.149]  device changed:     /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o002
[14:16:44.149]  device changed:     /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o002
[14:16:49.013]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:49.013]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:49.013]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:49.013]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:49.013]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:49.013]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:49.013]  device changed:     /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o002
[14:16:49.013]  device changed:     /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o002
[14:16:49.271]  device changed:     /org/freedesktop/UPower/devices/line_power_AC
[14:16:49.271]  device changed:     /org/freedesktop/UPower/devices/line_power_AC
[14:16:50.432]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:50.432]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:50.432]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:50.432]  device changed:     /org/freedesktop/UPower/devices/battery_BAT0
[14:16:50.432]  device changed:     /org/freedesktop/UPower/devices/line_power_ucsi_source_psy_USBC000o002

</details>

Was not able to easily capture the whole output of `sudo fwupdtool get-devices --verbose`. Will try again later when I have more time. 

Thanks for the help so far!

ChiWei-Chen, 2023-08-08:
@ethaning ,
thanks, I'll share this with the team.
Please keep me posted if can see the issue again.

ethaning, 2023-08-15:
@ChiWei-Chen Hey, thanks for responding. 

I just want to let you know that I have been experiencing unusually high temperatures on my laptop during the last few days while my laptop has been charging.

Here is an example from just moments ago of current temperatures:

<details>

<summary>sensors</summary>

$ sensors
ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max = +13.20 V)
curr1:         3.00 A  (max =  +3.21 A)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +82.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +82.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +70.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +74.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +71.0°C  (high = +100.0°C, crit = +100.0°C)

thinkpad-isa-0000
Adapter: ISA adapter
fan1:        4658 RPM
CPU:          +72.0°C  
GPU:          +65.0°C  
temp3:        +44.0°C  
temp4:         +1.0°C  
temp5:        +54.0°C  
temp6:        +72.0°C  
temp7:        +72.0°C  
temp8:            N/A  

BAT0-acpi-0
Adapter: ACPI interface
in0:          12.94 V  

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +72.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +64.8°C  (low  = -273.1°C, high = +83.8°C)
                       (crit = +84.8°C)
Sensor 1:     +64.8°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +67.8°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +72.0°C  (crit = +128.0°C)

</details>

My laptop never used to get this hot. It is mildly uncomfortable to touch, due to the heat. Do you have any advice?

Thanks

ChiWei-Chen, 2023-08-16:
@ethaning , I've shared this with the team, but most members are on holiday this week. I'll keep you updated if I got any reply.

ethaning, 2023-08-21:
@ChiWei-Chen Hey, thanks for responding again. 

Just wanted to bump this thread. Issue is persisting. Here is a sample of my laptop right now, currently charging...

<details>

<summary>sensors</summary>

$ sensors
ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max = +13.20 V)
curr1:         3.00 A  (max =  +3.21 A)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +83.0°C  

thinkpad-isa-0000
Adapter: ISA adapter
fan1:        6185 RPM
CPU:          +86.0°C  
GPU:          +71.0°C  
temp3:        +46.0°C  
temp4:         +1.0°C  
temp5:        +58.0°C  
temp6:        +86.0°C  
temp7:        +86.0°C  
temp8:            N/A  

BAT0-acpi-0
Adapter: ACPI interface
in0:          12.24 V  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +87.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +78.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +90.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +77.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +82.0°C  (high = +100.0°C, crit = +100.0°C)

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         3.00 A  (max =  +0.00 A)

nvme-pci-0400
Adapter: PCI adapter
Composite:    +71.8°C  (low  = -273.1°C, high = +83.8°C)
                       (crit = +84.8°C)
Sensor 1:     +71.8°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +74.8°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +86.0°C  (crit = +128.0°C)

</details>

ChiWei-Chen, 2023-08-22:
@ethaning , thanks for the updated information.
The FW team wants to know did you observe this temperature during heavy loading? Or it is light (normal) usage condition just in idle condition?

ethaning, 2023-08-22:
@ChiWei-Chen The laptop was not idle during this time. I was using it. Usually, while using my laptop, my system is under a rather medium load (Rather high RAM usage and CPU usage fluctuating often from low to high). I will take note of my system usage next time I observe these high temperatures. 

Also, what do you mean by the "duplicate instruction from the system boot"? Maybe you could link me to some instructions on how I can provide this information for you? Thanks :)

ChiWei-Chen, 2023-08-23:
@ethaning , thanks, I've shared with the team, I'll keep you posted.

ChiWei-Chen, 2023-08-29:
@ethaning ,
The thermal team asked some questions, could we have your comment for them?

1. Is this machine using internal iGPU or dGPU(NVIDIA graphic)?

2. Could you provide "top" output when seeing high temperature? 
The team suspects something running in the system. Since 20% CPU usage does NOT mean CPU power consumption is low.
It is suspected the hot area(battery area) means palm rest, like heat comes from the planar via boss.

3. Is the machine in performance or balance mode, is it with AC or DC mode?

Thanks!

ethaning, 2023-09-06:
@ChiWei-Chen Hey, sorry for the delay in my response. I've not been experiencing the symptoms so much in the last week.

But, after changing some of my computer configuration, I am experiencing the symptoms again. I'll describe the details below, and answer your questions as best as I can.

1. My machine has both internal and dedicated GPU. About a week ago, I reinstalled my NVIDIA drivers, and I set the GPU to the performance mode from the Nvidia settings ( see pic). I believe that using the Nvidia GPU in performance mode  results in a heavier usage of the dedicated GPU over the internal gpu?
[image]

2. Right now, I am experiencing high temperatures when touching the laptop. the machine is in a rather idle state, but it is plugged in and charging. It is the output from the top command: 

<details>

<summary>top</summary>

$ top

top - 10:42:21 up  1:45,  1 user,  load average: 3.89, 4.14, 4.04
Tasks: 578 total,   1 running, 577 sleeping,   0 stopped,   0 zombie
%Cpu(s):  3.3 us,  1.3 sy,  0.0 ni, 95.3 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :  15705.0 total,    297.4 free,  12575.8 used,   2831.8 buff/cache
MiB Swap:  15625.0 total,  15310.0 free,    315.0 used.   2148.7 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                             
  30002 ebae      20   0 2863792 342628 105884 S   6.3   2.1   1:49.74 Isolated Web Co                                                                                                                     
  11193 ebae      20   0   25.2g 210120  95360 S   5.3   1.3   7:10.45 Xorg                                                                                                                                
  19092 ebae      20   0 1037584  45656  29632 S   4.6   0.3   4:44.93 gnome-system-mo                                                                                                                     
  73290 ebae      20   0 1132.9g 227704 118104 S   4.3   1.4   0:21.74 chrome                                                                                                                              
  11534 ebae      20   0 4622656 381704  94360 S   2.6   2.4  16:41.57 gnome-shell                                                                                                                         
  48369 ebae      20   0 1037028  72476  40064 S   2.3   0.5   0:03.06 terminator                                                                                                                          
  11152 ebae       9 -11 3443468  33488  22596 S   1.7   0.2   2:22.24 pulseaudio                                                                                                                          
  13131 ebae      20   0   32.7g 292336 150580 S   1.7   1.8  11:15.99 chrome                                                                                                                              
  30170 ebae      20   0  327536  44608  33152 S   1.3   0.3   0:19.22 Utility Process                                                                                                                     
    844 root     -51   0       0      0      0 S   1.0   0.0   2:34.32 irq/188-nvidia                                                                                                                      
  12837 ebae      20   0   33.2g 647940 363944 S   1.0   4.0   3:50.46 chrome                                                                                                                              
   2503 root      20   0 9778.0m 676652  25472 S   0.7   4.2   1:14.42 java                                                                                                                                
   3314 root      20   0 9847.8m 623916  25216 S   0.7   3.9   1:20.15 java                                                                                                                                
   4163 root      20   0 5683760 195864   7168 S   0.7   1.2   0:51.84 java                                                                                                                                
  13135 ebae      20   0   32.4g 108520  74232 S   0.7   0.7   2:20.09 chrome                                                                                                                              
  29541 ebae      20   0 4012396 689072 237516 S   0.7   4.3  15:26.61 firefox                                                                                                                             
     15 root      20   0       0      0      0 I   0.3   0.0   0:09.07 rcu_preempt                                                                                                                         
   1671 root      20   0 1316428  71548  12928 S   0.3   0.4   0:43.43 tailscaled                                                                                                                          
   4031 systemd+  20   0 3183516 149584  66944 S   0.3   0.9   0:30.70 beam.smp                                                                                                                            
   4156 root      20   0 9770.2m 984.5m  17408 S   0.3   6.3   1:11.17 java                                                                                                                                
[... cut ...]
     25 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/1:0H-events_highpri                                                                                                         
     26 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/2  

</details>

3. Since I updated my Nvidia drivers, I also put the machine in performance mode, and since then I have not been experiencing the issues so much. right now, while I'm experiencing the issue, the machine is in Balance mode. Also, I think my machine using AC power; using the upower -d command, I only see info relating to AC devices.

Please let me know if you have any advice. 
Thanks

ethaning, 2023-09-07:
@ChiWei-Chen Hey, one more thing, which may or may not be related...

Currently, my laptop is plugged in, but it continues to discharge. The charging indicator on the side of the laptop is not on. Here is the output from the acpi command:

```
$ acpi -b
Battery 0: Discharging, 22%, 00:18:00 remaining
```
This is the 3rd or 4th time that this has happened. The first time it happened was maybe 1 or 2 weeks after the initial firmware update, but I didn't think much of it. But it has happened a few times now. 

When I restart the laptop, then the laptop is able to charge normally. 

Any advice?

ChiWei-Chen, 2023-09-07:
@ethaning thank you, I shared this with FW team, and I'll back to you once I get any feedback.

ethaning, 2023-09-20:
@ChiWei-Chen Hey there, any update? Thanks

ChiWei-Chen, 2023-09-21:
@ethaning sorry for the late feedback. Just got a few questions from FW team.
According to the system monitor and top above, the CPU usage > 20% seems very high, this could be the problem, they're checking.

Could you confirm what power mode is used? performance, balanced, or power saver?

ethaning, 2023-09-21:
@ChiWei-Chen Thanks for the update. 

I am using performance mode atm. While using this power mode, the laptop does not get as hot as when it is in balanced mode (but it still does get rather hot - hotter than before the firmware update from weeks ago).

ChiWei-Chen, 2023-09-22:
@ethaning thanks.
The team wants to know if have you tried rolling back all firmware except the battery firmware?
With that condition, still hot during charge?