fwupd/firmware-lenovo #357: ThinkPad T14s Gen 3 AMD (21CQ) - Can't upgrade EC firmware 1.15 to 1.25 (or 1.127)

j0ex10, 2023-07-27:
I am running Fedora 38 and I received the system updates regularly from gnome software. I recently received a BIOS update that installed fine (1.31). But it's been a while since I am not able to upgrade the Embedded Controller firmware.

fwupdmgr get-devices:
```
LENOVO 21CQCTO1WW
│
├─ELAN0678:00 04F3:3195:
│     Device ID:          f8bebdac7d9634f3890f6f6cf3eec0908c2c700b
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              4e8cda57-37ba-5dc6-b114-7610e38714e2 ← HIDRAW\VEN_04F3&DEV_3195
│                         cc456594-6f7e-5d1b-82a6-b4558a813152 ← HIDRAW\VEN_04F3&DEV_3195&MOD_0006
│                         4b808d27-c833-50cc-8177-9a401f5c08fe ← ELANTP\ICTYPE_14
│                         d15e804c-570f-511b-803a-b38a4eddae29 ← ELANTP\ICTYPE_14&MOD_0006
│                         296e64f7-9e08-5717-bff2-651e41b0477e ← ELANTP\ICTYPE_14&MOD_0006&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─AMD Ryzen 7 PRO 6850U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a404102
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│ │                       52f8f9af-1ca9-5352-bef4-ceb232c888a5 ← CPUID\PRO_0&FAM_19&MOD_44
│ │                       e94372a3-3ffb-5d1c-a579-c415b7313e52 ← CPUID\PRO_0&FAM_19&MOD_44&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │     Device ID:        fb2f6a4f66c638829126eef33d281666a63c1e64
│ │     Current version:  113-REMBRANDT-X37
│ │     Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │     GUIDs:            817e27b3-d1bb-591a-a1cc-e039d0a60be6 ← PCI\VEN_1002&DEV_1681
│ │                       f87061f4-ca92-521f-8ee6-bd240356cce8 ← PCI\VEN_1002&DEV_1681&SUBSYS_17AA50B4
│ │     Device Flags:     • Internal device
│ │   
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Summary:          Microcontroller used within CPU/APU program 4
│       Current version:  69.55.0
│       Vendor:           Advanced Micro Devices, Inc.
│       GUID:             79307ae6-a2ea-52e1-bf56-6abbaf3547ad ← /sys/devices/platform/AMDI0007:00
│       Device Flags:     • Internal device
│     
├─EM05-G:
│     Device ID:          593e609e6f7b35b0fe53f583b7f5548094b04c4f
│     Summary:            Mobile broadband device
│     Current version:    EM05GFAR07A07M1G_01.002.01.002
│     Vendor:             Quectel (USB:0x2C7C)
│     GUIDs:              83020735-53b8-5ce8-957b-95e9245a64ee ← USB\VID_2C7C&PID_030A&REV_0200&NAME_EM05GFA
│                         0322d845-8497-5cbe-bfd5-2c2a6c9ea129 ← USB\VID_2C7C&PID_030A&REV_0200
│                         df71f261-0f4c-55d5-8c0c-e6d2840feee5 ← USB\VID_2C7C&PID_030A
│                         22ae45db-f68e-5c55-9c02-4557dca238ec ← USB\VID_2C7C
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Unsigned Payload
│   
├─Embedded Controller:
│     Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
│     Summary:            UEFI ESRT device
│     Current version:    0.1.15
│     Minimum Version:    0.1.15
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               66d6a3ef-a771-4302-9cd0-d062c79c5ef2
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
[... 182 lines cut ...]
Devices with no available firmware updates: 
 • ELAN0678:00 04F3:3195
 • EM05-G
 • Fingerprint Sensor
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.15 to 0.1.27?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Modify the initialization sequence of expander.                            ║
║ • Add password protection for MPS2845A.                                      ║
║                                                                              ║
║ 21CQCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
```
When I run the update it all looks good, the device reboots and shows the progress bar going 0-100%, after which I see another "Lenovo" screen saying "Flashing firmware...!" but when it boots back to the OS the firmware version is still the old 1.15, BIOS also shows the same (old) version.

ChiWei-Chen, 2023-08-04:
Thanks, I'll give you have if any progress from FW team (LO-2582 for my internal tracking)

ChiWei-Chen, 2023-08-04:
@j0ex10 , the FW team replied this is a known issue at present:

> Users may need to update ECFW to 1.22 or later by shell, then users can use LVFS normally. The team is working in progress preparing tips for the user's reference, the application document is being reviewed, and it may take some time to complete.

I'll keep you posted.

j0ex10, 2023-08-04:
thank you for the update,  when the team will have EC FW standalone packages I am happy to give it a go

j0ex10, 2023-08-25:
Hi @ChiWei-Chen 
Thanks for the intructions provided, it worked fine!
After upgrading to the provided package I was able to upgrade to the newest version via fwupd.

However as a reference for people who could find this issue in the future, the guide assumes you are using windows.
Not all linux distros come with a efi shell binary already available. 
You can get it from tianocore's edk2 repo (https://github.com/tianocore/edk2/blob/UDK2018/EdkShellBinPkg/FullShell/X64/Shell_Full.efi)
Download it, rename it as _bootx64.efi_ and place it under _/boot/efi/_ on a fat32 formatted usb drive
Copy the unzipped package for your model on the usb (see guide)
Then keep following the official guide from step 5

closed 2023-08-25 by j0ex10