fwupd/firmware-lenovo #358: Firmware missing for most devices on Thinkpad T14 Gen 4 Intel

c0deaddict, 2023-08-09:
```
$ fwupdmgr get-updates
Devices with no available firmware updates:
 • ELAN0676:00 04F3:3195
 • Integrated Camera
 • Intel Management Engine
 • MZVL21T0HDLU-00BLL
 • Prometheus IOTA Config
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Prometheus
 • UEFI dbx
No updates available```

hughsie, 2023-08-09:
@mrhpearson that does seem like a lot of ESRT devices!

hughsie, 2023-08-09:
@c0deaddict that's on AC power and with a fully charged battery, right?

c0deaddict, 2023-08-09:
Yes, that is correct. Although not *fully* charged, TLP charges it up until around 80%, but I guess that counts as fully charged :)

c0deaddict, 2023-08-09:
> ESRT

What is an ESRT device?

mrhpearson, 2023-08-09:
ESRT devices - these are the devices that the BIOS says are available and enumerates in the ESRT table (and you can check under /sys/firmware/efi/esrt). I don't have one of these platforms on hand to check what they all are - but we've been moving more and more to EFI capsule updates but that is quite a long list...I'll have to see what is going on there.

For the origin of this ticket and why no FW is available - my guess is it's because we've not completed Linux enablement of this platform yet. Certification is done and preload testing is ongoing. The FW team like to use our preload image for doing their testing for LVFS releases so it does gate LVFS.

If you brought your system with Windows (as Linux isn't available yet) then you can update FW from that. FW for this platform will be on LVFS for future updates.

Let me know if any objections to closing this.

c0deaddict, 2023-08-10:
Thanks for your explanation! Unfortunately I don't have a Windows license with the laptop, but I can wait for the certification process to complete to install firmware updates.

Ticket can be closed :heavy_check_mark:

closed 2023-08-10 by c0deaddict