fwupd/firmware-lenovo #359: ThinkPad P14sG3 - Failed to update embedded controller & system firmware

ben-ranford, 2023-08-13:
Embedded controller fails to update to 0: authentication signing error.
System firmware fails to update to 117474352: authentication signing error

rolled both into one thread as they're on the same device, though i'm happy to make another issue if they need to be split for internal tracking purposes :)

###get-history
`LENOVO 21AK21AK0S
│
├─Prometheus:
│ │   Device ID:          0d5d05911800242bb1f35287012cdcbd9b381148
│ │   Previous version:   10.01.3333223
│ │   Update State:       Success
│ │   Last modified:      2023-08-13 07:58
│ │   GUID:               659f7e45-8d45-528d-b3c7-0695eed055f6
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Reported to remote server
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─Prometheus:
│       New version:      10.01.3478575
│       Remote ID:        lvfs
│       Release ID:       10696
│       Summary:          Firmware for the Synaptics Prometheus Fingerprint Reader device
│       License:          Proprietary
│       Size:             425.5 kB
│       Created:          2021-06-09
│       Urgency:          Medium
│       Vendor:           Synaptics
│       Duration:         2 seconds
│       Description:      
│       New features and enhancements:
│       
│       • Fix an authentication issue on FOP project.
│     
├─Embedded Controller:
│ │   Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│ │   Previous version:   0
│ │   Update State:       Failed
│ │   Update Error:       failed to update to 0: authentication signing error
│ │   Last modified:      2023-08-13 08:01
│ │   GUID:               7ddc47a5-a390-4571-a943-2636de3504d2
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad T14 Gen 3 / P14s Gen 3 / T16 / P16s:
│       New version:      0.1.15
│       Remote ID:        lvfs
│       Release ID:       27928
│       Summary:          Lenovo ThinkPad T14 Gen 3 / P14s Gen 3 / T16 / P16s Embedded controller Firmware
│       License:          Proprietary
│       Size:             1.2 MB
│       Created:          2023-02-09
│       Urgency:          Medium
│       Vendor:           Lenovo
│       Release Flags:    • Is upgrade
│       Description:      
│       Embedded controller version 1.15.Optional update by battery request.
│     
└─System Firmware:
  │   Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
  │   Previous version:   0.0.162
  │   Update State:       Failed
  │   Update Error:       failed to update to 117474352: authentication signing error
  │   Last modified:      2023-08-13 08:01
[... 329 lines cut ...]
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          a083ebc5138e5e071ef7270cc9a8280722cc7adf
      Summary:            UEFI ESRT device
      Current version:    3
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               240cf38f-3985-42c5-b77b-8e4bdaf0fcc1
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
`
###extra details
- running RHEL 9.5
- reboot has been tried, same error in Windows
- think this device may be a late QC/Engineering Sample, which may be related to the error (XXXX where model should be on display bezel, non-existent S/N when searched on website and SKU that doesn't exist)
- secure boot is enabled
- boot order lock is off
- disk is NVMe

ben-ranford, 2023-10-26:
still an issue. bump.

mrhpearson, 2023-11-01:
Hi Ben,
You're on old FW and have to tread very carefully if on a pre-production system (which with XXXX on the bezel you almost certainly are). It's very easy to brick these units (I have done it myself multiple times).
Where did you get the system? They're not supposed to be for general use?
Not sure we're going to be able to officially support this (so I'm closing the ticket) but ping me off forum (mpearson-lenovo at squebb dot ca) and we can figure out what HW you've got and if there's a way to bring it up to date.

closed 2023-11-01 by mrhpearson