fwupd/firmware-lenovo #360: Unable to install Intel Management Engine: "Failed to recover ME firmware"

latener, 2023-08-15:


riscidotorg, 2023-08-15:
I am experiencing a similar issue on a Thinkpad P1 Gen5. Fwupdmgr fails to update "Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Consumer ME Firmware, for Hybrid Nonvpro" from reported version 1.27.2176 -> 257.27.2176.

Note: the firmware update description references version [16.1.27.2176](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3JRP.firmware) which is the current version reported by bios.

fwupd device info:
LENOVO 21DC002QUS
│
├─ELAN0686:00 04F3:320D:
│     Device ID:          522cb97965758d9b1309d3217dfd41673f410e15
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0003
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              7e8dad8a-0b0e-527b-8810-d5dbca8804d7 ← HIDRAW\VEN_04F3&DEV_320D
│                         71de889a-ea8d-5c66-a0fb-7e11e860cd3d ← HIDRAW\VEN_04F3&DEV_320D&REV_00
│                         74e88f64-4d6b-5048-8199-fdcc7833b48a ← HIDRAW\VEN_04F3&DEV_320D&MOD_003D
│                         4b808d27-c833-50cc-8177-9a401f5c08fe ← ELANTP\ICTYPE_14
│                         e25cd415-0431-502a-9077-3c409b007897 ← ELANTP\ICTYPE_14&MOD_003D
│                         8cb8f87c-48a3-51e4-ad02-8dd596adddc7 ← ELANTP\ICTYPE_14&MOD_003D&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─12th Gen Intel Core™ i7-12700H:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0000042c
│ │   Vendor:             Intel
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│ │                       ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│ │                       3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│ │   Device Flags:       • Internal device
│ │ 
│ └─USB4 host controller:
│       Device ID:        0cc78012cc50f125c9a6752cad690c351fc7bf45
│       Summary:          Unmatched performance for high-speed I/O
│       Current version:  38.00
│       Vendor:           Lenovo (THUNDERBOLT:0x0109)
│       GUIDs:            0baf15fd-e7f5-5d90-b48a-24e756b578de ← THUNDERBOLT\VEN_0109&DEV_2112
│                         eb242131-d645-5dc8-b499-a9b21f001362 ← THUNDERBOLT\VEN_0109&DEV_2112&REV_00
│                         e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│       Device Flags:     • Internal device
│                         • System requires external power source
│                         • Signed Payload
│     
├─Alder Lake-P Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              eaad9970-8e4d-56da-88ab-41a8c1e2811f ← PCI\VEN_8086&DEV_46A6
│                         ed0b9458-c2f1-54c5-9063-dea8f75b4039 ← PCI\VEN_8086&DEV_46A6&REV_0C
│                         4a9b0c34-260f-5ee1-aebe-91d7a9e58d75 ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22F8
│                         404362be-ea2f-5e67-9841-a7449e9e68d5 ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22F8&REV_0C
│                         c4625510-a985-517c-8800-0ecfc6f68c8f ← PCI\VEN_8086&DEV_46A6&REV_00
│                         fbac133d-f4b3-5d47-9676-191e701541c9 ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22F8&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─E50:
│     Device ID:          1218eb1c6d074c70cf10feab6154b1f9711320b0
│     Current version:    1.5
│     Vendor:             Topping (USB:0x152A)
│     Serial Number:
│     GUIDs:              fc644312-30bf-5c98-b018-c4aa1d9e8260 ← USB\VID_152A&PID_8750
│                         d1224013-4302-5c7a-92ed-688b21d3ba6f ← USB\VID_152A&PID_8750&REV_0105
│     Device Flags:       • Updatable
│   
├─Embedded Controller:
│     Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│     Summary:            UEFI ESRT device
│     Current version:    0.1.9
[... 268 lines cut ...]
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          11a7fbdd09ce583e58b899660fe10f147fd155a2
      Summary:            UEFI ESRT device
      Current version:    3
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               240cf38f-3985-42c5-b77b-8e4bdaf0fcc1
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
    
────────────────────────────────────────────────
Devices that were not updated correctly:
 • Intel Management Engine (1.27.2176 → 257.27.2176)
Devices that have been updated successfully:
 • System Firmware (0.1.18 → 0.1.18)
 • System Firmware (0.1.18 → 0.1.18)

kmauleon, 2023-08-16:
hi I think this is a similar issue with #295 can you please try if this also works for you >>> https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135

latener, 2023-08-16:
> hi I think this is a similar issue with #295 can you please try if this also works for you >>> [#295 (comment)](https://github.com/fwupd/firmware-lenovo/issues/295#issuecomment-1362651135)

I came across that post prior to posting my own bug, and tried the suggestion without success. Here is the output of terminal:

```
fwupdmgr reinstall
0.	Cancel
1.	632acf4927c0b5fb53519d6beed3b60adb73f1d5 (Embedded Controller)
2.	5bfc5b91c18ef8b751d3d052af77b69ebf738038 (Intel Management Engine)
3.	a083ebc5138e5e071ef7270cc9a8280722cc7adf (System Firmware)
Choose device [0-3]: 2
Unable to locate release 1.27.2176 for Intel Management Engine
```

kmauleon, 2023-08-16:
how about downloading the firmware from LVFS site and reinstalling?

fwupdmgr --allow-older --allow-reinstall install <cab file>

latener, 2023-08-16:
After trying:
```fwupdmgr --allow-older --allow-reinstall install 893660fc23835b8a7d511abe3b2da854c7427f5dc02fdd0b4a9d97ab576bdf89-Lenovo-LenovoThinkPadP1Gen5X1ExtremeGen5-ConsumerMEFirmware-257.27.2176.cab```

The output of terminal is:
```Decompressing…           [      \                                ]
No supported devices found```

kmauleon, 2023-08-16:
you have to download the Corporate version
[ThinkPad P1 Gen5 X1 Extreme Gen5 iTBT Vpro Corporate ME Update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3JRM.firmware)
[ThinkPad P1 Gen5 X1 Extreme Gen5 dTBT Vpro Corporate ME Update](https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3JRO.firmware)

latener, 2023-08-16:
Yes, that's what I was referring to in my previous post. I downloaded the dTBT since I have a dedicated NVIDIA graphics card. The terminal didn't complain about the installation but, upon restarting, I still got the "Reading ME firmware..." message and it failed at 60% of "Recovering ME firmware..."

kmauleon, 2023-08-16:
ok... let me ping ODM dev @latener ... thank you!

apsinthium, 2023-08-18:
Once the ODM dev is contacted, is there usually a long turnaround time? I've never reported here before, but I'd love to have my audio back soon.

kmauleon, 2023-08-21:
sorry i'm afraid none... it depends actually... there are dev teams who actually reply faster... but rest assured once i get some updates from them I will feedback here

sj98ta, 2023-09-15:
Exactly the same problem on my ThinkPad P1 Gen5 X1 Extreme Gen5 Corporate ME Firmware for Hybrid Vpro.  Fails at 60%, and have lost all sound.  Reinstall results with the same error.

FlyingCurryMonster, 2023-09-15:
What's the current status of this?  I have a new thinkpad x1 extreme gen 5 stuck on firmware version 1.27.2176, and I'm trying to update the firmware to hopefully clear other issues.  Is there a way I can install a particular version of the intel management engine?

jbillingredhat, 2023-09-15:
> What's the current status of this? I have a new thinkpad x1 extreme gen 5 stuck on firmware version 1.27.2176, and I'm trying to update the firmware to hopefully clear other issues. Is there a way I can install a particular version of the intel management engine?

If fwupd is reporting that you have 1.27.2176, you actually have version 257.27.2176 installed. From what I've been told by Lenovo, there's a bug in how the version is represented by the firmware, and the first part of the version can't be higher than 256 (and 1 really means 256 + 1).  Hopefully there will be a firmware fix at some point to address this.

sj98ta, 2023-09-15:
I hope so,  because it now takes a LONG time to boot the machine, having to wait for it to read, and then at 60% fails.  :-(  Thanks for the update

mrhpearson, 2023-09-15:
Just to ack the above - your FW is updated, so for now just ignore the update.
The FW team are working on the fix, but my understanding is we need a BIOS fix to address this first as it's related to how the BIOS exports the ME version to the ESRT table and all the pieces fall out from there.

I saw kmauleon bug the FW team on the sound dropping update so hopefully she can find steps to resolve that.

kmauleon, 2023-09-18:
for sound issue please try if any of the following works for you >> https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1380085679
https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1381188172
https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1407809074

sj98ta, 2023-09-19:
> for sound issue please try if any of the following works for you >> [#250 (comment)](https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1380085679) [#250 (comment)](https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1381188172) [#250 (comment)](https://github.com/fwupd/firmware-lenovo/issues/250#issuecomment-1407809074)

Still all a no-go for me.

apsinthium, 2023-09-19:
I still have no sound as well. I booted Windows 10 from WindowsToGo and the firmware fails to read immediately instead of taking ten minutes. I still wasn't able to get the BIOS update to go through using the official .exe from Lenovo nor using Lenovo Vantage (I'm still on 1.18 and the current version for Windows states 1.19).

kmauleon, 2023-09-19:
@apsinthium @sj98ta can you please try to update BIOS only in Linux?

apsinthium, 2023-09-20:
@kmauleon The BIOS for Linux is only v1.18 which is what I currently have installed.

kmauleon, 2023-09-20:
ok... got it... i will contact ODM dev again and feedback if any... thank you very much

sj98ta, 2023-09-20:
I am on Linux as well,  and it is at v1.14.  It appears it tried to update to 0.1.19,  but fails, because the bloody ME firmware update fails (assumption), The update error simply says `failed to run update on reboot`.

apsinthium, 2023-09-20:
Yes, all other updates fail because the ME (apparently) tries to update before anything else.

kmauleon, 2023-09-20:
usually certain BIOS version is a prerequisite before updating MEFW and not the other way around. and these dependencies can be defined in metainfo file.  Just wondering... have you tried updating firmware separately in this sequence
update BIOS reboot
update EC reboot
update MEFW reboot

sj98ta, 2023-09-20:
1) Is there any reason to believe that a BIOS update contains a resolution?   Based on comments above, it does not seem likely.  

That said,  how do I update individually?    Last night, I downloaded n3jul12w.zip,  unpacked, and tried up update using N3JET34W.cab,  and that fails with some message about missing signature.

apsinthium, 2023-09-20:
I've also tried installing the BIOS (via Windows To Go), from the official .exe, and it wouldn't push past the failed ME firmware.

kmauleon, 2023-09-20:
BIOS >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3JET.firmware
EC >> https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3JHT.firmware
please download from above links and install separately via
fwupdmgr --allow-older --allow-reinstall install <cab_filename.cab>

OR

Open Gnome Software
Go to Updates tab
Choose firmware to update and restart

sj98ta, 2023-09-21:
Again, the upgrade of the BIOS does not work,  again because the ME update fails.

sj98ta, 2023-09-21:
In addition to having no sound,  ever since the failed update, the docking station (Thinkpad Thunderbolt 3 Dock Gen 2) works  functions very erratically.   By this I mean,  all connection is lost over the thunderbolt connection, and I lose keyboard, mouse, network, etc.   Basically the machine is in a state to where it is almost completely unusable.

kmauleon, 2023-09-21:
@sj98ta let's just wait for any feedback from ODM dev.  As for the docking station issue, this is handled by different owner so please open another issue for proper tracking... thank you very much

sj98ta, 2023-09-21:
> @sj98ta let's just wait for any feedback from ODM dev. As for the docking station issue, this is handled by different owner so please open another issue for proper tracking... thank you very much

I will wait to deal with the docking issue if/when the firmware update issue is resolved.  It is not a coincidence that the docking station started having issues immediately when the firmware update issue started. 

Thanks...

kmauleon, 2023-09-22:
we also have dock firmware issues raised against other thinkpads... please feel free to check if you have similar issues >> https://github.com/fwupd/firmware-lenovo/issues/246

sj98ta, 2023-09-22:
> we also have dock firmware issues raised against other thinkpads... please feel free to check if you have similar issues >> #246

Thanks.... I have had no issues at all with the dock, up until the ME update issue.  But I will check out.

sj98ta, 2023-09-25:
Any news yet this issue?   This is extremely painful.

kmauleon, 2023-09-28:
@sj98ta unfortunately no update yet from ODM dev.  BTW, have you seen this comment >> https://github.com/fwupd/firmware-lenovo/issues/246#issuecomment-1736260139

sj98ta, 2023-09-28:
@kmauleon   Thanks for the link to the comment.  I had not seen it,  but many times I have booted without the dock connected, and with the lid open.  Same issue however,  will not update.   

Something odd happened the other day, in that the bios updated to 0.1.19.  I was a bit surprised.  But, the ME issue remains

kmauleon, 2023-09-28:
please try to boot system without docking station and update it manually via below command if you have not tried this one....
fwupdmgr --allow-older --allow-reinstall install <cab file>
reboot system

sj98ta, 2023-09-28:
@kmauleon The problem is unchanged.  ME update fails at 60%

kmauleon, 2023-09-28:
what version can you see in BIOS Menu?

sj98ta, 2023-09-28:
@kmauleon  N3JET35W (1.19)

kmauleon, 2023-09-28:
sorry for not being clear... I meant MEFW version in BIOS Menu please....

sj98ta, 2023-09-28:
@kmauleon   Odd.... MEFW shows 16.1.27.2176

kmauleon, 2023-09-28:
this means your MEFW is updated already that is why the MEFW update does not continue since you have the same version... in the meantime since we have the version format issue... please ignore the update notice for MEFW... this is still ongoing investigation... thank you very much @sj98ta

sj98ta, 2023-09-28:
@kmauleon  Well, it is difficult to ignore the update notice, since  on every boot, it tries to read the firmeware, gets to 60% and dies.  The problem is, that takes a long time... 5 - 8 minutes.   

Also, since experiencing this issue, I no longer have audio or microphone since this failed update problem started.   Not sure about the camera, but I believe it too is no longer working.   Pretty difficult to attend WebEx and Slack sessions without that. 

If the firmware is really installed, why would the media devices no longer function?

sj98ta, 2023-09-28:
@kmauleon I retract what I said about the video.. the camera is working.... audio however (speakers and mic) are not

kmauleon, 2023-09-28:
hmmm please don't choose YES when prompted for MEFW update.... your audio issue might not be caused by MEFW update since you already have an updated MEFW and BIOS as well.... this could be related with your dock fw update.  But I am not entirely sure unless we heard any update from ODM dev... all I can do for now is to follow up ODM.... will update again once I heard anything from them.... thank you very much

sj98ta, 2023-09-28:
@kmauleon It is not at all related to the dock, since I don't have the dock plugged in.   I am certain it has to do with the MEFW update however.  

The dmesg output shows:
```
[   57.463019] sof-audio-pci-intel-tgl 0000:00:1f.3: cl_dsp_init: timeout with rom_status_reg (0x80000) read
[   57.463047] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
[   57.463055] sof-audio-pci-intel-tgl 0000:00:1f.3: Boot iteration failed: 3/3
[   57.463063] sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_IN_PROGRESS (2)
[   57.463069] sof-audio-pci-intel-tgl 0000:00:1f.3: 0x06000021: module: ROM, state: CSE_IPC_RESET_PHASE_1, waiting for: CSE_CSR, running
[   57.463081] sof-audio-pci-intel-tgl 0000:00:1f.3: extended rom status:  0x6000021 0x0 0x0 0x0 0x0 0x0 0x2560521 0x0
[   57.463082] sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
[   57.463605] sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp init failed after 3 attempts with err: -110
[   57.463633] sof-audio-pci-intel-tgl 0000:00:1f.3: Failed to start DSP
[   57.463633] sof-audio-pci-intel-tgl 0000:00:1f.3: error: failed to boot DSP firmware -110
[   57.514009] sof-audio-pci-intel-tgl 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[   57.514011] sof-audio-pci-intel-tgl 0000:00:1f.3: error: dsp core reset failed: core_mask 1
[   57.514108] sof-audio-pci-intel-tgl 0000:00:1f.3: error: sof_probe_work failed err: -110
```

I am not clear by with you mean by `don't choose YES when prompted for MEFW update`.   I am not being prompted.

kmauleon, 2023-09-28:
can you please send us a video of your symptom "Well, it is difficult to ignore the update notice, since on every boot, it tries to read the firmeware, gets to 60% and dies. The problem is, that takes a long time... 5 - 8 minutes." I think the MEFW update will start only once you chose to update it via fwupdmgr or gnome software update. that is why I said that you can just ignore the update notice for MEFW.
@hughsie @mrhpearson do you have any inputs regarding "dmesg output" sorry not familiar with this

sj98ta, 2023-09-28:
@kmauleon   I don't have a means to capture video of this.   

"I think the MEFW update will start only once you chose to update it via fwupdmgr or gnome software update. that is why I said that you can just ignore the update notice for MEFW."     No,  for me this happens on every boot of the machine.  It says it is reading the ME firmware,  it speeds up dramatically at 50%, then fails at 60% with "Failed to recover ME firmware" 

Please do note that the originator of this issued also reports no audio devices.

apsinthium, 2023-09-28:
Yes, I have the same issue. No audio output/input. I cannot even use headphones because the entire sound card is invisible to my machine. When I boot from Windows To Go then everything works as intended.

Unfortunately, when rebooting today the ME firmware started trying to read again. I had been getting an instant, "Unable to read ME firmware..." without the delay for the past week. Now, I'm back to the 7-8 minute wait time to boot while this update fails.

And, also, this happens on _every_ boot as I stated in the original bug. Not just when it is schedule from within fwupdmgr.

kmauleon, 2023-09-29:
@hughsie are you familiar with this issue >> MEFW update happens every boot 

"Unfortunately, when rebooting today the ME firmware started trying to read again. I had been getting an instant, "Unable to read ME firmware..." without the delay for the past week. Now, I'm back to the 7-8 minute wait time to boot while this update fails.

And, also, this happens on every boot as I stated in the original bug report. Not just when it is schedule from within fwupdmgr."

kmauleon, 2023-09-29:
BTW, please note that we might not receive any feedback from ODM dev until first week of October.  just learned that they are on holidays... sorry for the inconvenience

laustemil, 2023-09-29:
I had a similar problem and just solved it using

fwupdmgr reinstall

and choosing Intel Management Engine
https://superuser.com/questions/1803761/thinkpad-t15-gen-2-fail-to-recover-me-firmware-at-booting-then-ubuntu-22-04-has

sj98ta, 2023-09-29:
I am on RHEL,  but have tried the above, and does not resolve it for me.

apsinthium, 2023-09-29:
The above resolution has been suggested multiple times, and I've tried it multiple times to no avail. I'm on Debian 13 (Trixie/Testing), but was on 12 (Bookworm) when the issue arose.

apsinthium, 2023-10-11:
Not for me. I'm still waiting to hear back. It's the middle of the second week of October so the ODM dev hopefully has returned from vacation. Going on no sound for two months now.

kmauleon, 2023-10-12:
@sj98ta thanks for the good news!! do you remember updating anything... might it be for Linux or any firmware?
@apsinthium it seems ODM dev is still o holidays... haven't heard from them yet...sorry T_T

sj98ta, 2023-10-12:
@kmauleon    As far as I know I had not updated anything, which is the bizarre thing about it.    I came in, in the morning,  powered it up as always, and about to go get a cup of coffee while waiting for the ME failure,  and was shock that I didn't see anything other than it was booting linux.    I keep waiting for it to return. :-(    But, it has now resolve all of the problems I was having, including the dock station dropping out.  The dock has been rock solid.

apsinthium, 2023-10-31:
@kmauleon Any updates? Going on three months now.

kmauleon, 2023-10-31:
hi @apsinthium no major updates yet ... they just said they are still investigating... will follow up again

[41 more replies]

closed 2024-04-05 by apsinthium