fwupd/firmware-lenovo #361: Wrong Intel Management Engine version on ThinkPad P1 Gen 5

luudvig, 2023-08-17:
This issue seems to be just like #300 but for ThinkPad P1 Gen 5 and different version numbers.

I updated to what is released as version 16.1.27.2176 (from version 16.1.25.1932) of the Intel Management Engine Firmware a few days ago but after reboot it still shows up as an available update. Also the _Update State_ is set to "failed" when installed but BIOS shows the correct version (my output below says "success" because I have reformated to reenable encrypted system drive).

```
Intel Management Engine:
│   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│   Summary:            UEFI ESRT device
│   Current version:    1.27.2176
│   Minimum Version:    0.0.1
│   Vendor:             Lenovo (DMI:LENOVO)
│   Update State:       Success
│   GUID:               012b2727-3f35-c20a-fb78-d7d4216ce715
│   Device Flags:       • Internal device
│                       • Updatable
│                       • System requires external power source
│                       • Supported on remote server
│                       • Needs a reboot after installation
│                       • Device is usable for the duration of the update
│ 
└─ThinkPad P1 Gen5 X1 Extreme Gen5 dTBT Nonvpro Consumer ME Update:
      New version:      257.27.2176
      Remote ID:        lvfs
      Release ID:       49248
      Summary:          Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Hybrid Consumer ME Firmware
      License:          Proprietary
      Size:             13,3 MB
      Created:          2023-07-14
      Urgency:          High
      Details:          https://pcsupport.lenovo.com/de/en/search?query=N3JRP11W
      Vendor:           Lenovo
      Release Flags:    • Trusted metadata
                        • Is upgrade
      Description:      
      Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Consumer ME Firmware, for Hybrid Nonvpro
```

kmauleon, 2023-08-18:
hi @luudvig  can you please share results of the following then i'll forward them to ODM dev... thank you!
fwupdmgr get-devices
grep . /sys/firmware/efi/esrt/entries/*/*

DanielBoettner, 2023-08-19:
I also encounter this issue (I think) for a week or so 

```
"Metadata" : {
                             "DistroId" : "ubuntu",
                             "DistroVersion" : "23.04"
                           },
...

                             {

                               "ReleaseId" : null,
                               "Protocol" : "org.nvmexpress",
                               "UpdateState" : 2,

                               "Plugin" : "nvme",
                               "VersionOld" : "1107ANLA",
                               "VersionNew" : "1108ANLA",

                               "Created" : 1682316243,
                               "Modified" : 1682341048,
                               "Metadata" : {
                                 "RuntimeVersion(org.freedesktop.gusb)" : "0.3.10",
                                 "HostSku" : "LENOVO_MT_21C3_BU_Think_FM_ThinkPad L15 Gen 3",
                                 "DistroVersion" : "22.10",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.8.4",
                                 "FwupdSupported" : "True",
                                 "HostFamily" : "ThinkPad L15 Gen 3",
                                 "KernelName" : "Linux",
                                 "CompileVersion(com.hughsie.libjcat)" : "0.1.9",
                                 "RuntimeVersion(org.kernel)" : "6.0.9-060009-generic",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.3.10",
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "DistroId" : "ubuntu",
                                 "RuntimeVersion(org.freedesktop.fwupd-efi)" : "1.0",
                                 "CpuArchitecture" : "x86_64",

                                 "HostVendor" : "LENOVO",
                                 "HostProduct" : "21C3007HGE",
                                 "KernelVersion" : "6.0.9-060009-generic",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.8.4",
                                 "FwupdTainted" : "False"
                               }
                             },
```

```
{

                               "ReleaseId" : null,
                               "Protocol" : "com.usi.dock",
                               "UpdateState" : 2,

                               "Plugin" : "usi_dock",
                               "VersionOld" : "10.7",
                               "VersionNew" : "10.13",
                               "Flags" : 140737626783786,
                               "Created" : 1690871090,
                               "Modified" : 1690871221,
                               "Metadata" : {
                                 "HostEnclosureKind" : "a",
                                 "HostSku" : "LENOVO_MT_21C3_BU_Think_FM_ThinkPad L15 Gen 3",
                                 "DistroVersion" : "23.04",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.8.12",
                                 "RuntimeVersion(org.kernel)" : "6.2.0-26-generic",
                                 "CpuArchitecture" : "x86_64",
                                 "HostFamily" : "ThinkPad L15 Gen 3",
                                 "HostBiosVersion" : "R1XET46W (1.28 )",
                                 "HostVendor" : "LENOVO",
                                 "RuntimeVersion(org.freedesktop.gusb)" : "0.4.5",
[... 532 lines cut ...]
├─USB3.0 Hub:
│     Device ID:          7c5b6dadf8126e3820c86cb56d9e1777084a98a8
│     Current version:    38.0
│     Vendor:             Intel Corporation. (USB:0x8087)
│     GUIDs:              b3f8b518-ebdd-5969-9b44-6b096eff5b9b ← USB\VID_8087&PID_0B40
│                         27f4ed60-ed53-52a7-ae28-ea87d9a23302 ← USB\VID_8087&PID_0B40&REV_1234
│                         c6e7a14f-f13c-55e5-99de-d59461db1d27 ← USB\VID_8087&PID_0B40&HUB_0000
│                         350ec5bb-25c0-543e-8d4c-72c30d2b8df0 ← TBT-01082031
│     Device Flags:       • Updatable
│                         • Signed Payload
│   
└─USB4 Retimer:
      Device ID:          16d35c510618116e7899ae45e62605572872e713
      Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
      Current version:    10.00
      Vendor:             Intel (THUNDERBOLT:0x8087)
      GUIDs:              e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e ← THUNDERBOLT\VEN_8087&DEV_15EE
                          f00257c8-0e09-5a32-bbb3-f49748ae6cc5 ← THUNDERBOLT\VEN_8087&DEV_15EE&REV_00
                          6d71ac25-b368-5e7b-ac62-32df365fa640 ← TBT-808715ee-retimer0-0:1.1
      Device Flags:       • Internal device
                          • Updatable
                          • Device stages updates
```

The content of `/sys/firmware/efi/esrt/entries/` are 18 entry files (0-17)

kmauleon, 2023-08-21:
hi @DanielBoettner can i know the ESRT values for MEFW specifically.... thanks!

DanielBoettner, 2023-08-24:
Hi @kmauleon I hope this helps. I attached the small script which I used to generate the output at the end.

```
=== entry0 ===
capsule_flags: 0x0
fw_class: f8637dbe-5762-0000-873a-910aa71c49e4
fw_type: 2
fw_version: 0
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 15
=== entry1 ===
capsule_flags: 0x0
fw_class: 4e88068b-41b2-4e05-893c-db0b43f7d348
fw_type: 2
fw_version: 70151
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry10 ===
capsule_flags: 0x0
fw_class: 4eae73e8-1b3d-4684-bfe1-ef3c6094aedd
fw_type: 2
fw_version: 3
last_attempt_status: 0
last_attempt_version: 3
lowest_supported_fw_version: 0
=== entry11 ===
capsule_flags: 0x0
fw_class: 5dc5d695-c24f-4723-aa17-1b30f79c6d19
fw_type: 1
fw_version: 65564
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry12 ===
capsule_flags: 0x8010
fw_class: 10cba2e7-8db3-411e-957a-ab83c2083f25
fw_type: 2
fw_version: 18548864
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry13 ===
capsule_flags: 0x0
fw_class: 6f83eb9d-7ea4-4b96-93dd-55597f216fc8
fw_type: 2
fw_version: 65554
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 65554
=== entry14 ===
capsule_flags: 0x8010
fw_class: 9b8a58ae-ad1a-4b73-b45d-2ebc337ef708
fw_type: 2
fw_version: 16777219
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry15 ===
capsule_flags: 0x8010
fw_class: 3a244308-4c8e-4abf-adcc-c99e4f37363f
fw_type: 2
fw_version: 0
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry16 ===
capsule_flags: 0x8010
fw_class: d855de5e-0293-4cd7-be42-0bbcb7e3b394
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry17 ===
capsule_flags: 0x0
fw_class: dcde6a43-bdd2-4da4-ace7-d005c112bd13
fw_type: 2
fw_version: 66048
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 66048
=== entry2 ===
capsule_flags: 0x0
fw_class: 69585d92-b50a-4ad7-b265-2eb1ae066574
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry3 ===
capsule_flags: 0x0
fw_class: 23192307-d667-4bdf-af1a-6059db171246
fw_type: 2
fw_version: 18548864
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry4 ===
capsule_flags: 0x0
fw_class: 76ca0ad8-4a14-4389-b7e5-fd88791762ad
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry5 ===
capsule_flags: 0x0
fw_class: 626d93db-2c42-48c3-915a-71f968a81b04
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry6 ===
capsule_flags: 0x0
fw_class: 86a885ee-d71e-2ed6-0fc1-9d6ccc9677eb
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry7 ===
capsule_flags: 0x0
fw_class: 7c8c8f35-5986-2142-0a0c-43491d11746b
fw_type: 2
fw_version: 26388
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry8 ===
capsule_flags: 0x0
fw_class: 54cb734c-975c-4a74-9556-57ae6d9a2229
fw_type: 2
fw_version: 196608
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 57374
=== entry9 ===
capsule_flags: 0x0
fw_class: 88523ddf-3aac-41f6-b15f-dc7bea434b39
fw_type: 2
fw_version: 532
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
```

And if needed for other users. This is the small bash script used to generate this output.
Obviously needs root permissions to access the files.

```bash
#!/bin/bash

# Navigate to the ESRT entries directory
cd /sys/firmware/efi/esrt/entries || exit 1

# Iterate over each entry folder
for entry in entry*; do
  echo "=== $entry ==="
  cd "$entry" || continue
  
  # Iterate over each file in the entry folder and display its content with the filename prepended
  for file in *; do
    echo -n "$file: "
    cat "$file"
  done
  
  cd ..
done
```

kmauleon, 2023-08-25:
hi @DanielBoettner sorry I just noticed you have a different system which is L14/L15... would you please open a separate ticket for this for proper tracking?  we have different owner for each system... thank you very much...

lohedges, 2023-08-25:
I also have this problem on a P1 Gen 5:

```
fwupdmgr get-devices 
LENOVO 21DCCTO1WW
│
├─ELAN0686:00 04F3:320D:
│     Device ID:          522cb97965758d9b1309d3217dfd41673f410e15
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0003
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              7e8dad8a-0b0e-527b-8810-d5dbca8804d7 ← HIDRAW\VEN_04F3&DEV_320D
│                         74e88f64-4d6b-5048-8199-fdcc7833b48a ← HIDRAW\VEN_04F3&DEV_320D&MOD_003D
│                         4b808d27-c833-50cc-8177-9a401f5c08fe ← ELANTP\ICTYPE_14
│                         e25cd415-0431-502a-9077-3c409b007897 ← ELANTP\ICTYPE_14&MOD_003D
│                         8cb8f87c-48a3-51e4-ad02-8dd596adddc7 ← ELANTP\ICTYPE_14&MOD_003D&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│
├─12th Gen Intel Core™ i7-12800H:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0000042c
│ │   Vendor:             Intel
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│ │                       ab855c04-4ff6-54af-8a8a-d8193daa0cd8 ← CPUID\PRO_0&FAM_06&MOD_9A
│ │                       3ebbde86-d03e-549a-a8fd-02ebf9aa537a ← CPUID\PRO_0&FAM_06&MOD_9A&STP_3
│ │   Device Flags:       • Internal device
│ │
│ └─USB4 host controller:
│       Device ID:        85d7e6f391d98dfeed9cbe039684a07f316d3ec7
│       Summary:          Unmatched performance for high-speed I/O
│       Current version:  38.00
│       Vendor:           Lenovo (THUNDERBOLT:0x0109)
│       GUIDs:            e074e6b4-fab8-5ef9-bbe3-f49d2510704b ← THUNDERBOLT\VEN_0109&DEV_2113
│                         e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│       Device Flags:     • Internal device
│                         • System requires external power source
│                         • Signed Payload
│
├─Alder Lake-P Integrated Graphics Controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              eaad9970-8e4d-56da-88ab-41a8c1e2811f ← PCI\VEN_8086&DEV_46A6
│                         4a9b0c34-260f-5ee1-aebe-91d7a9e58d75 ← PCI\VEN_8086&DEV_46A6&SUBSYS_17AA22F8
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│
├─Embedded Controller:
│     Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│     Summary:            UEFI ESRT device
│     Current version:    0.1.9
│     Minimum Version:    0.1.9
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               f8963b51-51b6-4059-87c4-722967c16745
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─GA104 [Geforce RTX 3070 Ti Laptop GPU]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE, PCI:0x8086)
│     GUIDs:              276de2e3-50dd-55bf-afc0-6fb90f2e1958 ← PCI\VEN_10DE&DEV_24A0
[... 611 lines cut ...]
        Summary:          Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Hybrid Corporate ME Firmware
        Licence:          Proprietary
        Size:             13.3 MB
        Created:          2023-07-14
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N3JRO11W
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:
        Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Corporate ME Firmware, for Hybrid Vpro
        Checksum:         68b320dd67db5717062ddbc6d1ad7e42c7c4e82d90fab41c73d87685fe6446fd
```

Looking at the esrt entries, the only one with a failed attempt status reports:

```
capsule_flags : 0x0
fw_class : 4e78ce68-5389-4a95-bf10-e3568c30caf8
fw_type : 2
fw_version : 18548864
last_attempt_status : 6144
last_attempt_version : 984818
lowest_supported_fw_version : 0
```

pn-santos, 2023-08-25:
I also have this issue with a X1 Extreme Gen 5 (which is AFAIK exactly the same as the P1 but with another graphics card):
```
❯ fwupdmgr get-devices
21DECTO1WW
│
├─Embedded Controller:
│     Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│     Summary:            UEFI ESRT device
│     Current version:    0.1.9
│     Minimum Version:    0.1.9
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               f8963b51-51b6-4059-87c4-722967c16745
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          0fef0a0c55f6442bffaebd774ae771341c89571b
│     Current version:    56.20
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:              b687616d-e0bc-58d3-9022-8f0065eed4ca
│                         bcd2c020-e315-55aa-b7e1-7d2d7108037e
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│     Summary:            UEFI ESRT device
│     Current version:    1.27.2176
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               2d9ea625-256b-7882-2103-da49b102ef33
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    1.27.2176
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               4e78ce68-5389-4a95-bf10-e3568c30caf8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Micron MTFDKBA1T0TFH:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    7006P7LN
│     Vendor:             Micron Technology Inc (NVME:0x1344)
│     GUIDs:              aebc2e6e-4b99-5cb8-b72f-01039bcb849c
│                         07e304bc-bb66-563d-bb53-011f94039748
│                         0ae18239-216f-568f-951d-da402dfa1d70
│                         b70b3b98-b8a6-5e07-8a97-138d66b0dded
│                         dd4c8625-57b6-53a3-852a-c01304b27ab5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
[... 435 lines cut ...]
--- /sys/firmware/efi/esrt/entries/entry7
capsule_flags: 0x0
fw_class: 2a197802-e469-4fa7-a37b-2d681bcf416f
fw_type: 2
fw_version: 10551296
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 57374
--- /sys/firmware/efi/esrt/entries/entry8
capsule_flags: 0x0
fw_class: e5212228-46c9-4190-b064-830baa75ea8c
fw_type: 2
fw_version: 3615822
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
--- /sys/firmware/efi/esrt/entries/entry9
capsule_flags: 0x0
fw_class: 3a6cdf97-cfb9-479d-9c4c-135e0d4505c9
fw_type: 1
fw_version: 65554
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 65536
```

Kenterfie, 2023-08-31:
`21DCCTO1WW
│
└─Intel Management Engine:
  │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
  │   Summary:            UEFI ESRT device
  │   Current version:    1.27.2176
  │   Minimum Version:    0.0.1
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               012b2727-3f35-c20a-fb78-d7d4216ce715
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad P1 Gen5 X1 Extreme Gen5 dTBT Nonvpro:
        New version:      257.27.2176
        Remote ID:        lvfs
        Release ID:       49248
        Summary:          Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Hybrid Consumer ME Firmware
        License:          Proprietary
        Size:             13,3 MB
        Created:          2023-07-14
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N3JRP11W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Consumer ME Firmware, for Hybrid Nonvpro
`

kmauleon, 2023-09-01:
thank you all for reporting... forwarded to ODM CO... will feedback once I got any update

JonnyLoughlin, 2023-09-12:
Encountering the same issue so I figured I would provide more output:
```
Devices with no available firmware updates: 
 • ELAN0686:00 04F3:320D
 • Integrated Camera
 • Intel Management Engine
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • Prometheus
 • SKHynix HFS001TDE9X081N
 • System Firmware
 • UEFI dbx
LENOVO 21DC002VUS
│
└─Intel Management Engine:
  │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
  │   Summary:            UEFI ESRT device
  │   Current version:    1.27.2176
  │   Minimum Version:    0.0.1
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               2d9ea625-256b-7882-2103-da49b102ef33
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad P1 Gen5 X1 Extreme Gen5 dTBT Vpro Corporate ME Update:
        New version:      257.27.2176
        Remote ID:        lvfs
        Release ID:       49246
        Summary:          Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Hybrid Corporate ME Firmware
        License:          Proprietary
        Size:             13.3 MB
        Created:          2023-07-14
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N3JRO11W
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Corporate ME Firmware, for Hybrid Vpro
        Checksum:         68b320dd67db5717062ddbc6d1ad7e42c7c4e82d90fab41c73d87685fe6446fd
      
```
```
=== entry0 ===
capsule_flags: 0x0
fw_class: 4e88068b-41b2-4e05-893c-db0b43f7d348
fw_type: 2
fw_version: 70154
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry1 ===
capsule_flags: 0x0
fw_class: 69585d92-b50a-4ad7-b265-2eb1ae066574
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry10 ===
capsule_flags: 0x8010
fw_class: 2d9ea625-256b-7882-2103-da49b102ef33
fw_type: 2
fw_version: 18548864
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry11 ===
capsule_flags: 0x0
fw_class: f8963b51-51b6-4059-87c4-722967c16745
fw_type: 2
fw_version: 65546
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 65546
=== entry12 ===
capsule_flags: 0x8010
fw_class: edc6568f-6565-41df-9de6-18d69e5b17c4
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry13 ===
capsule_flags: 0x8010
fw_class: e5212228-46c9-4190-b064-830baa75ea8c
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry14 ===
capsule_flags: 0x0
fw_class: 87049c36-dfcf-431c-b506-41f4ac4fb172
fw_type: 2
fw_version: 117572352
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 117572352
=== entry2 ===
capsule_flags: 0x0
fw_class: 4e78ce68-5389-4a95-bf10-e3568c30caf8
fw_type: 2
fw_version: 18548864
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry3 ===
capsule_flags: 0x0
fw_class: 76ca0ad8-4a14-4389-b7e5-fd88791762ad
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry4 ===
capsule_flags: 0x0
fw_class: 626d93db-2c42-48c3-915a-71f968a81b04
fw_type: 2
fw_version: 1
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry5 ===
capsule_flags: 0x0
fw_class: a4bd26a4-8a04-48ea-8061-4867fe826f1e
fw_type: 2
fw_version: 4128
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 1
=== entry6 ===
capsule_flags: 0x0
fw_class: 0eb9bda9-3010-493a-a6a8-b5e80eddf870
fw_type: 2
fw_version: 117572352
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 0
=== entry7 ===
capsule_flags: 0x0
fw_class: 2a197802-e469-4fa7-a37b-2d681bcf416f
fw_type: 2
fw_version: 10551296
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 57374
=== entry8 ===
capsule_flags: 0x0
fw_class: 240cf38f-3985-42c5-b77b-8e4bdaf0fcc1
fw_type: 2
fw_version: 3
last_attempt_status: 0
last_attempt_version: 3
lowest_supported_fw_version: 0
=== entry9 ===
capsule_flags: 0x0
fw_class: 3a6cdf97-cfb9-479d-9c4c-135e0d4505c9
fw_type: 1
fw_version: 65555
last_attempt_status: 0
last_attempt_version: 0
lowest_supported_fw_version: 65554
```

T800-101A, 2023-09-12:
I have same problem, I updated it 3 times thinking it was something new. 
same versio Thinkpad P1 Gen 5.

kmauleon, 2023-09-13:
thank you @JonnyLoughlin @T800-101A for reporting.... unfortunately no update yet from dev.  will feedback once I got any

george-lorch, 2023-09-21:
Same issue here I think, Ubuntu 20.04 6 LTS
:~$ dmidecode | grep SKU
		Corporate SKU
	SKU Number: LENOVO_MT_21DD_BU_Think_FM_ThinkPad P1 Gen 5
	SKU Number: Not Specified

:~$ fwupdmgr get-history
21DDS5H800
│
├─Prometheus:
│ │   Device ID:          23cf6368c14a875f74c38a5a423518f38d8abbbc
│ │   Previous version:   10.01.3333223
│ │   Update State:       Success
│ │   Last modified:      2023-05-26 12:25
│ │   GUID:               d67c4925-e30f-53e6-abc6-e7af0e4c5884
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Is in bootloader mode
│ │                       • Cryptographic hash verification is available
│ │ 
│ └─Prometheus:
│       New version:      10.01.3478575
│       Remote ID:        lvfs
│       Release ID:       10696
│       Summary:          Firmware for the Synaptics Prometheus Fingerprint Reader device
│       License:          Proprietary
│       Size:             425.5 kB
│       Created:          2021-06-09
│       Urgency:          Medium
│       Vendor:           Synaptics
│       Duration:         2 seconds
│       Description:      
│       New features and enhancements:
│       
│       • Fix an authentication issue on FOP project.
│     
├─Embedded Controller:
│ │   Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│ │   Previous version:   0.1.9
│ │   Update State:       Success
│ │   Last modified:      2023-09-21 11:48
│ │   GUID:               f8963b51-51b6-4059-87c4-722967c16745
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─ThinkPad P1 Gen 5/X1 Extreme Gen 5:
│       New version:      0.1.10
│       Remote ID:        lvfs
│       Release ID:       58487
│       Summary:          Lenovo ThinkPad P1 Gen 5/X1 Extreme Gen 5 System Firmware
│       License:          Proprietary
│       Size:             1.1 MB
│       Created:          2023-08-09
│       Urgency:          High
│       Vendor:           Lenovo
│       Description:      
│       Changes in this release - version 1.10
│       
│       Problem fixes - Fixed an issue where USB-C/Thunderbolt port and charging function become unavailable rarely.
│     
├─System Firmware:
│ │   Device ID:          a083ebc5138e5e071ef7270cc9a8280722cc7adf
│ │   Previous version:   0.1.18
│ │   Update State:       Failed
│ │   Update Error:       failed to run update on reboot
│ │   Last modified:      2023-09-21 12:47
[... 67 lines cut ...]
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 1.27.2176 to 257.27.2176?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Corporate ME Firmware, for Hybrid    ║
║ Vpro                                                                         ║
║                                                                              ║
║ 21DDS5H800 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Any ideas on how to get our of this failed update on reboot and get my audio back?  This has all been working for months until I did this update today.  Thanks.

timblaktu, 2023-09-21:
Another P1 Gen5 (Arch) Linux (`6.5.3-arch1-1`) user here (linux-hardware probe results here](https://linux-hardware.org/?probe=8c65f7a68a) encountering the odd "new version" for Intel Mgmt Engine in `fwupd` (below).

I'm troubleshooting kernel errors ("can't change power state") and panics at shutdown that indicate they are related to pci device ids associate with thunderbolt controller. I'm starting by ensuring my fw is updated. For now I'm going to assume that this "new version" 257.27.2176 is incorrect and not-to-be-updated-to. Please advise.

```
LENOVO 21DCCTO1WW
│
└─Intel Management Engine:
  │   Device ID:          a083ebc5138e5e071ef7270cc9a8280722cc7adf
  │   Summary:            UEFI ESRT device
  │   Current version:    1.27.2176
  │   Minimum Version:    0.0.1
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUID:               012b2727-3f35-c20a-fb78-d7d4216ce715
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad P1 Gen5 X1 Extreme Gen5 dTBT Nonvpro Consumer ME Update:
        New version:      257.27.2176
        Remote ID:        lvfs
        Release ID:       49248
        Summary:          Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Hybrid Consumer ME Firmware
        License:          Proprietary
        Size:             13.3 MB
        Created:          2023-07-14
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N3JRP11W
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:
        Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Consumer ME Firmware, for Hybrid Nonvpro
        Checksum:         893660fc23835b8a7d511abe3b2da854c7427f5dc02fdd0b4a9d97ab576bdf89
```

kmauleon, 2023-09-22:
@george-lorch @timblaktu thanks for reporting... these are the same firmware version so please ignore the update notice... 1 version format should be retained. ODM dev is working for fix... no outlook date yet unfortunately... will feedback once I received updates from them... thank you very much

laustemil, 2023-09-29:
I had a similar problem and just solved it using

fwupdmgr reinstall

  https://superuser.com/questions/1803761/thinkpad-t15-gen-2-fail-to-recover-me-firmware-at-booting-then-ubuntu-22-04-has

AImenes, 2023-10-03:
Hi @kmauleon! Any updates from ODM?

justinsg, 2023-10-15:
My P1 Gen 5 system was attempting to upgrade the ME firmware every boot (making me wait up to 5min for it to fail & proceed to the OS), and then once booted into Ubuntu 22.04 I had no sound. However, I fixed it by running:

```
fwupdmgr refresh --force
fwupdmgr update
```

After one more reboot, the system had working audio and I am no longer seeing the "Failed to install" message every boot. After boot, I still get a prompt to upgrade from `1.27.2176  -->  257.27.2176` which I'm ignoring because accepting that caused the issue in the first place.

kmauleon, 2023-10-16:
@justinsg please ignore the update, the correct one should be 1.27.2176
will follow up ODM dev regarding the versioning issue

kmauleon, 2023-11-06:
sorry ODM ME dev is still in discussion with BIOS dev regarding the fix. will feedback again once I get any update

padamstx, 2023-11-24:
I'm also having this problem on a P1 Gen 5 with Fedora 38. Ever since the first failed firmware install occurred, my trackpointer is completely unresponsive after resuming from suspend (a reboot brings it back, thankfully).  Is there any way to back out whatever has failed so I can get my trackpointer back?

padamstx, 2023-11-28:
@kmauleon Any progress on this issue yet?   Is there any way for me to get my trackpoint back after a resume?  Trackpoint mouse is completely unresponsive after a resume ever since I tried to install this failed firmware update.

kmauleon, 2023-11-29:
hi @padamstx no update yet for MEFW version issue fix.  But i don't think it's the one causing your trackpoint issue.  have you updated your BIOS or other firmware as well?  And I think when you check your BIOS Setup Menu, you should have the latest MEFW already.  The MEFW being offered for update is just the same level but different version format so you can just ignore the MEFW update notice for now.  @mrhpearson @hughsie are you familiar with trackpoint issue?

padamstx, 2023-11-29:
My trackpoint issues started immediately after I did the MEFW update that failed.  I have heard reports from others that their touchpad is also unresponsive after a resume (after the failed MEFW update), but I use only the trackpoint and have disabled my touchpad :)

hughsie, 2023-11-30:
@kmauleon i don't know why the trackpad would be affected -- audio we know is affected due to the HDCP stuff -- but trackpad would be weird indeed.

hughsie, 2023-11-30:
Ohh hang on -- I remember something like this. I've got a pre-prod X1 nano that the touchpad works fine and then disappears after resume -- it's fixed in newer firmwares, but I can't run those because it's pre-prod and unlocked. No idea on the trackpoint I'm afraid.

padamstx, 2023-11-30:
I've heard reports of other folks with a P1 Gen 5 that observe no touchpad after a resume.  I prefer the trackpoint and have my touchpad disabled.  When I can spare a few minutes, I'll do an experiment to see if my touchpad is also lost after a resume.

padamstx, 2023-11-30:
One more data point... I downgraded the System Firmware from v 0.1.21 to 0.1.19 (the version I was at before this "no mouse after resume" problem started), and that didn't seem to affect anything.  Problem still occurs after a resume.

padamstx, 2023-12-01:
@kmaulean This morning I did the following:
1. I ran `sudo fwupdmgr upgrade` to update the System Firmware from v 0.1.19 back to the latest (v 0.1.21)
2. reboot (restart)
3. suspend
4. resume (have to wait ~75 secs for screen to come back to life): no trackpoint response (touchpad is fully operational)
6. reboot: still no trackpoint response
7. reboot again: still no trackpoint response
8. poweroff, then boot up: trackpoint is now responsive

So I guess this problem is not related to firmware or so it seems.
Please advise on next steps to diagnose.
For the record, I'm running Fedora 38 on this P1 Gen5 with the most recent kernel (`6.6.2-101.fc38.x86_64`)

hughsie, 2023-12-01:
> have to wait ~75 secs for screen to come back to life

That's odd too -- are there any warnings in dmesg?

padamstx, 2023-12-02:
There are some suspicious messages, but I'm not skilled enough to know if they are related to either problem (slow resume or no mouse after resume).
As an experiment, I cleared out dmesg, then I suspended and resumed with a short wait before resuming.  Here's what dmesg looks like right after I logged back in after resume:
```
[Fri Dec  1 18:03:09 2023] wlp0s20f3: deauthenticating from 94:10:3e:01:c1:34 by local choice (Reason: 3=DEAUTH_LEAVING)
[Fri Dec  1 18:03:11 2023] rfkill: input handler enabled
[Fri Dec  1 18:03:16 2023] PM: suspend entry (deep)
[Fri Dec  1 18:03:16 2023] Filesystems sync: 0.410 seconds
[Fri Dec  1 18:03:17 2023] Freezing user space processes
[Fri Dec  1 18:03:17 2023] Freezing user space processes completed (elapsed 0.001 seconds)
[Fri Dec  1 18:03:17 2023] OOM killer disabled.
[Fri Dec  1 18:03:17 2023] Freezing remaining freezable tasks
[Fri Dec  1 18:03:18 2023] Freezing remaining freezable tasks completed (elapsed 1.597 seconds)
[Fri Dec  1 18:03:18 2023] printk: Suspending console(s) (use no_console_suspend to debug)
[Fri Dec  1 18:03:20 2023] PM: suspend devices took 1.559 seconds
[Fri Dec  1 18:03:20 2023] ACPI: EC: interrupt blocked
[Fri Dec  1 18:03:20 2023] ACPI: PM: Preparing to enter system sleep state S3
[Fri Dec  1 18:03:20 2023] ACPI: EC: event blocked
[Fri Dec  1 18:03:20 2023] ACPI: EC: EC stopped
[Fri Dec  1 18:03:20 2023] ACPI: PM: Saving platform NVS memory
[Fri Dec  1 18:03:20 2023] Disabling non-boot CPUs ...
[Fri Dec  1 18:03:20 2023] smpboot: CPU 1 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 2 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 3 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 4 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 5 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 6 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 7 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 8 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 9 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 10 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 11 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 12 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 13 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 14 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 15 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 16 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 17 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 18 is now offline
[Fri Dec  1 18:03:20 2023] smpboot: CPU 19 is now offline
[Fri Dec  1 18:03:20 2023] ACPI: PM: Low-level resume complete
[Fri Dec  1 18:03:20 2023] ACPI: EC: EC started
[Fri Dec  1 18:03:20 2023] ACPI: PM: Restoring platform NVS memory
[Fri Dec  1 18:03:20 2023] Enabling non-boot CPUs ...
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 1 APIC 0x1
[Fri Dec  1 18:03:20 2023] CPU1 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 2 APIC 0x8
[Fri Dec  1 18:03:20 2023] CPU2 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 3 APIC 0x9
[Fri Dec  1 18:03:20 2023] CPU3 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 4 APIC 0x10
[Fri Dec  1 18:03:20 2023] CPU4 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 5 APIC 0x11
[Fri Dec  1 18:03:20 2023] CPU5 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 6 APIC 0x18
[Fri Dec  1 18:03:20 2023] CPU6 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 7 APIC 0x19
[Fri Dec  1 18:03:20 2023] CPU7 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 8 APIC 0x20
[Fri Dec  1 18:03:20 2023] CPU8 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 9 APIC 0x21
[Fri Dec  1 18:03:20 2023] CPU9 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 10 APIC 0x28
[Fri Dec  1 18:03:20 2023] CPU10 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 11 APIC 0x29
[Fri Dec  1 18:03:20 2023] CPU11 is up
[Fri Dec  1 18:03:20 2023] smpboot: Booting Node 0 Processor 12 APIC 0x30
[Fri Dec  1 18:03:20 2023] core: cpu_atom PMU driver: PEBS-via-PT 
[Fri Dec  1 18:03:20 2023] ... version:                5
[Fri Dec  1 18:03:20 2023] ... bit width:              48
[Fri Dec  1 18:03:20 2023] ... generic registers:      6
[... 126 lines cut ...]
[Fri Dec  1 18:04:39 2023] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[Fri Dec  1 18:04:39 2023] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[Fri Dec  1 18:04:39 2023] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[Fri Dec  1 18:04:39 2023] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[Fri Dec  1 18:04:39 2023] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[Fri Dec  1 18:04:39 2023] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[Fri Dec  1 18:04:39 2023] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[Fri Dec  1 18:04:39 2023] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[Fri Dec  1 18:04:39 2023] sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at snd_soc_dai_hw_params on iDisp3 Pin: -22
[... cut ...]
[Fri Dec  1 18:04:43 2023] wlp0s20f3: authenticated
[Fri Dec  1 18:04:43 2023] wlp0s20f3: associating to AP 94:10:3e:01:c1:34 with corrupt probe response
[Fri Dec  1 18:04:43 2023] wlp0s20f3: associate with 94:10:3e:01:c1:34 (try 1/3)
[Fri Dec  1 18:04:43 2023] wlp0s20f3: RX AssocResp from 94:10:3e:01:c1:34 (capab=0x511 status=0 aid=12)
[Fri Dec  1 18:04:43 2023] wlp0s20f3: associated
[Fri Dec  1 18:04:43 2023] wlp0s20f3: Limiting TX power to 20 (23 - 3) dBm as advertised by 94:10:3e:01:c1:34
```

padamstx, 2023-12-02:
Also just confirmed something else with respect to the "no trackpoint after resume" problem.  It turns out that after I resume (trackpoint is unresponsive), I can do a reboot (restart, not power off/on) any number of times (I tried 4) and the trackpoint doesn't come back.  But if I do a full power off/on type reboot, then I get the trackpoint back.   Somehow, the suspend is putting the trackpoint device in some sort of unusable state that even a restart won't fix.

padamstx, 2023-12-04:
@kmauleon I've opened https://github.com/fwupd/firmware-lenovo/issues/388
Thanks!

wtogami, 2023-12-09:
Note #387 is related where this bogus ME firmware version can result in a corrupted firmware when upgraded simultaneously as a legitimate system firmware. It is recoverable and bootable but the resulting system is crippled while the ME is corrupted.

kmauleon, 2023-12-11:
fix is under test by ODM owner... will update again for the release date... thank you

AImenes, 2023-12-12:
Hello,

I've observed a potential explanation for the version number discrepancy. This isn't a fix but an insight into why we might overlook the persistent message.

The latest driver is 16.1.27.2176 according to the LVFS website (https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN3JRP.firmware). However, within fwupd, it's also labeled as version 257.27.2176, as you can see on the website.

Key points:

The version appears as "16.1.27.2176" but is represented as "257.27.2176" in fwupd.

In the code, a limitation might prevent using four numbers in versioning. "16.1" seems hex-concatenated (0x10 . 0x1) to 257 (0x101).

BIOS displays the Intel ME Version as 16.1.27.2176.

During fwupd's version check, it retrieves "1.27.2176," potentially omitting the initial 16.1 merge, and hence ignoring "16". 
However, the 16.1.27.2176 version on LVFS is labeled 257.27.2176 on the website, causing a mismatch.

The issue may lie in fwupd not hex-concatenating "16.1" to 257 before comparing versions.

The hypothesis is that the driver isn't faulty; and the reason for crashing during update is that it is indeed already installed (verify this by looking in your BIOS for 16.1.27.2176). There's likely a parsing issue somewhere. 

I am not sure this has any practical consequenses. Linux does erroneously believe it lacks the latest driver. Maybe it potentially withholds other updates dependent on Intel ME due to this perceived discrepancy?

wtogami, 2023-12-12:
Practical consequences I've seen in #387 
* In isolation the ME flash process fails without consequence.
* In combination with a system firmware update the ME firmware can end up corrupted.

kmauleon, 2024-03-20:
ODM says they will target to set the firmware to stable by next week....

kmauleon, 2024-04-02:
ODM just set the firmware to stable please check... thank you very much

kmauleon, 2024-04-25:
closing this issue... please reopen if needed.  thank you very much

closed 2024-04-25 by kmauleon