fwupd/firmware-lenovo #363: Lenovo T580 Upgrade Intel Management Engine from 184.86.3909 to 184.94.4494 failing

dim6003, 2023-08-20:
**Describe the bug**
Attempts to upgrade UEFI and/or ME firmware (I've tried both together as well as separately) fail during the following POST with below messages
I have tested updating both or one at a time, without success. Every time I get : 
```
fwupd-efi version 1.4
WARNING: QueryCapsuleCapabilities failed, assuming EFIResetWarm: Unsupported
WARNING: Could not apply capsule update: Not Found
WARNING: Could not apply capsules: Not Found
Reset System
```
**OS**
Ubuntu 22.04.3 LTS up-to-date

**Steps to reproduce**
```
fwupdmgr  update
Devices with no available firmware updates: 
 • KINGSTON SNVS2000G
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 184.86.3909 to 184.94.4494?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ MEFW version update to 11.8.94.4494.and be aware that, in Intel® CSME FW     ║
║ 11.8.94.4491, the VCN increased to “331”.Therefore, downgrading from         ║
║ Intel® CSME FW 11.8.94.4494 version 3 to earlier versions is not possible.   ║
║                                                                              ║
║ 20L9CTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Downloading…             [***************************************]
Décompression…           [***************************************]
Décompression…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Redémarrage du périphérique…[***************************************]
Écriture…                [***************************************]
Planification...         [***************************************]
Décompression…           [***************************************]
Redémarrage du périphérique…[***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.30 to 0.1.36?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Important updates                                                            ║
║                                                                              ║
║ -Update includes a security fix.                                             ║
║                                                                              ║
║ New functions or enhancements                                                ║
║                                                                              ║
║ -Updated the Diagnostics module to version 04.30.000.                        ║
║                                                                              ║
║ -Updated the CPU microcode.                                                  ║
║                                                                              ║
║ 20L9CTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Downloading…             [***************************************]
Décompression…           [***************************************]
Décompression…           [***************************************]
Authenticating…          [***************************************]
[... 618 lines cut ...]
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img -> initrd.img-6.2.0-26-generic
├── initrd.img-5.19.0-50-generic
├── initrd.img-6.2.0-26-generic
├── initrd.img.old -> initrd.img-5.19.0-50-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.19.0-50-generic
├── System.map-6.2.0-26-generic
├── vmlinuz -> vmlinuz-6.2.0-26-generic
├── vmlinuz-5.19.0-50-generic
├── vmlinuz-6.2.0-26-generic
└── vmlinuz.old -> vmlinuz-5.19.0-50-generic

52 directories, 454 files
```

kmauleon, 2023-08-21:
hi @dim6003 you may want to check #319 for this one... thank you very much

dim6003, 2023-08-23:
Hi @kmauleon
Have read it, not sure where I should find a hint. Perhaps the way my efi partition has been formatted? 
[image]
Thanks

kmauleon, 2023-08-25:
hi @dim6003 this particular statement >> https://github.com/fwupd/firmware-lenovo/issues/319#issuecomment-1459620317

kmauleon, 2023-10-17:
@dim6003 question from ODM dev as below...
ODM DEV: For the issue , I could not reproduce the problem on my side , 
                  May I known is the user have tried modified the ESP setting ?

kmauleon, 2023-10-17:
@dim6003 ODM dev is asking if you have modified your partition settings. 
this link might help >> https://peterbabic.dev/blog/updating-uefi-bios-via-fwupd-on-thinkpad-t14-gen3/

kmauleon, 2024-05-06:
closing... no update since oct 2023

closed 2024-05-06 by kmauleon