fwupd/firmware-lenovo #365: ThinkPad L15 Gen3 - Firmware not updating

DanielBoettner, 2023-08-25:
This issues seems to be related to at least #300 and #361
As mentioned in #300 before. I was not able to fix this on my own for about 2 weeks by now.

The following update is "stuck"

[image]

> Lenovo ThinkPad L14/L15 Gen 3 Consumer ME: R1XMA13W/R1XMA13N Firmware Update
 • Update MEFW version to 16.1.27.2176.
 • Update PCHC version to 16.1.0.1014.
 • Update IOM version to 36.6.0.0.
 • Update PHY version to 14.530.508.8257.
 • Update TBT version to 16.0.0.1901.
 • Update PMC version to 160.1.0.1029.

In this post you'll finde the following information attached.

* general json like info
* output of fwupdmgr get-devices
* output of requested ESRT values from "/sys/firmware/efi/esrt/entries"
* bash script for the ESRT values

```bash
"Metadata" : {
                             "DistroId" : "ubuntu",
                             "DistroVersion" : "23.04"
                           },
...

                             {

                               "ReleaseId" : null,
                               "Protocol" : "org.nvmexpress",
                               "UpdateState" : 2,

                               "Plugin" : "nvme",
                               "VersionOld" : "1107ANLA",
                               "VersionNew" : "1108ANLA",

                               "Created" : 1682316243,
                               "Modified" : 1682341048,
                               "Metadata" : {
                                 "RuntimeVersion(org.freedesktop.gusb)" : "0.3.10",
                                 "HostSku" : "LENOVO_MT_21C3_BU_Think_FM_ThinkPad L15 Gen 3",
                                 "DistroVersion" : "22.10",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.8.4",
                                 "FwupdSupported" : "True",
                                 "HostFamily" : "ThinkPad L15 Gen 3",
                                 "KernelName" : "Linux",
                                 "CompileVersion(com.hughsie.libjcat)" : "0.1.9",
                                 "RuntimeVersion(org.kernel)" : "6.0.9-060009-generic",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.3.10",
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "DistroId" : "ubuntu",
                                 "RuntimeVersion(org.freedesktop.fwupd-efi)" : "1.0",
                                 "CpuArchitecture" : "x86_64",

                                 "HostVendor" : "LENOVO",
                                 "HostProduct" : "21C3007HGE",
                                 "KernelVersion" : "6.0.9-060009-generic",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.8.4",
                                 "FwupdTainted" : "False"
                               }
                             },
```
```
{

                               "ReleaseId" : null,
                               "Protocol" : "com.usi.dock",
[... 722 lines cut ...]
```

And if needed for other users. This is the small bash script used to generate this output.
Obviously needs root permissions to access the files.

```bash
#!/bin/bash

# Navigate to the ESRT entries directory
cd /sys/firmware/efi/esrt/entries || exit 1

# Iterate over each entry folder
for entry in entry*; do
  echo "=== $entry ==="
  cd "$entry" || continue
  
  # Iterate over each file in the entry folder and display its content with the filename prepended
  for file in *; do
    echo -n "$file: "
    cat "$file"
  done
  
  cd ..
done
```

kmauleon, 2023-08-25:
thanks for opening a separate issue @DanielBoettner ... already forwarded this to correct owner... will update once I received any feedback from them... thank you very much...

clixclix2, 2023-08-30:
I have the same identical issue: same HW (Lenovo L15 Gen3), same OS ( (K)Ubuntu 23.04 )

dominiksr, 2023-10-16:
I have the same problem, L14 gen 3. The new version has been out for a month, every attempt to update is unsuccessful. Debian testing

kmauleon, 2023-11-09:
@clixclix2 @dominiksr @stratosk thank you for reporting.  unfortunately, ODM dev is still investigating the issue

kmauleon, 2023-12-07:
everyone if you are already on MEFW 1.27.2176 which is the latest, please ignore 257.27.2176 update notice

dominiksr, 2024-02-09:
This incorrect version disappeared about a week ago. I think it can be closed.

closed 2024-02-11 by hughsie