fwupd/firmware-lenovo #366: Missing system firmware 1.61 cab for X1 Carbon gen 9

davemosk, 2023-08-25:
Getting error message "cannot find Lenovo-ThinkPad-X1Carbon9th-X1Yoga6th-SystemFirmware-1.62.cab in archive"

Full session:

 dave@hutt  ~  sudo fwupdmgr update                                                                       ✔  48s  09:24:06 AM 
[sudo] password for dave: 
Decompressing…           [-                                      ]Devices with no available firmware updates: 
 • ELAN0672:00 04F3:3187
 • Integrated Camera
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • MZVL2512HCJQ-00BL7
 • Prometheus
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.61 to 0.1.62?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo System Firmware Version 1.62                                          ║
║                                                                              ║
║ Important updates                                                            ║
║                                                                              ║
║ • Enhancement to address security vulnerabilities                            ║
║                                                                              ║
║ New functions or enhancements                                                ║
║                                                                              ║
║ • Updated the Diagnostics module to version 04.31.000                        ║
║                                                                              ║
║ 20XWCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
cannot find Lenovo-ThinkPad-X1Carbon9th-X1Yoga6th-SystemFirmware-1.62.cab in archive

hughsie, 2023-08-26:
what fwupd version please?

hughsie, 2023-08-26:
I suspect updating to fwupd >= 1.8.3 will fix this, although I can probably mitigate this on the LVFS too

hughsie, 2023-08-26:
I've also resigned the firmware on the LVFS without the extra tag that seems to be confusing things.

hughsie, 2023-08-26:
It'll take at least 4 hours before it'll appear on the CDN then you can do `fwupdmgr refresh --force` and then update again.

davemosk, 2023-08-27:
That seemed to work - thanks @hughsie !

davemosk, 2023-08-27:
Version is 1.7.9 fwiw

dave@hutt  ~  fwupdmgr --version                                                                               ✔  03:19:56 PM 
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.2.0-26-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

closed 2023-08-27 by davemosk