fwupd/firmware-lenovo #367: ThinkPad L14 Gen4 AMD: failed to update to 0: authentication signing error

Lucas-Florin, 2023-08-28:
I have a problem on my ThinkPad L14 Gen4 AMD with updates. 
`fwupdmgr get-devices`:
```
LENOVO 21H5CTO1WW
[...]
│   
├─UEFI Device Firmware:
│     Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               d6fb7af8-7862-42a3-b442-8f990f194941
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Failed
│     Update Error:       failed to update to 0: authentication signing error
│     GUID:               f601b368-963d-43a6-ae78-c4dfff74231d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    532
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               88523ddf-3aac-41f6-b15f-dc7bea434b39
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    0
│     Minimum Version:    15
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               f8637dbe-5762-0000-873a-910aa71c49e4
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
[...]
Devices that were not updated correctly:
 • UEFI Device Firmware (0 → 0.1.17)
Devices that have been updated successfully:
 • Prometheus (10.01.3478575 → 10.01.3478575)

```

`fwupdmgr get-history`:
```
LENOVO 21H5CTO1WW
│
├─Prometheus:
│ │   Device ID:          23ec719b6aabc2d2dac5176c232f0da7a21881b0
│ │   Previous version:   10.01.3478575
│ │   Update State:       Success
│ │   Last modified:      2023-08-25 15:42
│ │   GUID:               659f7e45-8d45-528d-b3c7-0695eed055f6
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Signed Payload
│ │ 
│ └─Prometheus:
│       New version:      10.01.3478575
│       Remote ID:        lvfs
│       Release ID:       10696
│       Summary:          Firmware for the Synaptics Prometheus Fingerprint Reader device
│       License:          Proprietary
│       Size:             425,5 kB
│       Created:          2021-06-09
│       Urgency:          Medium
│       Vendor:           Synaptics
│       Duration:         2 seconds
│       Description:      
│       New features and enhancements:
│       
│       • Fix an authentication issue on FOP project.
│     
└─UEFI Device Firmware:
  │   Device ID:          97b6fe9b220c7b9e1a3a1d9f404c00d4fe77ae7e
  │   Previous version:   0
  │   Update State:       Failed
  │   Update Error:       failed to update to 0: authentication signing error
  │   Last modified:      2023-08-25 15:44
  │   GUID:               f601b368-963d-43a6-ae78-c4dfff74231d
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad L14 L15 Gen3 AMD:
        New version:      0.1.17
        Remote ID:        lvfs
        Release ID:       27500
        Summary:          Lenovo ThinkPad L14 L15 Gen3 AMD Embedded controller Firmware
        License:          Proprietary
        Size:             1,1 MB
        Created:          2023-03-17
        Urgency:          Critical
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad L14 L15 Gen3 AMD Embedded controller Firmware
        
        • Fix "=", "(", ") "3 keys, actually output error characters on Numpad Keyboard for CR (hrvatski)
        • Add ECRAM[0x7F] bit5 GPIO control for TP4 FW update

```
Which LVFS package would I need to install? 
Here 0.1.17 is the newest stable version: 
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR1YHT.firmware
should I just wait until 0.1.18 becomes stable?

_Originally posted by @Lucas-Florin in https://github.com/fwupd/firmware-lenovo/issues/325#issuecomment-1693621202_

Lucas-Florin, 2023-08-28:
I tried installing the following: 
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR25HT.firmware
https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR25ET.BIOS.firmware
But I get the following messages: 
```
Decompressing…           [ -                                     ]
Specified firmware is older than installed '0.1.03 < 65539'

```
and 
```
Decompressing…           [             /                         ]
Specified firmware is older than installed '0.1.08 < 65544'
```
respectively.

hughsie, 2023-08-28:
@Lucas-Florin I think you need to use a newer fwupd version.

ChiWei-Chen, 2023-08-29:
@Lucas-Florin , FW team replied this is a known issue, your Gen 4 machine was offered a Gen 3 EC firmware, and they will release an official BIOS to resolve this issue around the middle of Sep.
But it would be great if you can use newer fwupd version to upgrade R25 FW manually.

Lucas-Florin, 2023-08-29:
Is there an easy way to install a newer fwupd version along with dependencies (such as through apt)? If not I think I will just wait for Ubuntu 23.10 to come out with the new fwupd version.

ChiWei-Chen, 2023-08-30:
@Lucas-Florin you can install via [snap](https://snapcraft.io/fwupd).
If you're using Ubuntu, please try this.
```
sudo apt update
sudo apt install snapd
sudo snap install fwupd
```

femto-code, 2023-10-09:
I have a somewhat similar issue...my system:

- OS: Kubuntu 23.04 (lunar)
- Hardware: 21H6S00900 ThinkPad L14 Gen 4 AMD Ryzen

In Discover (KDE Plasma default software store app) which is the only place I seem to be able to interact with lvfs firmware updates, I can see the following:

[image]

1. Why does it say "**Gen3** AMD System Firmware" when the hardware actually is **Gen4**?
2. Installation was apparently tried before (automatically) and it failed with the same error message as provided by @Lucas-Florin ) as I can tell from the `fwupdmgr` commands

Does anybody have a clue?

ChiWei-Chen, 2023-10-10:
@femto-code , this issue is expected to be resolved by the next BIOS version (1.11 30W), I'm checking the status with FW team.