fwupd/firmware-lenovo #368: Incorrect ME Firmware for ThinkPad X1 Extreme Gen 5

rustyx, 2023-09-07:
According to [Lenovo website](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-extreme-gen-5-type-21de-21df/21de/21de001qmh/downloads/ds559275-intel-management-engine-161-firmware-for-windows-11-version-21h2-or-later-10-version-20h2-or-later-thinkpad-p1-gen-5-x1-extreme-gen-5?category=Motherboard%20Devices%20%28core%20chipset,%20onboard%20video,%20PCIe%20switches%29), the latest Intel Management Engine Firmware is 16.1.27.2176.

According to the BIOS, ME version 16.1.27.2176 is already installed on my system.

However, `fwupdtool upgrade` shows:

```
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 1.27.2176 to 257.27.2176?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo Lenovo ThinkPad P1 Gen5 X1 Extreme Gen5 Consumer ME Firmware, for     ║
║ Hybrid Nonvpro                                                               ║
║                                                                              ║
║ 21DE001QMH must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
```

If I proceed with this upgrade, the flashing process in BIOS aborts at around 60%.

`fwupdtool get-devices` shows:

```
21DE001QMH
│
├─Embedded Controller:
│     Device ID:          632acf4927c0b5fb53519d6beed3b60adb73f1d5
│     Summary:            UEFI ESRT device
│     Current version:    0.1.10
│     Minimum Version:    0.1.10
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               f8963b51-51b6-4059-87c4-722967c16745
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:          0fef0a0c55f6442bffaebd774ae771341c89571b
│     Current version:    56.20
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Serial Number:      0001
│     GUIDs:              b687616d-e0bc-58d3-9022-8f0065eed4ca ← USB\VID_04F2&PID_B74F&REV_5620
│                         bcd2c020-e315-55aa-b7e1-7d2d7108037e ← USB\VID_04F2&PID_B74F
│     Device Flags:       • Updatable
│   
├─Intel Management Engine:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    1.27.2176
│     Vendor:             Lenovo (DMI:LENOVO)
│     GUID:               23192307-d667-4bdf-af1a-6059db171246
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
│     Summary:            UEFI ESRT device
│     Current version:    1.27.2176
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
[... 183 lines cut ...]
│     Summary:            UEFI ESRT device
│     Current version:    10551296
│     Minimum Version:    57374
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               2a197802-e469-4fa7-a37b-2d681bcf416f
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          11a7fbdd09ce583e58b899660fe10f147fd155a2
      Summary:            UEFI ESRT device
      Current version:    3599446
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               7dbe1162-8854-491e-a3c4-8b6d92e9db52
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

apsinthium, 2023-09-07:
Duplicate: [#360](https://github.com/fwupd/firmware-lenovo/issues/360)

I have the same issue as well. Could you let me know if your audio has stopped functioning since trying to install the firmware upgrade?

kmauleon, 2023-09-07:
@rustyx thanks for reporting... forwarded to ODM CO.... will feedback once I have an update

rustyx, 2023-09-08:
Audio still works for me @apsinthium. But I think I have a consumer version, not a corporate one. Model code 21DE001QMH

kmauleon, 2023-09-15:
hi have not yet heard from dev but just wondering if you could try as below...
fwupdmgr refresh --force
fwupdmgr reinstall

apsinthium, 2023-09-15:
@kmauleon Unfortunately, same as before. A month in, and no response from the developer; is this coming straight from Lenovo? If so, they should have better response times for something that renders audio unusable on many of their machines because of a bugged firmware released by their company.

kmauleon, 2023-09-15:
hi @apsinthium we are forwarding these issues to ODM dev/owner... and follow ups .... in the meantime, you may want to check #250 with similar issue and users have tried some procedures that worked.... thank you

kmauleon, 2023-12-11:
fix is under test by ODM owner... will update again for the release date... thank you

kmauleon, 2024-04-02:
ODM just set the firmware to stable please check... thank you very much

rustyx, 2024-04-04:
Thanks! Seems to be fixed now.

```none
# fwupdtool upgrade
Loading…                 [*****                                  ]09:50:45.922 GLib-GObject         value "6144" of type 'guint' is invalid or out of range for property 'last-attempt-status' of type 'guint'
Loading…                 [************************************** ]
Devices with no available firmware updates: 
 • Integrated Camera
 • Intel Management Engine
 • MZVL22T0HBLB-00B00
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • Prometheus
 • System Firmware
 • UEFI dbx
```

```none
# fwupdtool get-devices 5bfc5b91c18ef8b751d3d052af77b69ebf738038
LENOVO 21DE001QMH
│
└─Intel Management Engine:
      Device ID:          5bfc5b91c18ef8b751d3d052af77b69ebf738038
      Summary:            UEFI ESRT device
      Current version:    1.27.2176
      Minimum Version:    0.0.1
      Vendor:             Lenovo (DMI:LENOVO)
      Update State:       Success
      GUID:               012b2727-3f35-c20a-fb78-d7d4216ce715
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

closed 2024-04-04 by rustyx

kmauleon, 2024-04-05:
great! thank you very much for the feedback!